openvehicles/Open-Vehicle-Monitoring-System-3 #35: Vehicle Password confusing term

markwj, 2018-04-15:
The web interface uses the term 'vehicle password' that is confusing. Module or Server?

For v2, we standardised on using 'server password' for that. Best to change our interface, and the user guide, to match.

dexterbg, 2018-04-15:
I disagree. "Server password" is confusing more, I had used that term earlier and users filled in their user account password. To remove any doubt what is meant, I already added a help text to the field reading "Note: enter the password for the **vehicle ID account**, not your user account password".

How about changing the label to just "Password" (as on the v3 setup page) and the help text to "Note: enter the password for the **vehicle server account**, not your user account password"?

closed 2018-05-10 by dexterbg

demetan71, 2019-05-16:
Hi, I'm a Canadian user. I spent about 10 hours to figure it out. I'm not a newbies. I make every password the same. After passed another 5 hours whit my engeneer friend, we are able to finish the security set-up and keep data. But we can't connect the android app.

So, for a programmer who built this thing, it's maybe easy, but it's not.

Someone want to buy my unit? Or can I post a video of me put a hammer on this thing?