openvehicles/Open-Vehicle-Monitoring-System-3 #43: Use google DNS by default

markwj, 2018-04-18:
Rather than using ISP assigned DNS (either wifi or ppp), we should just simply use 8.8.8.8 and 8.8.4.4 by default. This simplifies things when switching between wifi and modem network connections.

mstegen, 2018-04-20:
There's also 1.1.1.1  (https://1.1.1.1/) if privacy is a concern.

Marvin8228, 2018-04-26:
I strongly disagree. All the wifi networks I'm connecting to, provide DNS servers and do not allow direct DNS traffic to public DNS servers (as for example 8.8.8.8). So with a fixed DNS server setting I'm stuck (as I'm now at the moment with the actual firmware release). I had to use the IP-address instead of the the server name to overcome this limitation.
I kindly request/suggest to reconsider this approach...

dexterbg, 2018-04-26:
There is no fixed DNS in the actual firmware release. You can set a DNS on the modem configuration page or by `config set network dns "…"`. Leave the DNS config empty to use the DNS assigned by your DHCP / provider.

We had issues on some mobile networks (i.e. Hologram in Germany) with their DNS not resolving some servers, that's why we introduced the DNS configuration. We do not intend to replace the configuration by a fixed DNS setup, this issue is just about the factory default DNS configuration.

Marvin8228, 2018-04-27:
Ok, thanks for the clarification.

I haven't had the chance to check the source-code yet, but I can clearly see that, as soon as I enter a hostname in the "Server V2 (MP)" configuration page, the system is trying to connect to the 8.8.8.8 server for DNS.

The output of the "network status" shows correct DNS server settings (the local ones). I have no static servers defined on the modem configuration page (in fact the modem is powered off).

If I allow traffic to 8.8.8.8 or set the host of the Server V2 configuration to an IP-address the communication and the secure channel is established immediately.

I also can see the requests in the tcpdump and/or firewall-log.
I'm using the latest build (3.1.004)

Anything I can check/provide?

dexterbg, 2018-04-27:
See above, this commit should solve all DNS issues.

The original problem was caused by the ppp interface overwriting the wifi DNS settings. As wifi has routing priority, the ppp DNS became unreachable. The manager now keeps the DNS assigned by the wifi and modem network and restores them along with the default route.

The commit also disables the builtin resolver of mongoose and uses the LWIP resolver for all lookups, so the 8.8.8.8 accesses should be gone as well. @Marvin8228 can you please test this? If you can't build a firmware, you can download my latest build from ovms.dexters-web.de/f/v3.1/current.bin (log: ovms.dexters-web.de/f/v3.1/current.log).

A manual DNS configuration has priority over all interface/connection specific DNS.

Marvin8228, 2018-04-28:
I tried the mentioned version from ovms.dexters-web.de: 3.1.004-88-g9d6bfb3/ota_1/main (build idf v3.1-dev-454-gdaef4b5c Apr 27 2018 20:58:26) and I can confirm that the OTA firmware update works with the dns name (ovms.dexters-web.de). So the module seems to use the correct DNS for that.

Unfortunately the connection to the V2 server still doesn't work. So for the metric upload the device is still trying to connect to 8.8.8.8 for a DNS lookup. I still see the requests in the tcpdump. The network status shows the correct DNS servers and also the default interface is correct (Wifi).

I tried with and without the modem enabled. No difference.

Should I perform a factory reset or anything else to try?
Sorry for the bad news.

dexterbg, 2018-04-28:
No, a factory reset would only have the same effect as clearing the DNS config.
I'll have another look at the mongoose code.

dexterbg, 2018-04-28:
Found & fixed a bug in mongoose that kept getaddrinfo() from actually being used. @Marvin8228 can you please try again? Update is on my server.

Marvin8228, 2018-04-29:
I can confirm that it's using the correct DNS server now.

No more requests to 8.8.8.8 but only to the DNS servers specified by DHCP.

Thank you very much!

dexterbg, 2018-04-29:
Thank you for tests & reports :)

closed 2018-04-29 by dexterbg