openvehicles/Open-Vehicle-Monitoring-System-3 #47: SSH memory issue on large outputs

dexterbg, 2018-04-24:
SSH occasionally cuts off large outputs due to low memory, i.e. when dumping the memory status of the "OVMS Events" task:

```
OVMS# mo m 3FFBFC50
Free 8-bit 49368/281968, 32-bit 8052/24308, SPIRAM 4126420/4194252
--Task--     Total DRAM D/IRAM   IRAM SPIRAM   +/- DRAM D/IRAM   IRAM SPIRAM
wifi               1400    128      0   6288         +0     +0     +0  +1684
============================ blocks dumped = 844
  t=OVMS Events s=   4 a=0x3f8073dc  3FFEAC84 BAAD5678 3F807439 3F809B40 3FFD96FC 00000018 00000000 3F80BB4A
  t=OVMS Events s=  12 a=0x3f807448  3FFC9814 3FFD1960 FEFEFE01 BAAD5678 3F8074C0 ABBA1234 3FFBFC50 00000054
  t=OVMS Events s=  84 a=0x3f807468  3F4108E4 3F40C8DC 3F448270 00000000 3F4102EC 3F807484 00000000 FEFEFE00
  t=OVMS Events s=  84 a=0x3f8074d0  3F4108E4 3F44825C 3F448270 40138C00 3F4799B4 3F8074EC 00000000 FEFEFE00
…
  t=OVMS Events s=  12 a=0x3ffdde1c  3FFE37F4 3FFD03A0 3FFDC7F4 BAAD5678 3FFDDE6C ABBA1234 3FFBFC50 0000002C
  t=OVMS Events s=  44 a=0x3ffdde3c  3F40ED6C 3FFDDE48 00000007 69686576 00656C63 FEFEFEFE FEFEFEFE 3FFDDE7C
  t=OVMS Events s=  12 a=0x3ffdde7c  40227100 00000000 3FFDC8DC BAAD5678 3FFDDEAC ABBA1234 3FFBFC50 0000000C

[49832 bytes lost due to low free memory]

E (38429735) ssh: 49832 output bytes lost due to low free memory
```

slcasner, 2018-04-25:
This should fix it.  Someone with v3.1 hardware please confirm.

dexterbg, 2018-04-26:
Confirmed.

closed 2018-04-26 by dexterbg