openvehicles/Open-Vehicle-Monitoring-System-3 #48: Web login is not secure

slcasner, 2018-04-24:
The web login is not secure, so in Safari the nav bar changes to "Website not secure" when the focus goes to the username field.  This is really not good enough for users.  What is needed to support https?

dexterbg, 2021-03-21:
Installing a certificate to enable https is now supported, but you still won't get any browser to accept a self-signed certificate as a "secure site".

closed 2021-03-21 by dexterbg

leres, 2021-03-21:
> Installing a certificate to enable https is now supported, but you still won't get any browser to accept a self-signed certificate as a "secure site".

What if you install the self-signed root CA in the browser?

dexterbg, 2021-03-21:
That should also work. Try and report.

leres, 2021-03-21:
> That should also work. Try and report.

It works!

slcasner, 2021-03-22:
Works for me, too.  Should our webserver implement redirect from http to https if a cert and key are installed?

dexterbg, 2021-03-22:
No, or at most optionally, and if so probably only for the main UI (web app). HTTPS has a much higher connect overhead so would introduce some seconds additional load time for all scripts calling the REST API or fetching files.