openvehicles/Open-Vehicle-Monitoring-System-3 #65: Quickstart configuration system

markwj, 2018-04-27:
Implement a quickstart configuration system for new deployments. Initial spec includes:

We can lead them through the setup in the following stages:

1. Require, as a pre-requisite, users to initially setup the module within range of a home wifi / cellular hotspot connection.

2. Factory default, use Access Point mode, SSID OVMS password “OVMSinit”, and no module password. Have them connect to the module over wifi Access Point, and http://192.168.4.1/.

3. The web interface then shows them a list of WiFi access points in range, and asks them to choose one and enter the password for that access point (clearly labelled as SSID, and "<SSID> Password”).

4. The module then reconfigures as Client + Access Point mode, and tries to connect to the client wifi, as well as keeping the access point connection open. If not ok, tell them and go back to #3 for them to try again.

5. When ok, proceed to ask them to update the firmware. This will do a simple OTA download over HTTP, then reboot.

6. When back up, in one screen ask them to enter all the details for their network, and configure appropriately:
* Use Wifi?
  * WiFi Mode (AccessPoint+SpecificClient, or ScanningClient)
  * OVMS Access Point SSID
  * New OVMS Access Point password (as OVMSinit is insecure beyond initial setup)
* DNS (default: google, or custom)
* Module password (as default empty is insecure beyond initial setup)
* Use OVMS Server v2?
  * Vehicle type
  * Vehicle ID
  * Vehicle Server password
  * Vehicle server (openvehicles, dexter, or custom)
* Use modem?
  * Modem APN, username and password (default: hologram)

7. Test each of the above, and show them the results. Set auto appropriately, and disable quickstart then reboot.

The point is to lead them through each of the steps, test each setting and show them the result.

dexterbg, 2018-04-29:
Prerequisites:
- allow background wifi scans without disrupting ap mode
- provide wifi command to retrieve scan results
- allow transition from ap to apclient mode and changing client SSIDs without disrupting ap mode
- add client connection status and failure reason to `wifi status`

dexterbg, 2019-01-06:
Status note: the setup wizard has been implemented without the wifi scan, so the user still needs to enter the SSID manually.

Options to further enhance the process:
- Auto server account creation → issue #182
- Support SmartConfig or WPS → issue #185
- Wifi scan w/o network disruption → issue #186

closed 2019-01-06 by dexterbg