openvehicles/Open-Vehicle-Monitoring-System-3 #75: Twizy: add command to read actual tuning from SEVCON

dexterbg, 2018-04-27:
V2 could only store tuning profiles and write them into the SEVCON. V3 should also be able to provide the other direction, i.e. create a profile from the actual registers of any SEVCON.

Gpcas9, 2018-06-11:
Also it would be nice if the V could lookup the current profile in the SEVCON and compare it with a saved one.

So the webbrowser would display the correct profile after a loss voltage/hard reboot.

Now you have to reprogam the profile after each reboot.

Does the automatic power control work if the OVMS thinks that the default profile is set?

dexterbg, 2018-06-17:
I think you misunderstood "base" and "live" in the display. "live" is the active profile, and that's saved and loaded on boot. There's no need to reprogram and auto power adjustments work based on the live profile as well.

The last "base" profile number is currently lost on a reboot, I'll look for a better solution.

dexterbg, 2018-06-17:
The last successfully applied base profile key is now also saved.
(commit 18b58c486b78d488d034fff8a8dcb423b99a71cc)

Gpcas9, 2018-06-17:
"Reprogramming" the base was only for the purpose that the OVMS shows it correcty in the Browser.
I am shure that the profile is still in the SEVON oin time of the reboot beeing done.

Wasn´t shure if the autopower will be active if he base profile is shown to be std.
Had the feeling that it only is enabled when you have choosen a tuning profile.

dexterbg, 2018-06-17:
You can see the auto power levels in the telemetry and the adjustment actions in the log (v-twizy, level debug).

dexterbg, 2021-12-30:
No active Twizy maintainer & no sponsor for this, so closing this.

closed 2021-12-30 by dexterbg