openvehicles/Open-Vehicle-Monitoring-System-3 #100: Optional IP routing between AP and PPP

dexterbg, 2018-04-28:
There are mobile service sites like www.datapass.de that need to be accessed from the mobile network because they use the calling IP address as a key.

Also, most mobile phones stop their GPRS connection when in wifi client mode. Bandwidth of the internal modem bus does not allow for a full speed mobile hotspot, but is sufficient to stay connected to messaging channels and/or a V2 server.

dexterbg, 2018-04-28:
Additional benefit: will stop phones nagging about a network without internet connectivity.

markwj, 2018-08-14:
This is not going to happen any time soon. Massive amount of work to do in the core freertos and lwip stacks (that don't support this functionality at the moment).