openvehicles/Open-Vehicle-Monitoring-System-3 #99: Can't turn off SIMCOM power

slcasner, 2018-04-28:
On v3.0 hardware, in case that is significant:

OVMS# power simcom off
Power mode of simcom is now off
I (62232813) simcom: State: Enter PoweringOff state
I (62232813) gsm-ppp: Shutdown (direct)
I (62232813) gsm-nmea: Shutdown (direct)
I (62232813) netmanager: Interface priority is st1 (192.168.1.84/255.255.255.0 gateway 192.168.1.254)
I (62232823) simcom: Power Cycle
I (62242353) simcom: State timeout, transition to 1
I (62242353) simcom: State: Enter CheckPowerOff state
I (62253363) simcom: State: Enter PoweringOff state
I (62253363) gsm-ppp: Shutdown (direct)
I (62253363) gsm-nmea: Shutdown (direct)
I (62253363) simcom: Power Cycle
I (62253363) netmanager: Interface priority is st1 (192.168.1.84/255.255.255.0 gateway 192.168.1.254)
I (62263353) simcom: State timeout, transition to 1
I (62263353) simcom: State: Enter CheckPowerOff state
I (62274363) simcom: State: Enter PoweringOff state
I (62274363) gsm-ppp: Shutdown (direct)
I (62274363) gsm-nmea: Shutdown (direct)
I (62274363) simcom: Power Cycle
I (62274363) netmanager: Interface priority is st1 (192.168.1.84/255.255.255.0 gateway 192.168.1.254)
I (62284353) simcom: State timeout, transition to 1
I (62284353) simcom: State: Enter CheckPowerOff state
I (62295363) simcom: State: Enter PoweringOff state
I (62295363) gsm-ppp: Shutdown (direct)
I (62295363) gsm-nmea: Shutdown (direct)
I (62295363) netmanager: Interface priority is st1 (192.168.1.84/255.255.255.0 gateway 192.168.1.254)
I (62295363) simcom: Power Cycle
I (62305353) simcom: State timeout, transition to 1
I (62305353) simcom: State: Enter CheckPowerOff state
OVMS# simcom setstate PoweredOff
I (62314923) simcom: State: Enter PoweredOff state
I (62314923) gsm-mux: Stop MUX

This stops the power cycle loop for a while, but later it starts up again (not sure what is the trigger).
More info:

OVMS# simcom status
SIMCOM
  Network Registration: NotRegistered
  State: PoweredOff
  Ticker: 383
  User Data: 0

  Mux
    Status: down
    Open Channels: 0
    Framing Errors: 0
    Last RX frame: 0 sec(s) ago
    RX frames: 0
    TX frames: 0

  PPP
    Not Connected
    Last Error: Undefined

  GPS
    Status: disabled
    Time: disabled
    NMEA: GPS/GLONASS Not Connected
OVMS# config list auto
auto (readable writeable)
  ext12v: no
  init: yes
  modem: no
  obd2ecu: 
  server.v2: no
  server.v3: no
  vehicle.type: TR
  wifi.mode: client
  wifi.ssid.ap: OVMS
  wifi.ssid.client: Revelstoke
OVMS#

dexterbg, 2018-04-28:
Steve, can you please add a log of that cycle at level verbose?

slcasner, 2018-04-28:
Logging level for * set to verbose
I (82923353) simcom: State timeout, transition to 1
I (82923353) simcom: State: Enter CheckPowerOff state
V (82934353) simcom: tx: 41 54 0d 0a                                     | AT..            
D (82934353) simcom: tx scmd ch=0 len=4   : AT||
V (82934363) simcom: rx: 41 54 0d                                        | AT.             
I (82934363) simcom: State: Enter PoweringOff state
I (82934363) gsm-ppp: Shutdown (direct)
I (82934363) gsm-nmea: Shutdown (direct)
V (82934363) simcom: tx: f9 11 ff 17 41 54 2b 43 47 50 53 3d 30 0d 0a f3 | ....AT+CGPS=0...
V (82934363) simcom: tx: f9                                              | .               
I (82934373) netmanager: Interface priority is st1 (192.168.1.84/255.255.255.0 gateway 192.168.1.254)
I (82934373) simcom: Power Cycle
V (82934373) simcom: rx: 0d 0a 4f 4b 0d 0a                               | ..OK..          
V (82934383) simcom: rx: 41 54 2b 43 47 50 53 3d 30 0d                   | AT+CGPS=0.      
V (82934393) simcom: rx: 0d 0a 4f 4b 0d 0a                               | ..OK..          
I (82944353) simcom: State timeout, transition to 1
I (82944353) simcom: State: Enter CheckPowerOff state
V (82955353) simcom: tx: 41 54 0d 0a                                     | AT..            
D (82955353) simcom: tx scmd ch=0 len=4   : AT||
V (82955363) simcom: rx: 41 54 0d                                        | AT.             
I (82955363) simcom: State: Enter PoweringOff state
I (82955363) gsm-ppp: Shutdown (direct)
I (82955373) gsm-nmea: Shutdown (direct)
V (82955373) simcom: tx: f9 11 ff 17 41 54 2b 43 47 50 53 3d 30 0d 0a f3 | ....AT+CGPS=0...
V (82955373) simcom: tx: f9                                              | .               
I (82955383) netmanager: Interface priority is st1 (192.168.1.84/255.255.255.0 gateway 192.168.1.254)
I (82955383) simcom: Power Cycle
V (82955383) simcom: rx: 0d 0a 4f 4b 0d 0a                               | ..OK..          
V (82955393) simcom: rx: 41 54 2b 43 47 50 53 3d 30 0d                   | AT+CGPS=0.      
V (82955393) simcom: rx: 0d 0a 4f 4b 0d 0a                               | ..OK..          
I (82965353) simcom: State timeout, transition to 1
I (82965353) simcom: State: Enter CheckPowerOff state
V (82976353) simcom: tx: 41 54 0d 0a                                     | AT..            
D (82976353) simcom: tx scmd ch=0 len=4   : AT||
V (82976363) simcom: rx: 41 54 0d                                        | AT.             
I (82976363) simcom: State: Enter PoweringOff state
I (82976363) gsm-ppp: Shutdown (direct)
I (82976373) gsm-nmea: Shutdown (direct)
V (82976373) simcom: tx: f9 11 ff 17 41 54 2b 43 47 50 53 3d 30 0d 0a f3 | ....AT+CGPS=0...
V (82976373) simcom: tx: f9                                              | .               
I (82976383) netmanager: Interface priority is st1 (192.168.1.84/255.255.255.0 gateway 192.168.1.254)
I (82976383) simcom: Power Cycle
V (82976383) simcom: rx: 0d 0a 4f 4b 0d 0a                               | ..OK..          
V (82976393) simcom: rx: 41 54 2b 43 47 50 53 3d 30 0d                   | AT+CGPS=0.      
V (82976403) simcom: rx: 0d 0a 4f 4b 0d 0a                               | ..OK..          
I (82986353) simcom: State timeout, transition to 1
I (82986353) simcom: State: Enter CheckPowerOff state

dexterbg, 2018-04-28:
Very strange, your modem echos all commands even after a power cycle, so the CheckPowerOff state assumes it's still alive. I cannot reproduce this even with an explicit `simcom cmd ATE1` before `power simcom off`, my modem doesn't echo anything in powered off state.

You can try giving `simcom cmd ATE0` before power-off, but this may be a hardware issue. Maybe TX and RX have a shortcut?

slcasner, 2018-04-28:
I have just now flashed to OTA0 the 3.1.003 image.  It starts up fine and powers off the modem:

I (735) cpu_start: Pro cpu up.
I (735) cpu_start: Starting app cpu, entry point is 0x400811c8
0x400811c8: call_start_cpu1 at /Users/casner/src/github/esp-idf/components/esp32/./cpu_start.c:243

I (719) cpu_start: App cpu up.
I (738) heap_init: Initializing. RAM available for dynamic allocation:
I (745) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (751) heap_init: At 3FFBBB20 len 000244E0 (145 KiB): DRAM
I (757) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (763) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (771) heap_init: At 400929F8 len 0000D608 (53 KiB): IRAM
I (776) cpu_start: Pro cpu start user code
I (122) ovms_main: Set default logging level for * to INFO
I (123) command: Initialising COMMAND (1000)
I (124) boot: Initialising BOOT (1100)
I (128) boot: Boot #1 reasons for CPU0=12 and CPU1=12
I (133) boot: Soft reset by user
I (137) events: Initialising EVENTS (1200)
I (142) config: Initialising CONFIG (1400)
I (147) time: Initialising TIME (1500)
I (152) script: Initialising SCRIPTS (1600)
I (156) script: Using DUKTAPE javascript engine
I (162) metrics: Initialising METRICS (1810)
I (166) metrics: Expanding DUKTAPE javascript engine
I (177) notify: Initialising NOTIFICATIONS (1820)
I (178) notify: Registered notification type info
I (183) notify: Registered notification type error
I (188) notify: Registered notification type alert
I (194) notify: Registered notification type data
I (199) location: Initialising LOCATIONS (1900)
I (205) location: Expanding DUKTAPE javascript engine
I (210) vehicle: Initialising VEHICLE Factory (2000)
I (216) pcp: Initialising POWER (4000)
I (220) max7317: Initialising MAX7317 EGPIO (4200)
I (225) sdcard: Initialising SD CARD (4400)
I (230) ota: Initialising OTA (4400)
I (235) can: Initialising CAN (4500)
I (240) simcom: Initialising SIMCOM (4600)
I (244) test: Initialising TEST (5000)
I (248) ovms-module: Initialising MODULE (5100)
I (253) vfs: Initialising VFS (5200)
I (258) ovms-server: Initialising OVMS Server (6000)
I (263) ovms-server-v2: Initialising OVMS V2 Server (6100)
I (269) ovms-server-v3: Initialising OVMS V3 Server (6200)
I (276) obd2ecu: Initialising OBD2ECU (7000)
I (281) canopen: Initialising CANopen (7000)
I (286) esp32wifi: Initialising ESP32WIFI (8000)
I (291) ovms-mdns: Initialising MDNS (8100)
I (296) webserver: Initialising WEBSERVER (8200)
I (302) ssh: Initialising SSH (8300)
I (305) re: Initialising RE Tools (8800)
I (310) netmanager: Initialising NETMANAGER (8999)
I (317) v-track: Registering Vehicle: TRACK (9000)
I (321) v-teslaroadster: Registering Vehicle: Tesla Roadster (9000)
I (327) v-teslamodels: Registering Vehicle: Tesla Model S (9000)
I (334) v-twizy: Registering Vehicle: Renault Twizy (9000)
I (340) v-obdii: Registering Vehicle: OBDII (9000)
I (346) v-none: Registering Vehicle: NONE (9000)
I (351) v-nissanleaf: Registering Vehicle: Nissan Leaf (9000)
I (358) v-kiasoulev: Registering Vehicle: Kia Soul EV (9000)
I (364) v-demo: Registering Vehicle: DEMO (9000)
I (369) version: Initialising Versioning (9900)
I (378) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (424) ovms_main: Executing on CPU core 0
I (424) ovms_main: Mounting CONFIG...
I (994) ovms_main: Registering default configs...
I (994) ovms_main: Starting HOUSEKEEPING...
[... 83 lines cut ...]
I (143564) webserver: HTTP GET /cfg/firmware
OVMS> simcom status
SIMCOM
  Network Registration: NotRegistered
  State: PoweredOff
  Ticker: 267
  User Data: 0

  Mux
    Status: down
    Open Channels: 0
    Framing Errors: 0
    Last RX frame: 0 sec(s) ago
    RX frames: 0
    TX frames: 0

  PPP
    Not Connected
    Last Error: Undefined

  GPS
    Status: disabled
    Time: disabled
    NMEA: GPS/GLONASS Not Connected
OVMS>

markwj, 2018-05-02:
Is this ok now? Seems ok for me.

slcasner, 2018-05-02:
On Tue, 1 May 2018, Mark Webb-Johnson wrote:

> Is this ok now? Seems ok for me.

No, it still loops.  If I issue "simcom setstate PoweredOff" I can
stop the loop for a while.  Not sure what event restarts it.  The
power is not off, though, this just stops the code that tries to cycle
power.

The blue LED is on.  When I issue "power simcom off" should that turn
it off?  It does not.

                                                        -- Steve

slcasner, 2018-05-02:
Nevermind, this was my fault.  I was trimming down the configuration to minimize RAM usage on my v3.0 hardware and that included CONFIG_OVMS_COMP_MAX7317.  Tonight I read the simcom.cpp code to discover that the power control for the SIMCOM modem goes through the MAX7317, so that is why the power would not turn off.

closed 2018-05-02 by slcasner