openvehicles/Open-Vehicle-Monitoring-System-3 #112: Wifi/netmanager: reconnect/fallback strategy if STA w/o internet

dexterbg, 2018-05-08:
The following log shows a situation where the wifi client could connect to a network that normally provides internet access. After the reconnect on timestamp 51458942, no internet access and no network internal access was possible, leaving the module stuck without any connectivity -- user needed to reboot manually.

Possibly another issue here is the empty DNS assigned by the wifi network -- but it otherwise works because the user manually set the Google DNS servers.

```
I (39740862) netmanager: Interface priority is pp2 (10.52.38.134/255.255.255.255 gateway 10.64.64.64)
I (39740912) netmanager: Set DNS#0 8.8.8.8
I (39740962) netmanager: Set DNS#1 8.8.4.4
I (39741022) netmanager: WIFI client down (with MODEM up): reconfigured for MODEM priority
I (39741082) esp32wifi: STA disconnected with reason 200
I (51022802) esp32wifi: Found SSID Hausnetz - trying to connect
I (51025262) wifi: n:11 0, o:11 0, ap:255 255, sta:11 0, prof:1
I (51026072) wifi: state: init -> auth (b0)
I (51026142) wifi: state: auth -> assoc (0)
I (51026212) wifi: state: assoc -> run (10)
I (51026292) wifi: connected with Hausnetz, channel 11
D (51026532) netmanager: Saved DNS#0 192.168.10.1
D (51026582) netmanager: Saved DNS#1 0.0.0.0
D (51026632) netmanager: Saved DNS#2 0.0.0.0
I (51026692) netmanager: Interface priority is st1 (192.168.10.27/255.255.255.0 gateway 192.168.10.1)
I (51026742) netmanager: Set DNS#0 8.8.8.8
I (51026782) netmanager: Set DNS#1 8.8.4.4
I (51026832) netmanager: WIFI client up (with MODEM up): reconfigured for WIFI client priority
I (51026882) esp32wifi: STA got IP with SSID: Hausnetz, MAC: 30:ae:a4:3f:6b:4c, IP: 192.168.10.27, mask: 255.255.255.0, gw: 192.168.10.1
I (51029272) wifi: pm start, type:0
I (51043782) wifi: bcn_timout,ap_probe_send_start
I (51054362) wifi: bcn_timout,ap_probe_send_start
I (51067372) wifi: bcn_timout,ap_probe_send_start
I (51076472) wifi: bcn_timout,ap_probe_send_start
I (51085552) wifi: bcn_timout,ap_probe_send_start
I (51097372) wifi: bcn_timout,ap_probe_send_start
I (51106692) wifi: bcn_timout,ap_probe_send_start
I (51124382) wifi: bcn_timout,ap_probe_send_start
I (51133462) wifi: bcn_timout,ap_probe_send_start
I (51159232) wifi: bcn_timout,ap_probe_send_start
I (51168312) wifi: bcn_timout,ap_probe_send_start
I (51209482) wifi: bcn_timout,ap_probe_send_start
I (51218552) wifi: bcn_timout,ap_probe_send_start
I (51227642) wifi: bcn_timout,ap_probe_send_start
I (51236742) wifi: bcn_timout,ap_probe_send_start
I (51272162) wifi: bcn_timout,ap_probe_send_start
I (51330732) wifi: bcn_timout,ap_probe_send_start
I (51333292) wifi: ap_probe_send over, resett wifi status to disassoc
I (51333352) wifi: state: run -> init (1)
I (51333412) wifi: pm stop, total sleep time: 0/304087149
I (51333412) wifi: n:11 0, o:11 0, ap:255 255, sta:11 0, prof:1
I (51333512) netmanager: Interface priority is pp2 (10.52.38.134/255.255.255.255 gateway 10.64.64.64)
I (51333562) netmanager: Set DNS#0 8.8.8.8
I (51333622) netmanager: Set DNS#1 8.8.4.4
I (51333672) netmanager: WIFI client down (with MODEM up): reconfigured for MODEM priority
I (51333722) esp32wifi: STA disconnected with reason 200
I (51452802) esp32wifi: Found SSID Hausnetz - trying to connect
I (51455262) wifi: n:11 0, o:11 0, ap:255 255, sta:11 0, prof:1
I (51456072) wifi: state: init -> auth (b0)
I (51456142) wifi: state: auth -> assoc (0)
I (51456212) wifi: state: assoc -> run (10)
I (51458292) wifi: connected with Hausnetz, channel 11
E (51458452) v-twizy: Twizy type unknown: SEVCON type 0x00000000
D (51458562) netmanager: Saved DNS#0 192.168.10.1
D (51458612) netmanager: Saved DNS#1 0.0.0.0
D (51458662) netmanager: Saved DNS#2 0.0.0.0
I (51458732) netmanager: Interface priority is st1 (192.168.10.27/255.255.255.0 gateway 192.168.10.1)
I (51458782) netmanager: Set DNS#0 8.8.8.8
I (51458832) netmanager: Set DNS#1 8.8.4.4
I (51458882) netmanager: WIFI client up (with MODEM up): reconfigured for WIFI client priority
I (51458942) esp32wifi: STA got IP with SSID: Hausnetz, MAC: 30:ae:a4:3f:6b:4c, IP: 192.168.10.27, mask: 255.255.255.0, gw: 192.168.10.1
I (51459262) wifi: pm start, type:0
E (51496002) ovms-server-v2: Status: Error: Connection failed
E (51533302) ovms-server-v2: Status: Error: Connection failed
E (51570302) ovms-server-v2: Status: Error: Connection failed
E (51608332) ovms-server-v2: Status: Error: Connection failed
E (51646302) ovms-server-v2: Status: Error: Connection failed
E (51683302) ovms-server-v2: Status: Error: Connection failed
E (51721372) ovms-server-v2: Status: Error: Connection failed
E (51759372) ovms-server-v2: Status: Error: Connection failed
```
… and so on until reboot.

markwj, 2018-05-11:
Hard to know without 'network status', 'wifi status', and other test output (such as whether the module was reachable via ping to 192.168.10.27. But, that zero DNS is really strange. Either the user's wifi is not correctly setup, or the wifi IP allocation and ancillary stuff didn't work correctly.

I guess if N connection attempts over wifi failed, and modem was available, we could set interface priority to the modem? Could do that via a callback to NetManager to let it know a connection attempt failed; the logic could then be in NetManager (which is the best place for it).

Wireheadbe, 2021-03-08:
I've had this a couple of times as well. OVMS is registered to my home network (for e.g. updates etc..) - but when driving away, it tries to hold on to it, causing the modem not to be used, and thus missing updates. Maybe make use of the watchdog in this case to reboot the device, if connection to V2/V3 fails.

akovacs21, 2021-08-02:
I have a similar issue. When i leave home WIFI the modem is not  connecting correctly to the local GSM network. After a reboot it connects without a problem. I would be happy if home WIFI connection lost for X amount of time and no active GSM network connection then a reboot happens in every 30 minutes or something similar  solution...

RowdyDoc, 2021-08-15:
I seem to be having a similar issue.
I don't use a SIM in the unit and just rely on home wifi and my mobile hotspot wifi when driving (ie always using wifi for connectivity - scanning mode).  However, the unit seems to disconnect from wifi and then will not ever reconnect again.  I power cycle it to get access again, and after doing that I notice that it has been collecting the 12VDC data throughout the period that it has been offline (ie the unit was at least partially operational).
I am getting this on a daily basis - so if there is some log data that would be useful I can probably supply it.
My network shows the correct DNS at all times when the unit is online.
Also - the direct access to the web client on the unit drops out at the same time as the OVMS app (which is why I suspect a failure to reconnect on wifi).