openvehicles/Open-Vehicle-Monitoring-System-3 #120: Crash in mongoose mbuf_insert()

dexterbg, 2018-05-10:
This crash always has the same signature, the assertion `a->len <= a->size` in mbuf_insert() fails on a transmission from server_v2.

```
0x40102b13 is in __assert_func (../../../.././newlib/libc/stdlib/assert.c:63).
0x400fcb5e is in mbuf_insert (/home/balzer/esp/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/components/mongoose/mongoose/mongoose.c:1507).
1502	size_t mbuf_insert(struct mbuf *a, size_t off, const void *buf, size_t) WEAK;
1503	size_t mbuf_insert(struct mbuf *a, size_t off, const void *buf, size_t len) {
1504	  char *p = NULL;
1505	
1506	  assert(a != NULL);
1507	  assert(a->len <= a->size);
1508	  assert(off <= a->len);
1509	
1510	  /* check overflow */
1511	  if (~(size_t) 0 - (size_t) a->buf < len) return 0;
0x400fcc26 is in mbuf_append (/home/balzer/esp/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/components/mongoose/mongoose/mongoose.c:1537).
1532	  return len;
1533	}
1534	
1535	size_t mbuf_append(struct mbuf *a, const void *buf, size_t len) WEAK;
1536	size_t mbuf_append(struct mbuf *a, const void *buf, size_t len) {
1537	  return mbuf_insert(a, a->len, buf, len);
1538	}
1539	
1540	void mbuf_remove(struct mbuf *mb, size_t n) WEAK;
1541	void mbuf_remove(struct mbuf *mb, size_t n) {
0x400fcc3c is in mg_socket_if_tcp_send (/home/balzer/esp/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/components/mongoose/mongoose/mongoose.c:3436).
3431	  return 0;
3432	}
3433	
3434	size_t mg_socket_if_tcp_send(struct mg_connection *nc, const void *buf,
3435	                           size_t len) {
3436	  return mbuf_append(&nc->send_mbuf, buf, len);
3437	}
3438	
3439	size_t mg_socket_if_udp_send(struct mg_connection *nc, const void *buf,
3440	                           size_t len) {
0x400ffa52 is in mg_send (/home/balzer/esp/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/components/mongoose/mongoose/mongoose.c:2688).
2683	size_t mg_send(struct mg_connection *nc, const void *buf, int len) {
2684	  nc->last_io_time = (time_t) mg_time();
2685	  if (nc->flags & MG_F_UDP) {
2686	    len = nc->iface->vtable->udp_send(nc, buf, len);
2687	  } else {
2688	    len = nc->iface->vtable->tcp_send(nc, buf, len);
2689	  }
2690	
2691	  if (!(nc->flags & MG_F_SEND_IMMEDIATELY) ||
2692	      (nc->flags & MG_F_CLOSE_IMMEDIATELY) ||
0x4012343e is in OvmsServerV2::Transmit(char const*) (/home/balzer/esp/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/components/ovms_server_v2/src/ovms_server_v2.cpp:692).
687	  RC4_crypt(&m_crypto_tx1, &m_crypto_tx2, (uint8_t*)s, len);
688	
689	  char* buf = new char[(len*2)+4];
690	  base64encode((uint8_t*)s, len, (uint8_t*)buf);
691	  strcat(buf,"\r\n");
692	  mg_send(m_mgconn, buf, strlen(buf));
693	
694	  delete [] buf;
695	  delete [] s;
696	  }
0x4012845d is in OvmsServerV2::TransmitNotifyData() (/home/balzer/esp/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/main/./ovms.h:63).
58	  T* allocate(std::size_t n)
59	    {
60	    auto p = static_cast<T*>(ExternalRamMalloc(n*sizeof(T)));
61	    return p;
62	    }
63	  void deallocate(T* p, std::size_t) noexcept { std::free(p); }
64	  };
65	template <class T, class U>
66	bool operator==(const ExtRamAllocator<T>&, const ExtRamAllocator<U>&) { return true; }
67	template <class T, class U>
0x4012878c is in OvmsServerV2::Ticker1(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, void*) (/home/balzer/esp/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/components/ovms_server_v2/src/ovms_server_v2.cpp:1643).
1638	    if (m_now_firmware) TransmitMsgFirmware();
1639	    if (m_now_capabilities) TransmitMsgCapabilities();
1640	
1641	    if (m_pending_notify_alert) TransmitNotifyAlert();
1642	    if (m_pending_notify_error) TransmitNotifyError();
1643	    if (m_pending_notify_info) TransmitNotifyInfo();
1644	
1645	    if (m_pending_notify_data)
1646	      {
1647	      TransmitNotifyData();
```

dexterbg, 2018-05-12:
Just happened again, this time with last words in the SD log:

```
I (23624140) ovms-server-v2: Status: Disconnected
I (23624190) ovms-server-v2: Send MP-0 h2716,1,RT-BAT-C,11,86400,2,1,4005,3820,4005,10,29,27,29,1
```

So maybe there still is a race condition on disconnection.

dexterbg, 2018-05-22:
No disconnect issue, just happened again without a prior disconnect.

```
0x4010598b is in __assert_func (../../../.././newlib/libc/stdlib/assert.c:63).
0x400ff9d6 is in mbuf_insert (/home/balzer/esp/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/components/mongoose/mongoose/mongoose.c:1507).
1502	size_t mbuf_insert(struct mbuf *a, size_t off, const void *buf, size_t) WEAK;
1503	size_t mbuf_insert(struct mbuf *a, size_t off, const void *buf, size_t len) {
1504	  char *p = NULL;
1505	
1506	  assert(a != NULL);
1507	  assert(a->len <= a->size);
1508	  assert(off <= a->len);
1509	
1510	  /* check overflow */
1511	  if (~(size_t) 0 - (size_t) a->buf < len) return 0;
0x400ffa9e is in mbuf_append (/home/balzer/esp/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/components/mongoose/mongoose/mongoose.c:1537).
1532	  return len;
1533	}
1534	
1535	size_t mbuf_append(struct mbuf *a, const void *buf, size_t len) WEAK;
1536	size_t mbuf_append(struct mbuf *a, const void *buf, size_t len) {
1537	  return mbuf_insert(a, a->len, buf, len);
1538	}
1539	
1540	void mbuf_remove(struct mbuf *mb, size_t n) WEAK;
1541	void mbuf_remove(struct mbuf *mb, size_t n) {
0x400ffab4 is in mg_socket_if_tcp_send (/home/balzer/esp/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/components/mongoose/mongoose/mongoose.c:3436).
3431	  return 0;
3432	}
3433	
3434	size_t mg_socket_if_tcp_send(struct mg_connection *nc, const void *buf,
3435	                           size_t len) {
3436	  return mbuf_append(&nc->send_mbuf, buf, len);
3437	}
3438	
3439	size_t mg_socket_if_udp_send(struct mg_connection *nc, const void *buf,
3440	                           size_t len) {
0x401028ca is in mg_send (/home/balzer/esp/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/components/mongoose/mongoose/mongoose.c:2688).
2683	size_t mg_send(struct mg_connection *nc, const void *buf, int len) {
2684	  nc->last_io_time = (time_t) mg_time();
2685	  if (nc->flags & MG_F_UDP) {
2686	    len = nc->iface->vtable->udp_send(nc, buf, len);
2687	  } else {
2688	    len = nc->iface->vtable->tcp_send(nc, buf, len);
2689	  }
2690	
2691	  if (!(nc->flags & MG_F_SEND_IMMEDIATELY) ||
2692	      (nc->flags & MG_F_CLOSE_IMMEDIATELY) ||
0x401263ba is in OvmsServerV2::Transmit(char const*) (/home/balzer/esp/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/components/ovms_server_v2/src/ovms_server_v2.cpp:692).
687	  RC4_crypt(&m_crypto_tx1, &m_crypto_tx2, (uint8_t*)s, len);
688	
689	  char* buf = new char[(len*2)+4];
690	  base64encode((uint8_t*)s, len, (uint8_t*)buf);
691	  strcat(buf,"\r\n");
692	  mg_send(m_mgconn, buf, strlen(buf));
693	
694	  delete [] buf;
695	  delete [] s;
696	  }
0x4012b431 is in OvmsServerV2::TransmitNotifyData() (/home/balzer/esp/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/components/ovms_server_v2/src/ovms_server_v2.cpp:1389).
1384	      << e->m_id
1385	      << ","
1386	      << monotonictime - e->m_created
1387	      << ","
1388	      << msg;
1389	    Transmit(buffer.str().c_str());
1390	    m_pending_notify_data_last = e->m_id;
1391	
1392	    // use max 500 ms of the ticker.1 event time per run:
1393	    if (esp_log_timestamp() - starttime > 500)
0x4012b760 is in OvmsServerV2::Ticker1(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, void*) (/home/balzer/esp/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/components/ovms_server_v2/src/ovms_server_v2.cpp:1647).
1642	    if (m_pending_notify_error) TransmitNotifyError();
1643	    if (m_pending_notify_info) TransmitNotifyInfo();
1644	
1645	    if (m_pending_notify_data)
1646	      {
1647	      TransmitNotifyData();
1648	      }
1649	    else if (m_pending_notify_data_retransmit > 0 && --m_pending_notify_data_retransmit == 0)
1650	      {
1651	      // check for retransmissions:
```

markwj, 2018-05-23:
Is this easily repeatable?

I am wondering if we have a ‘mongoose is not thread safe’ issue here? Perhaps mg_send being called from two threads (housekeeping for a ticker, and mongoose task for a callback)? Unless something is fundamentally broken inside mongoose, I can’t think of any other explanation for this.

We have a m_mgconn_mutex and that is used everywhere I can find, except OvmsServerV2::SendLogin - so I’ve just added it. But, it doesn’t seem that this could cause the issue.

In normal operation, mg_send is called from threads other than the mongoose event loop. Could that be causing the issue? Could the mongoose task be removing stuff from the send buffer, as it is being added using mg_send by another task? In that case, our mutex wouldn’t help.

https://groups.google.com/forum/#!topic/mongoose-users/YgcFd0Sss3o <https://groups.google.com/forum/#!topic/mongoose-users/YgcFd0Sss3o>

One option would be to put something like mg_send (and perhaps others similar functions) in OvmsNetManager. Then add a mutex there. But that is messy and slow (as the actual work is in mg_mgr_poll).

The mongoose documentation for mg_broadcast has this:

/*
 * Passes a message of a given length to all connections.
 *
 * Must be called from a thread that does NOT call `mg_mgr_poll()`.
 * Note that `mg_broadcast()` is the only function
 * that can be, and must be, called from a different (non-IO) thread.
 *
 * `func` callback function will be called by the IO thread for each
 * connection. When called, the event will be `MG_EV_POLL`, and a message will
 * be passed as the `ev_data` pointer. Maximum message size is capped
 * by `MG_CTL_MSG_MESSAGE_SIZE` which is set to 8192 bytes.
 */
void mg_broadcast(struct mg_mgr *mgr, mg_event_handler_t cb, void *data,
                  size_t len);

Maybe we need to have OvmsNetManager provide a mg_send style function that adds it to a buffer and then calls mg_broadcast to do the actual transmission? That callback would be in the mongoose task context.

Alternatively, perhaps a simpler implementation would be to use mg_set_timer to trigger actions in the mongoose loop, rather than Ticker1 (and others)?

Regards, Mark.

dexterbg, 2018-05-23:
Mark,

m_mgconn_mutex was meant to protect the connection from being invalidated during a transmit on an established connection, so it wasn't necessary for SendLogin.

The crash is not reproducable, but I think you may be right. The intended way to feed data into an mbuf is from within the event callback, and we violate that rule in multiple ways.

mg_set_timer() would only cover timer events, not events like "system.modem.received.ussd" or sends from the notification callback.

I've experimented with mg_broadcast() for the webserver, it wasn't working then but there is now a fix in the mongoose dev branch for the issue. mg_broadcast() itself isn't thread safe, but that's easy to fix by a wrapper in Netmanager. The advantage is, this wakes up the mongoose select immediately, so should be efficient.

Disadvantage is, we would need to make the broadcast call synchronous. That's easy to add to mongoose, I already thought about doing that, but the real issue with that is we cannot know how long it takes to execute. The mongoose loop may be busy with a large transmission or a slow server connection or stuck somewhere else for even seconds -- we've seen this happen.

Also, needing to supply callback functions for all sends isn't really elegant.

How about solving this on the mbuf level? I.e. adding a mutex … OK, I just remembered having read this in a mongoose issue, and here it is:
https://github.com/cesanta/mongoose/issues/758#issuecomment-279121941
…from the mongoose master himself ;)

A general implementation of course should be using a POSIX mutex as Sergey sketched. Not sure how much overhead a POSIX mutex has for us. Is it safe to assume the esp-idf implementation maps that API directly to RTOS mutexes…

…yes, it does, there's really not much overhead (see pthread.c). So we can do this even as a general addition for the mongoose project. I think I'll give that a try.

Regards,
Michael

dexterbg, 2018-05-23:
Just saw,  an implementation has also just been added to that issue:
https://github.com/saidinesh5/mongoose/commit/2f24adfeb39eb12b98c03667f4e40f7c0ac0dc74

But having had a look at the mbuf system already, I think that is a very inefficient solution, as it adds a mutex to every mbuf, and it needs to recreate the mutex on every free/reinit. mbufs are used at various places for temporary copies, so this IMO adds unnecessary overhead.

We basically only need to protect the send and recv mbufs, so I'll try another way.

dexterbg, 2018-05-23:
It's also a wrong implementation, as the send and recv mbufs are not initialized explicitly.

saidinesh5, 2018-05-24:
@dexterbg wow. Didn't expect this to get picked up so fast!
I made the implementation based on what cpq mentioned here: https://github.com/cesanta/mongoose/issues/758

Could you share your fixes once you they are ready?

dexterbg, 2018-05-24:
I've been running the module all day long and stressed it by doing a slow (long) charge with multiple web and ssh sessions on both wifi AP & client networks as well as a running server v2 connection.

saidinesh5, 2018-05-27:
@dexterbg sorry for my delayed reply. Thanks for the fixes. I wasn't sure If mg_connection was the only place which needs the mutexes - so i ended up initializing the mutexes unconditionally in mbuf_init ..

So do you have any plans on upstreaming your changes? I don't like the idea of "maintaining" my fork of mongoose - so I was wondering if i should apply your improvements to my mongoose and upstream it or if you'd send a pull reuqest to them yourself.

dexterbg, 2018-05-27:
@saidinesh5 We've a running pull request:
https://github.com/cesanta/mongoose/pull/894

It's been open for a while and contains earlier changes as well.

dexterbg, 2018-06-14:
Not a single crash of this type since the , so I guess this is fixed.

closed 2018-06-14 by dexterbg