openvehicles/Open-Vehicle-Monitoring-System-3 #127: Wifi TCP connections not closing

dexterbg, 2018-05-18:
This has happened both with ssh and webserver (websocket) connections and with both AP and STA mode: an open connection does not get terminated when the client gets out of wifi range.

The server then continues to send messages, growing the send buffer until running out of memory -- or reconnecting to the client: one time an open ssh session continued after the module had been disconnected from the wifi network for about 50 minutes. The ssh session then tried to output all log messages stored during the 50 minutes (log monitoring was enabled).

slcasner, 2018-05-18:
This a deficiency in Mongoose or LWIP, right?  Or do I need to receive wifi events in ssh and force the connection closed?

dexterbg, 2018-05-18:
I don't think the application (ssh/web server) should need to handle interface specific events. LWIP should close active connections that can no longer be routed after an interface went down. Mongoose would then call the application's event handler with MG_EV_CLOSE just like with a normal termination.

It seems either the Wifi stack doesn't properly promote the interface event to LWIP, or our wifi component and/or netmanager need to take care of this.

Forcing a close on wifi down could be a possible workaround but involves checking the IP routing of all connections. Another option may be to check if the send buffer didn't get flushed for some time in the MG_EV_TIMER callback.

dexterbg, 2018-05-18:
Yet another esp-idf wifi stack anomaly?
It seems we actually do currently need to handle the esp-idf wifi events:

http://esp-idf.readthedocs.io/en/latest/api-guides/wifi.html#system-event-sta-disconnected
> Upon receiving this event, the event task will shut down the station’s LwIP netif and notify the LwIP task to clear the UDP/TCP connections which cause the wrong status to all sockets. For socket-based applications, the application callback needs to close all sockets and re-create them, if necessary, upon receiving this event.

http://esp-idf.readthedocs.io/en/latest/api-guides/wifi.html#system-event-ap-stadisconnected
> When this event happens, the event task will do nothing, but the application event callback needs to do something, e.g., close the socket which is related to this station, etc.

That "wrong status" possibly also explains why the SOF_KEEPALIVE doesn't catch this case.
So we need to close wifi sockets on receiving these events.

dexterbg, 2018-05-20:
FYI: I've done a first draft implementation of a central auto cleanup of disconnected sockets. It's done by the netmanager, so the components don't need to take care of this. It works but needs some more refinement.

closed 2018-05-21 by dexterbg