openvehicles/Open-Vehicle-Monitoring-System-3 #131: Status site is visible without login

Gpcas9, 2018-06-11:
I allways thougt the first page the browser should dislplay would be the login page if no user is logged in.

Instead the statuspage will be shown.
Is this intentional?

Every click after that will lead to the login page.

dexterbg, 2018-06-12:
Intended behaviour is the browser can start with any valid URI (i.e. from a bookmark or homescreen launch icon), login page is shown as needed.

If you see the status page first, that's because your browser opens that page. If it's shown without a login, it's because your browser has resumed a previous session from the App cache, it doesn't load the page from the web server in that case. On Android, browser sessions are killed by pushing the "back" icon twice or by removing the App from the history.

closed 2018-06-12 by dexterbg

dexterbg, 2018-06-14:
That's the home/menu screen, and it's open by design to allow access to open functions. There are currently no open functions in the framework, but later on you will be able to add some by the plugin mechanism.