openvehicles/Open-Vehicle-Monitoring-System-3 #136: API documentation

boazdejong, 2018-07-01:
### Question
Where can I find documentation for the public API? I'd like to prototype a React Native mobile client application similar to the current mobile apps.

markwj, 2018-07-05:
You can find the v2 protocol guide here:

https://github.com/openvehicles/Open-Vehicle-Monitoring-System/raw/master/docs/OVMS_Protocol.pdf

That is the protocol that the current apps use. You can also reference the Apps source code:

https://github.com/openvehicles/Open-Vehicle-iOS

https://github.com/openvehicles/Open-Vehicle-Android

Regards, Mark.

closed 2018-07-05 by markwj

glynhudson, 2019-03-08:
@boazdejong did you manage to implement the V2 API? I would like to use the API for an OpenEVSE integration I'm working on to obtain SOC from OVMS. However, I'm having a tough time figuring out the V2 API. Is there a working example of a basic V2 API implementation?

boazdejong, 2019-03-08:
@glynhudson No sorry I completely dropped this.