openvehicles/Open-Vehicle-Monitoring-System-3 #139: Commands do not work for KIA SOUL EV

rwk1501, 2018-07-08:
I have the KIA SOUL EV ro of the 2014 oms firmware 3.1.008 and the android 3.12.4 application and I use the V 2.1.1 server. and a new type of OBD II cable. I found out that I do not even run the specific instructions for this vehicle that are listed in your manual: xks IGN1 <on / off> <pin>
Turn off or off IGN1-relay. Can be used to wake up part of the car.
xks IGN2 <on / off> <pin>
Turn off or off IGN2-relay. Can be used to wake up part of the car.
xks ACC <on / off> <pin>
Turn or off the ACC-relay. Can be used to wake up part of the car.xks START <on / off> <pin>
Turn-off or off START-relay. Can be used to wake up part of the car.
The vehicle does not respond to these commands at all. So I do not know where the problem might be. When I place these commands, the vehicle is locked and parked. Otherwise, when I lock or unlock the vehicle, but the charging port door prints an unrecognized command at the command line in the application. I hope you can get it done to make it work.

chrisli2018, 2019-02-21:
Can anyone solve this problem? If so, it will be like Christmas

rwk1501, 2019-03-13:
I wrote it here eight months ago that these commands do not work and still nobody has solved it to work as written in the manual. As far as support is concerned, it's pretty bad. I'm starting to regret buying this product.

chrisli2018, 2019-03-13:
Problem is that it is a open software, the hardware has nothing to do with the software. It's people/programmers who codes at their free time

rwk1501, 2019-03-13:
So in the eight months that I am dealing with, it would be good if one of them found time to solve or delete the instructions in the manual and not post them that work when the truth is that they don't work

rwk1501, 2019-06-03:
This problem with non-working commands on Kia Soul Ev has not yet been removed by firmware 3.2.002. and it has been written for some years in the manual that it should be usable. So I do not know if I ever live to see that it will work as it should as it is presented. The whole device is useful for getting information from the car to lock and unlock the car and open the charging door with the key and that's all it can do. When I do not think it will discharge in a couple of hours when the car is parked, the 12V battery.

chrisli2018, 2019-06-03:
Hope someone can read from the new e-soul, the UVO system and get some info from it. The UVO app will be released in Sweden in a couple of weeks