openvehicles/Open-Vehicle-Monitoring-System-3 #140: Compiling v3 fails

prensel, 2018-07-11:
Following the dev manual, succesfully able to compile the 'Hello World' example but compiling v3 fails:

> In file included from /home/paul/esp/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/components/console_ssh/src/console_ssh.cpp:39:0:
> /home/paul/esp/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/main/ovms_netmanager.h:47:22: fatal error: mongoose.h: No such file or directory
>

M4GNV5, 2018-07-11:
Did you initialize the git submodules?
```sh
git submodule update --init --recursive
```

prensel, 2018-07-11:
No didnt because the dev manual didnt told me so :-) 

> Git clone the OVMS v3 repository and init the build configuration:
> 
> $ git clone https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3.git
> $ cd Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3
> $ cp support/sdkconfig.default.hw31 sdkconfig
> $ make menuconfig
> 

So it should add this to it:

`git submodule update --init --recursive`

M4GNV5, 2018-07-11:
Yes, or add `--recursive` to git clone. I'll mention this on the mailing list.
Does compiling work for you after initializing submodules?

leres, 2018-07-11:
On 07/11/18 05:12, prensel wrote:
> No didnt because the dev manual didnt told me so :-)
>
>     Git clone the OVMS v3 repository and init the build configuration:
>
>     $ git clone [...]

I thought for sure that command was listed when I made some suggestions 
last month. I tried to look at the document history but apparently "view 
only" permissions can't do that. Anyway I think that line was lost 
accidentally.

Also, the "make menuconfig" isn't really needed, it seems to always 
happen anytime you run make on any target. (Although I would argue it 
shouldn't run when you "make distclean"...)

		Craig

prensel, 2018-07-11:
Compiling works now :-)
I have created a new 'vehicle' based on the Zeva BMS system (aftermarket BMS from zeva.com.au)).
It works very well and i have just installed it in my test-vehicle to do some real-life testing next days.

markwj, 2018-07-12:
Developer guide updated.

closed 2018-07-12 by markwj