openvehicles/Open-Vehicle-Monitoring-System-3 #146: Sleepy leaf returning 0% SOC

glynhudson, 2018-08-01:
Hi, I'm one of the developers at [OpenEnergyMonitor](http://openenergymonitor.org) project and a contributor to the [OpenEVSE WiFi gateway](http://github.com/openevse/ESP8266_WiFi_v2.x/). I'm keen to integrate the data with OpenEnergyMonitor Emoncms graphing platform and OpenEVSE.

I've recently obtained a OVMS V3 kit and I'm testing on my 24kWh Nissan LEAF and e-NV200. I'm *very impressed* with the OVMS project. The HW, FW and setup experience has been fantastic. Within minutes I had the unit hooked to the car and successfully logging to OVMS sever, Android app access and MQTT server via V3. Using Emoncms MQTT input getting the data logging was very easy. Nice work guys 👍 

I've got a few little issues / questions, sorry if these are very basic. I'm still getting to grips with the platform. 

*** 

After a while of being parking (30min+) the SOC starts returning 0%. This is presumably since the Leaf CAN bus has gone to sleep. Attempting to start a charge or turning on the A/C wakes up the car and the correct SOC gets returned again. Returning 0% causes issues when graphing the data in Emoncms or using the data to inform a smart charging algorithm. How would you recommend I handle this? Any integration would need to tell the differance between acutal 0% SOC or just the car sleeping! Obviously forcing the CAN bus to stay away 24/7 is not a good idea. 

- Could OVSM detect when the car has gone to sleep then return the last known SOC value? 
- Is there a command I could issue remotely which would wake up the car? The wake up button on the app doesn't work. 
- Is there a way to determine if the CAN bus has gone to sleep? 

Returning 0% SOC whenever the car is parked does not look good on Emoncms graphs! 

[image]

Or on the mobile app interface: 

[image]

carrott, 2018-08-01:
I'm glad to hear you're having such a positive experience. I wrote a python client for the v2 protocol and sent data to Grafana. Its great to hear the MQTT support allows easier mashups of the OVMS data.

The Leaf CAN bus goes to sleep very shortly after you turn the car off. The OVMS remembers the SoC until it gets updated data or it restarts. Your graph is strongly indicative of a crash resetting the remembered values.

Try plotting the monotonic metric which goes up by I think 1 every second. If that ever goes backwards you have had a restart probably caused by a crash.

My car is reasonably stable:

[image]

I'm running prototype hardware with a somewhat different configuration. Most notably I am not using the v3 server, and the code for this is very new. Try temporally turning off the v3 server support and seeing if it becomes more stable. You could also join the developer mailing list and request assistance debugging the actual cause.

The known issues document is wrong. Remote wake up for some pre-2013 models requires additional hardware. 2013-2015 is fully supported and 2016-2017 is more complicated, see the Leaf section of the user manual: https://docs.google.com/document/d/16JrXR7rybp-18DrEoeh1rg6GqQT_jVBvhaXh2oEWRHw/edit#heading=h.z0stoofvfzd3

I will update the known issues document shortly.

glynhudson, 2018-08-02:
Thanks, I think your right. Experiencing 0% SOC could be due to a crash. The SOC plot has been stable today and has not returned 0% even though the car has been parked up. 

>Try plotting the monotonic metric 

Good idea. Will do, What is the code for this? Is there a list of all MQTT topic codes somewhere or shall I try and sniff it out?

closed 2018-08-02 by glynhudson