openvehicles/Open-Vehicle-Monitoring-System-3 #147: Calibrate 12V reading?

glynhudson, 2018-08-01:
The 12V aux battery reading from a Nissan LEAF from OVMS is lower than the actual voltage. As measured using a voltmeter. This is probably no fault of OVMS, we're just reading what on the CAN bus right? Has anyone else noticed/experienced this? Is there anyway to apply a calibration factor?

M4GNV5, 2018-08-01:
The 12V voltage is measured using the incoming voltage that powers the OVMS. You can calibrate it using 
```
config set system.adc factor12v <factor>
```
the `<factor>` has to be calculated using: `oldFactor * (displayedVoltage / actualVoltage)`.
- `oldFactor` is the old value set. If you have not changed it yet it is `195.7`
- `displayedVoltage` is the Voltage as displayed by the OVMS.
- `actualVoltage` is the Voltage as measured bu hand using a voltmeter.

dexterbg, 2018-08-01:
I've added a section to the user guide:
https://docs.google.com/document/d/16JrXR7rybp-18DrEoeh1rg6GqQT_jVBvhaXh2oEWRHw/edit#heading=h.usa0sgzemy9l

closed 2018-08-01 by dexterbg

glynhudson, 2018-08-02:
Awesome! Thanks 👍

glynhudson, 2018-08-02:
Assuming the `config set` command is issued via the SHELL page on the module webpage?

cavram, 2018-08-02:
You can also issue commands via the app messages screen.

Try "ME   L" an abbreviation for metrics list. It's not case sensitive.

Also try ?
Or "me?" Com and completion.

Chris Avram

dexterbg, 2018-08-02:
…or you can send commands from my server based shell if you're on my server (https://dexters-web.de/), or you can send commands via ssh shell or the USB terminal. The server shell works like the app shell, so will work via modem as well as wifi. The builtin web and ssh servers are currently only available via wifi.

But please note: commands **are** case sensitive in v3 now (they haven't been in v2).

glynhudson, 2018-08-02:
Awesome, that's great to know that shell commands can be issued via the server remotely. 

I've just tried to issue to command via the local web server connected via station WiFi to the module. The command does not seem to have been applied. Should there be some sort of confirmation? 

[image]

dexterbg, 2018-08-02:
See above, the command is `config set system.adc factor12v <factor>`.

Also see `config ?`, `config list`, `config list ?` and `config list system.adc`. Configuration is segmented into "params" (categories) with "instances" (variables).

slcasner, 2018-08-02:
The "Usage" line _was_ the response from the command interpreter.  It
was telling you that the input did not fit the required command syntax
as shown.  This follows a convention from Unix-like systems.  It is
sufficient to remind a user who knows the command how to use it, but
obviously falls short of being a user guide for the command.

Michael explained this command in more detail.

                                                        -- Steve

glynhudson, 2018-08-02:
Doh! Sorry my mistake, that's much better 👍 

[image]