openvehicles/Open-Vehicle-Monitoring-System-3 #148: Switch off OVSM when 12V aux is critically low?

glynhudson, 2018-08-01:
Is there a way fo OVSM module to turn off / go to sleep when the 12V aux battery get's too low? I would be worried to leave OVMS connected to my EV (nissan LEAF) for a long period if the car is not in use. The Leaf doesn't have the ability to automatically top-up the 12V from the traction battery if it runs low. A flat 12V battery would result in being unable to startup the car. 

How much power does OVMS use? It seems the GSM, GPS and Wifi AP are all powered up 12/7 on the V3 module. Has anyone experienced any battery drain issues?

dexterbg, 2018-08-01:
The power requirements depend on the network, wifi uses less power than modem. Average current on the 12V line is around 80-120 mA, adding up to ~ 2-3 Ah per day.

The sleep modes are TODO and will reduce the power usage by some magnitude. For now, you get an alert notification when the 12V level gets too low (see user guide).

When leaving the car parked for longer periods, it's best to unplug the module if the car can't top up the 12V battery.

BrainBug0815, 2018-12-06:
In my case (Kia Soul EV), the battery was dead within 24 hours. 
Lucky me, it happened at home and I own a 12v car battery charger

Is there a reason to keep the device running after car is stopped and no charge is started, planned or even a cable plugged in?

mjuhanne, 2019-09-08:
Heads up for this feature. There's an implementation of Power Management module in my  [experimental powersave branch](https://github.com/mjuhanne/Open-Vehicle-Monitoring-System-3/commits/powersave).

- Switch off SIMCOM and WiFi after certain (configurable) time period of idling/non-charging of 12V battery has lapsed.
- Turn on previously switched off modules if 12V charging is initiated
- If 12V battery alert is received, shut down all the modules and OVMS if voltage hasn't been restored after certain configurable grace period (e.g. 30 minutes)
 - Web page to configure these features

I've made following measurements:
 - WiFi module (idle) : 25 mA
 - SIMCOM (idle) : 12 mA
 - CAN1 : 1.3 mA
 - CAN2/3 : neglible
 - SDCARD / ADC / SPI: all neglible
 - SWCAN/TH8056 transceiver: 4.4 mA
 - ESP32 core : 15 mA

Shutting down all of these just leaves 13.2 mA consumption. Finally found out the culprit: Plugged in GPS antenna consumes 11 mA even when SIMCOM is powered off ??!! After removing the antenna the consumption is only 0.8 mA. When SWCAN module was not attached, there was residual 2 mA consumption, but this may be caused by some pullup/pulldown resistor issue (haven't looked into this yet). 

Anyhow, with this feature it is much safer to leave the car parked and unattended for longer periods in those places where charging might not be feasible (airport parking for example). This is now similar to Volt/Ampera Onstar module, which has a shutdown delay after 5 days of inactivity if I recall correctly.

mjuhanne, 2019-09-08:
TODO: Recover from shutdown by using ESP32 sleep mode and wake up periodically (or maybe via interrupt caused by CAN traffic) to measure the 12V voltage.

dexterbg, 2019-09-08:
The GPS antenna is an active one. We first tried passive antennas, but they had very bad signal levels. I don't think the antenna power is switchable.

dexterbg, 2019-10-19:
Merged in commit 1cbdf4c4d964b0cb6031347303ccccc77f52b55d

The todo is still to do. Recovering from a complete shutdown currently needs plugging out the module, if the todo needs more time we should mention that on the config page.

slcasner, 2019-10-19:
Or pushing the reset button?  I cut a hole in the top of the case and installed the plastic button and rod that was the reset button in the GFCI plug of a hair dryer.  This pushes on the reset switch on the OVMS PCB.  It has been handy to recover from the periodic system hangups more easily than interrupting power or hooking up a laptop.  I hope this is needed less often since the PSRAM fix.  I think it would be a worthwhile to add a reset button like this to the OVMS design.

Dimitrie78, 2019-10-20:
I made a branch to test Wake up when 12V is higher then 12v ref Volatage. [powersave-test](https://github.com/Dimitrie78/Open-Vehicle-Monitoring-System-3/commits/powersave-test) .
This needs the ULP coprocessor programming toolchain:
https://docs.espressif.com/projects/esp-idf/en/release-v3.1/api-guides/ulp.html

Also i have remove the automatic set 12v ref Volatage,
to prevent automatic set wrong 12v ref values.
Example: 12v Battery is almost gone and have ~11,8V. When now charging starts for like 5 min. The Function will set 12v ref to ~ 11,9V.
With a 12v alert threshold from 1,6v you will get an alarm at 10,3V. This wheres to late.

dexterbg, 2019-10-20:
Please do not simply remove existing functionality. If you need to disable something, add a config for that.
Note there is a config reserved to control the initial 12v ref. If you add a config to disable setting the ref on charge stop, you can use the 12v.ref config to set a fixed ref level.

Dimitrie78, 2019-10-21:
added config to disable 12v ref

jheredianet, 2020-05-23:
> In my case (Kia Soul EV), the battery was dead within 24 hours.
> Lucky me, it happened at home and I own a 12v car battery charger
> 
> Is there a reason to keep the device running after car is stopped and no charge is started, planned or even a cable plugged in?

I'm experimenting the same issue with a Hyundai Kona.

jheredianet, 2020-07-14:
Hi guys, after trying a lot of suggestions you made. What I do with my KonaEV is to power off the CAN bus, then I start it manually when needed. These are the commands I use:
power can1 off

power can1 on

Controlling this manually, the aux battery does not drain a lot, it could last days without problems.
So I guess it's something with that module that need to be fixed.

RowdyDoc, 2021-07-25:
I am not sure if it has been adequately demonstrated (ie conclusive) - but there seems to be a pretty strong suggestion that aux battery drain problems in the Kia e-Niro and Hyundai Kona EVs may be due to the OVMS module keeping other systems in the car in a higher power consumption mode - maybe somehow due to the CAN bus of the OVMS being active?
So in addition to the overall shutdown of the OVMS module on low aux battery volts, wouldn't it be a good option feature to:
1. "power can1 off" on the condition of: (!charging & car-off & doors-closed) for some delay time (mins)
2. "power can1 on" on the condition of: (aux-batt-volts>aux-ref-volts) for some delay time (secs) [this should work because aux batt volts is directly measured from the OVMS module supply voltage?]
3. "power can1 on" periodically to do a check of the system (mins) [would stay on until condition 1]
Obviously this isn't perfect as the OVMS module would not be able to notify vehicle events while the CAN monitoring is off, however for users more concerned about impact on the aux battery than continuous monitoring while the car is effectively asleep it may be a good option.

Alternatively, is it possible that just putting the OVMS module CAN into a purely passive (listen only) mode (instead of fully off) would allow detection of a reduced set of status without the risk of keeping other car systems awake?

dexterbg, 2021-07-25:
You can implement that easily as a plugin. As a general change, that would be a rather nasty hack, and would only be needed for the e-Niro/Kona adaption.

The far better option is to determine which part of the e-Niro/Kona polling causes the ECUs to wake up or stay awake, and to add some logic to only poll extended data when you know the vehicle is actually online for driving or charging.

The VW e-Up ECUs also would be kept awake if we wouldn't determine the vehicle status from a single PID on an ECU that doesn't trigger wakeup / inhibit sleep if polled.

Regards,
Michael

RowdyDoc, 2021-07-26:
ok - will start digesting the scripting info in the docs and have a crack.
Thanks for taking the time to provide the feedback.

dexterbg, 2021-12-30:
> The todo is still to do. Recovering from a complete shutdown currently needs plugging out the module, if the todo needs more time we should mention that on the config page.

I think I've fixed that todo in commit d8028d5d3fa1792823ac349211d0fb82055c1ec0, so closing this.

If there was another open point feel free to open a new specific issue.

closed 2021-12-30 by dexterbg