openvehicles/Open-Vehicle-Monitoring-System-3 #149: Nissan LEAF charging current incorrect

glynhudson, 2018-08-01:
OVMS seems to report a charging current that's lower than the actual charging current. 

e.g. 

- With the pilot set to 13A OVSM reports 7A actual.
- With the pilot set to 32A the OVSM reports a pilot of 30A setting and 17A acutal. 

Doing some more testing adjusting the charge pilot (CP) in real-time using an openevse I observe that at low CP OVSM repots the pilot setting correclty below 13A then the OVMS pilot is always slightly higer until 30A when it then reports a bit lower above 30A. The CP reading is not a big deal, it's close enough! 

However, the OVSM real-time current is always much lower than the actual current, as measued by openevse. 

CP  OVSMCP   OVMS real-time
6A     6A             6A
10A   10A           6A
12A   12A           7A
13A   14A           7A
14A   15A           8A
15A   16A          8A
20A    21A         10A
27      28A         14A
30A    30A         16A
32A    30A         16A

The pilot setting 

[image]

[image]

carrott, 2018-08-01:
As far as I can know, the Leaf does not expose the grid voltage and current on the CAN bus. As we don't have the grid side data, we report battery voltage and current. This is different from the other car modules and should really be handled on the client (by displaying either grid or battery depending on what is available) rather than faking the grid side data.

Looking at your two screenshots and guessing you're on approximately a 240V grid:

240V * 13A / 381V = 8A
240V * 30A / 367V = 19A

The first example is close enough to 7A, while the second example is a bit further away. On my Leafs with 3.3kW (ie 16A) chargers I see the reported control pilot shows no more than 18A, even though the EVSE control pilot is much higher. It's possible the 6kW charger cannot draw the full 30A from the grid.

If you know of a way to get the grid side charger data I'd be pleased to update the firmware.

glynhudson, 2018-08-02:
Ok, that makes sense. Yes, you're correct the leaf doesn't see to report the grid voltage on the CAN bus. How about displaying the charging power i.e multiply the battery voltage by the charging current? This is what leafspy does, this works well since it's the power most EV drivers are familiar with rather than current and voltage and power remained constant as the voltage changes. 

I'm happy if you want to close this issue since the OVMS readings are not technically incorrect. But could possibly be better displayed. I'll see if I can get involved and add charging power display to the Android app.

closed 2018-08-02 by glynhudson