openvehicles/Open-Vehicle-Monitoring-System-3 #158: Modem loosing connection (NetLoss) after 3.1.010 update

glynhudson, 2018-09-18:
After updating to `V3.1.010` I've noticed that the modem seems to loose connection every few days. Stopping and starting the modem via the web interface doesn't seem to make it re-connect. The only method I've found to re-connect is to physically power cycle the whole OVMS unit. 

The location I'm in has got excellent GSM signal and I've not heard of any local issues with the GSM network. Previously on older FW versions the connection was solid 

When disconnected the modem status shows:
 
```
Network Registration: Searching
Provider: O2 - UK Hologram
Signal: -89 dBm
State: NetLoss
PPP: Not connected
Last Error: User Interrupt
GPS: Connected on channel: #1
```

I've got logging set to WARNING level and the log is full of 100's of lines queue overflow, I'm not sure if this has got anything to do with NetLoss issue:

````
[0;31mE (108709612) events: SignalEvent: queue overflow, event 'ticker.1' dropped[0m
[0;31mE (108710612) events: SignalEvent: queue overflow, event 'ticker.1' dropped[0m
[0;31mE (108710612) events: SignalEvent: queue overflow, event 'ticker.10' dropped[0m
[0;31mE (108711612) events: SignalEvent: queue overflow, event 'ticker.1' dropped[0m
[0;31mE (108712612) events: SignalEvent: queue overflow, event 'ticker.1' dropped[0m
[0;31mE (108713612) events: SignalEvent: queue overflow, event 'ticker.1' dropped[0m
[0;31mE (108714612) events: SignalEvent: queue overflow, event 'ticker.1' dropped[0m
[0;31mE (108715612) events: SignalEvent: queue overflow, event 'ticker.1' dropped[0m
[0;31mE (108716612) events: SignalEvent: queue overflow, event 'ticker.1' dropped[0m
[0;31mE (108717612) events: SignalEvent: queue overflow, event 'ticker.1' dropped[0m
[0;31mE (108718612) events: SignalEvent: queue overflow, event 'ticker.1' dropped[0m
[0;31mE (108719612) events: SignalEvent: queue overflow, event 'ticker.1' dropped[0m
[0;31mE (108720612) events: SignalEvent: queue overflow, event 'ticker.1' dropped[0m
[0;31mE (108720612) events: SignalEvent: queue overflow, event 'ticker.10' dropped[0m
[0;31mE (108720622) events: SignalEvent: queue overflow, event 'ticker.60' dropped[0m
[0;31mE (108721612) events: SignalEvent: queue overflow, event 'ticker.1' dropped[0m
[0;31mE (108722612) events: SignalEvent: queue overflow, event 'ticker.1' dropped[0m
[0;31mE (108723612) events: SignalEvent: queue overflow, event 'ticker.1' dropped[0m
[0;31mE (108724612) events: SignalEvent: queue overflow, event 'ticker.1' dropped[0m
[0;31mE (108725612) events: SignalEvent: queue overflow, event 'ticker.1' dropped[0m
[0;31mE (108726612) events: SignalEvent: queue overflow, event 'ticker.1' dropped[0m
[0;31mE (108727612) events: SignalEvent: queue overflow, event 'ticker.1' dropped[0m
[0;31mE (108728612) events: SignalEvent: queue overflow, event 'ticker.1' dropped[0m
[0;31mE (108729612) events: SignalEvent: queue overflow, event 'ticker.1' dropped[0m
[0;31mE (108730612) events: SignalEvent: queue overflow, event 'ticker.1' dropped[0m
[0;31mE (108730612) events: SignalEvent: queue overflow, event 'ticker.10' dropped[0m
[0;31mE (108731612) events: SignalEvent: queue overflow, event 'ticker.1' dropped[0m
[0;31mE (108732612) events: SignalEvent: queue overflow, event 'ticker.1' dropped[0m
````

glynhudson, 2018-10-15:
Is anyone else experiencing this issue? Or is it just me?

The unit was rock solid before this update. I'm still experiencing the lock ups every few days requiring a hard reset.

markwj, 2018-11-26:
I'm not seeing it, or had this issue reported elsewhere.

Can you try the latest firmware:

http://api.openvehicles.com/firmware/ota/v3.1/main/3.1.011.ovms3.bin

or the one before that:

http://api.openvehicles.com/firmware/ota/v3.1/main/3.1.008.ovms3.bin

Make sure that it is the new firmware, and not something else?

Please also provide the output of:
  config list auto
  module tasks

so we can see what you have running.

glynhudson, 2018-11-26:
I'm currently running `3.1.011`, it updated automatically via OTA. I'm afraid the issue still seems to persist. The disconnect seems to happen every few weeks. One thing I didn't mention in my first post ist that I have the V3 server enabled. 

```
OVMS# config list auto
auto (readable writeable)
  ext12v: no
  init: yes
  modem: yes
  obd2ecu: 
  ota: yes
  server.v2: yes
  server.v3: yes
  vehicle.type: NL
  wifi.mode: apclient
  wifi.ssid.ap: env200
  wifi.ssid.client: XXXXXX
```

***

```
OVMS# module tasks
Number of Tasks = 17      Stack:  Now   Max Total    Heap 32-bit SPIRAM C# PRI
3FFAFB64  1 Blk esp_timer         400   688  4096   22480    644  22488  0  22
3FFBD9B8  2 Blk eventTask         436   868  4608       0      0      0  0  20
3FFBFCF8  3 Blk OVMS Events       436  3508  8192   70652      0  22592  1   5
3FFC17CC  4 Blk OVMS CanRx        424   872  2048    2364      0  23220  0  23
3FFC221C  5 Blk ipc0              392   488  1024    7776      0      0  0  24
3FFC281C  6 Blk ipc1              392   488  1024      12      0      0  1  24
3FFC4644  9 Rdy IDLE              384   560  1024       0      0      0  0   0
3FFC4BD8 10 Rdy IDLE              372   548  1024       0      0      0  1   0
3FFC596C 11 Blk Tmr Svc           344   984  3072      88      0      0  0   1
3FFC2AF4 16 Blk tiT               492  2556  3072    1328      0   8444  *  18
3FFCB374 17 Blk OVMS SIMCOM       672  1872  4096       0      0   3912  1   5
3FFCCB34 18 Blk wifi              464  2096  3584   10768      0   5720  0  22
3FFD1774 19 Blk OVMS Vehicle      464  1104  6144       0      0     32  1  10
3FFD4C08 20 Blk OVMS Console      548  1780  6144       0      0     20  1   5
3FFD6D9C 21 Blk OVMS NetMan       856  4616  8192    7064      0   5920  1  22
3FFD8350 22 Blk mdns              460  2188  4096     104      0      8  0   1
3FFDD79C 47 Rdy module tasks      296  1944  5120       0  15488  12080  1   4
```

I will now flash `3.1.008` report back if the issue persisits or not. Unfortnalty it will take a number of weeks to find out!

glynhudson, 2018-12-10:
Ok, I'm still experiencing the issue with `3.1.008` therefore the issue must not relate to `3.1.010`. I'll revert back to running latest version. It only happens about once per month so not a big issue. Maybe it's something do with my hadware :-/ 

I'll close this issue for the moment.

closed 2018-12-10 by glynhudson