openvehicles/Open-Vehicle-Monitoring-System-3 #167: send homelink command outside of app

imperialjoy, 2018-11-08:
I'm trying to send the homelink 1 and homelink 2 command to my Leaf.  I can do this manually from the app just fine, but can't seem to do this from the http api or MQTT.  I'd like to be able to send these commands via my home automation system so the car is preconditioned in the morning.

Sending homelink commands from the http api returns 'not yet implemented', so I've moved on to MQTT which seems to be the new way.

 I'm not sure on the syntax for MQTT, which seems to require a 'ClientID'.  Is that my login for the v2 server?  Where can I find the ClientID?

I'm basing the commands on: http://lists.openvehicles.com/pipermail/ovmsdev/2018-July/005297.html
since I can't find any docs or examples.

slcasner, 2018-11-08:
On Thu, 8 Nov 2018, imperialjoy wrote:

> I'm trying to send the homelink 1 and homelink 2 command to my Leaf.
> I can do this manually from the app just fine, but can't seem to do
> this from the http api or MQTT.  I'd like to be able to send these
> commands via my home automation system so the car is preconditioned
> in the morning.

This is curious to me.  What I know of homelink commands is that they
are sent *from* the car to control things like garage door openers.

                                                        -- Steve

imperialjoy, 2018-11-09:
On the LEAF, homelink 1 is hardcoded to start climate control, and homelink 2 is coded to stop it.  Not sure if you can configure homelink 3 to do something or not.  It would be great to send a command whenever the car got home, announcing its arrival but I'll leave that for another day.

I figured MQTT commands out, and its working.  The syntax for me is:
mosquitto_pub -h macmini.local -t ovms//vehicleID/client/clientID/command/homelink -m "homelink 1"

Where clientID is your website login (server login). 

It works very fast, but I haven't had a chance to check if the MQTT commands work over cellular yet. I can confirm it works when everything is on WiFi.

carrott, 2018-11-10:
The Leaf's 'Homelink is Remote Climate Control' feature is a temporary hack to allow control of the climate control by clients which don't support the climate control command (Android now supports it natively but you may not be able to access it because the necessary capability isn't being sent, I'm not sure about iOS).

You should be able to directly invoke the remote climate control commands the same way you use the homelink hack, but it turns out that isn't implemented. I've just submitted a pull request to implement it. The command is called climatecontrol and accepts an argument of on or off, see PR 168

glynhudson, 2018-11-13:
You mention the OVMS Android app supports the climate control command, it seems currently this command is not accessible and still needs to be invoked by the Homelink function. Could this now be fixed after your PR?

https://github.com/openvehicles/Open-Vehicle-Android/issues/59

imperialjoy, 2018-12-03:
I was able to get this working via MQTT.  I browsed the source as this wasn't documented but the command line for me is:  /usr/local/bin/mosquitto_pub -h macmini.local -t ovms//MYVEHICLEID/client/MYUSERNAME/command/homelink -m "homelink 1" 

Works great over cellular too :)

dexterbg, 2021-12-30:
If I read this issue correctly, the only thing missing here is command invocation via the HTTP API, which isn't a module but a server issue. So I'll close this.

closed 2021-12-30 by dexterbg