openvehicles/Open-Vehicle-Monitoring-System-3 #172: Module answers the phone when called

ltjessem, 2018-11-15:
This is a very fringe scenario but I have a twin sim (ie. when my main sim rings, that one rings too) that I'd like to use in my OVMS, but when somebody calls me the OVMS will answer the phone and just leave the line silent, and therefore not letting me receive call on my phone. Is it possible to disable the behavior?

markwj, 2018-11-23:
Very strange. From what I can see, the auto-answer behaviour is controlled by S0. The factory default for that is ATS0=0, which means don't auto-answer.

I only have data plans on my SIMs here, so hard to test this.

Can you connect to your module as a terminal (either USB cable, or ssh), and:

  log level debug

Then call the number, and paste the output here. In particular, I am interesting in seeing what the simcom shows as rx/tx.

ltjessem, 2018-12-05:
Finally got around to this, here's the output when it answers:

````
D (251627) simcom: rx line ch=3 len=10  : +CREG: 1,1
D (251627) simcom: rx line ch=3 len=29  : +CCLK: "18/12/05,08:51:13+04"
D (251627) simcom: rx line ch=3 len=11  : +CSQ: 15,99
D (251637) simcom: rx line ch=3 len=22  : +COPS: 0,0,"TELENOR",2
D (251637) simcom: rx line ch=3 len=2   : OK
D (256697) simcom: rx line ch=3 len=4   : RING
D (256697) simcom: rx line ch=4 len=4   : RING
D (256707) simcom: rx line ch=3 len=26  : +CLIP: "5*******",129,,,,0
D (256707) simcom: rx line ch=4 len=26  : +CLIP: "5*******",129,,,,0
D (262917) simcom: rx line ch=3 len=17  : VOICE CALL: BEGIN
D (262917) simcom: rx line ch=4 len=17  : VOICE CALL: BEGIN
D (271457) simcom: rx line ch=3 len=23  : VOICE CALL: END: 000008
D (271457) simcom: rx line ch=4 len=23  : VOICE CALL: END: 000008
D (271467) simcom: rx line ch=3 len=10  : NO CARRIER
D (271467) simcom: rx line ch=4 len=10  : NO CARRIER
D (281587) simcom: rx line ch=3 len=10  : +CREG: 1,1
D (281587) simcom: rx line ch=3 len=29  : +CCLK: "18/12/05,08:51:43+04"
D (281587) simcom: rx line ch=3 len=11  : +CSQ: 15,99
D (281597) simcom: rx line ch=3 len=22  : +COPS: 0,0,"TELENOR",2
D (281597) simcom: rx line ch=3 len=2   : OK
````

ltjessem, 2018-12-05:
Finally got around to this, 

Here's the output when it initializes: 
````
D (21567) events: Signal(system.modem.poweringon)
D (21567) simcom: tx scmd ch=0 len=4   : AT||
D (22567) simcom: tx scmd ch=0 len=4   : AT||
D (23567) simcom: tx scmd ch=0 len=4   : AT||
D (24567) simcom: tx scmd ch=0 len=4   : AT||
D (25567) simcom: tx scmd ch=0 len=4   : AT||
D (26567) simcom: tx scmd ch=0 len=4   : AT||
I (27377) simcom: State: Enter PoweredOn state
D (27377) events: Signal(system.modem.poweredon)
D (30367) simcom: rx line ch=0 len=12  : +CPIN: READY
D (30367) simcom: rx line ch=0 len=12  : OPL UPDATING
D (30367) simcom: rx line ch=0 len=12  : PNN UPDATING
D (30877) simcom: rx line ch=0 len=8   : SMS DONE
D (31937) simcom: rx line ch=0 len=10  : CALL READY
D (31947) simcom: rx line ch=0 len=7   : PB DONE
D (36567) simcom: tx scmd ch=0 len=103 : AT+CPIN?;+CREG=1;+CTZU=1;+CTZR=1;+CLIP=1;+CMGF=1;+CNMI=1,2,0,0,0;+CSDH=1;+CMEE=2;+CSQ;+AUTOCSQ=1,1;E0||
D (36627) simcom: rx line ch=0 len=101 : AT+CPIN?;+CREG=1;+CTZU=1;+CTZR=1;+CLIP=1;+CMGF=1;+CNMI=1,2,0,0,0;+CSDH=1;+CMEE=2;+CSQ;+AUTOCSQ=1,1;E0
D (36657) simcom: rx line ch=0 len=12  : +CPIN: READY
D (36657) simcom: rx line ch=0 len=11  : +CSQ: 14,99
D (36657) simcom: rx line ch=0 len=2   : OK
D (38567) simcom: tx scmd ch=0 len=16  : AT+CGMR;+ICCID||
D (38577) simcom: rx line ch=0 len=23  : +CGMR: 35316B11SIM5360E
D (38577) simcom: rx line ch=0 len=28  : +ICCID: 8*****************15
D (38577) simcom: rx line ch=0 len=2   : OK
D (41567) simcom: tx scmd ch=0 len=8   : AT+COPS?
D (42567) simcom: tx scmd ch=0 len=20  : AT+CMUXSRVPORT=3,1||
D (42577) simcom: rx line ch=0 len=5   : ERROR
D (43567) simcom: tx scmd ch=0 len=20  : AT+CMUXSRVPORT=2,1||
D (43607) simcom: rx line ch=0 len=2   : OK
D (44567) simcom: tx scmd ch=0 len=20  : AT+CMUXSRVPORT=1,1||
D (44607) simcom: rx line ch=0 len=2   : OK
D (45567) simcom: tx scmd ch=0 len=20  : AT+CMUXSRVPORT=0,5||
D (45587) simcom: rx line ch=0 len=2   : OK
D (46567) simcom: tx scmd ch=0 len=11  : AT+CMUX=0||
D (46587) simcom: rx line ch=0 len=2   : OK
````

Here's the output when it answers a call after two rings:
````
D (251627) simcom: rx line ch=3 len=10  : +CREG: 1,1
D (251627) simcom: rx line ch=3 len=29  : +CCLK: "18/12/05,08:51:13+04"
D (251627) simcom: rx line ch=3 len=11  : +CSQ: 15,99
D (251637) simcom: rx line ch=3 len=22  : +COPS: 0,0,"TELENOR",2
D (251637) simcom: rx line ch=3 len=2   : OK
D (256697) simcom: rx line ch=3 len=4   : RING
D (256697) simcom: rx line ch=4 len=4   : RING
D (256707) simcom: rx line ch=3 len=26  : +CLIP: "5*******",129,,,,0
D (256707) simcom: rx line ch=4 len=26  : +CLIP: "5*******",129,,,,0
D (262917) simcom: rx line ch=3 len=17  : VOICE CALL: BEGIN
D (262917) simcom: rx line ch=4 len=17  : VOICE CALL: BEGIN
D (271457) simcom: rx line ch=3 len=23  : VOICE CALL: END: 000008
D (271457) simcom: rx line ch=4 len=23  : VOICE CALL: END: 000008
D (271467) simcom: rx line ch=3 len=10  : NO CARRIER
D (271467) simcom: rx line ch=4 len=10  : NO CARRIER
D (281587) simcom: rx line ch=3 len=10  : +CREG: 1,1
D (281587) simcom: rx line ch=3 len=29  : +CCLK: "18/12/05,08:51:43+04"
D (281587) simcom: rx line ch=3 len=11  : +CSQ: 15,99
D (281597) simcom: rx line ch=3 len=22  : +COPS: 0,0,"TELENOR",2
D (281597) simcom: rx line ch=3 len=2   : OK
````

dexterbg, 2018-12-22:
Please try if
```simcom cmd "ATS0=0"```
disables the auto answering or if e.g.
```simcom cmd "ATS0=10"```
causes the modem to answer after 10 rings.

grnbrg, 2019-01-25:
Just a note that as bugs go, this one has a nasty external side effect.  It could have cost me over $300.

I picked up a SIM for the module, and had it assigned on a data only plan on my phone account, and more or less forgot about it.  It wasn't until a couple of months later that I bothered to check my bill -- the account just auto-debits, and since the amount never really changes, there's no need.  :)  It turns out that some one or some thing was calling the number that had been assigned to the new SIM as many as 10 times or more per day, and since the SIM was on a data only plan, voice "calls" were billed at $1 per minute or portion.  A couple of hundred incoming calls logged, at $1 each.  Oops.  

Fortunately my service provider was reasonable, and believed me when I insisted that the device was incapable of voice traffic.  They reversed the charges and added a call block to the number.  But it could have been worse.

markwj, 2020-02-02:
Simple fix is to add S0=0 to the modem init string. Implemented in e6ac0c75.

closed 2020-02-02 by markwj