openvehicles/Open-Vehicle-Monitoring-System-3 #174: i_MIEV 3.1.011 issues

gravtix, 2018-11-20:
This version was a big improvement to OVMS functionality on the i-MIEV but I've noticed some odd things that didn't exist in the previous version.

1) Charger and PEM show -40C temperature.  Occasionally I will see a correct temperature for a minute or two before it reverts to -40C.  Battery and motor temperature are fine.  This is in the app or running "metrics list"
2) Ideal range is always 0km.  I've seen it show a non-zero value once since this version was posted.
3) Using a L1 charger I seem to see 1 amp less than the actual value.  on a 120V/8A charger I see 120V/7A and 120V/12A I see 120V/11A.  This might just be a quirk with the car although I see 208V/14A on any public level2 charger which is correct and matches the output as reported by the charger.

I've just finished setting up the dev environment.  Still learning the code and the documentation but I hope to be able to contribute in the future.

KommyKT, 2018-11-20:
I have a own version of i-Miev code. Only tested on Peugeot iOn 2012 (88 cell), if you would like to test my code, you can update the software OTA. This compiled today with all updates newer then 3.1.011. This show all cell temps and volts on web interface, charging status, quick charge (not tested, no quick charger in my region). If you want to test my code, go to web interface on OVMS, Config/firmware, on firmware page select Flash from web tab and add this webpage address: http://kommykt-nas.ddns.net:  22289/ovms3.bin   I add two whitespace before 22289 the correct is ....net:22289.... If you test it please send me a note of working /not working

gravtix, 2018-11-22:
Thanks!
I have a 2012 North American i-MIEV with 88 cells.

I managed to install your firmware, the download kept timing out but it seems my WiFI is too weak where the car is parked, it worked before.

The firmware works! Some feedback:

1) Might want to turn BMS notifications off by default.  The app absolutely spammed my phone until I signed into to the web interface and turned them all off.
2) I am seeing the temperature for the charger and PEM correctly now.
3) Ideal range is working correctly now.  I think it assumes an ideal value of 150km which is extremely optimistic :)
4) CAC shows 2.0Ah.  I am not sure if this is the same variable as Canion but my value there is much higher.
5) The Level1 charging speeds seems to be fixed.  If I run "stat" I see 11.2A current but the app only shows 11.0A.  So the app is either rounding down of truncating the decimal as the values mismatch
6) I am seeing all 88 cells in the BMS webpage as well as the 66 temperature sensors.  The car is charging now so I can see the animation.
7) I will test DC charging as soon as I can.  There is many of them in my area.

If you'd like me to test anything, please let me know, I will keep testing.

gravtix, 2018-11-22:
metrics list v.b.c

v.b.c.temp                               2;2;3;4;4;4;3;4;2;3;2;2;2;3;3;3;3;4;3;3;1;3;5;2;1;0;3;2;3;2;3;2;2;1;1;1;2;2;3;2;3;2;3;1;2;0;0;0;2;1;2;1;1;1;1;1;1;1;2;1;2;2;2;2;1;1°C
v.b.c.temp.alert                         2;1;2;1;1;1;2;1;2;2;1;1;2;1;1;1;1;1;1;1;2;1;1;1;1;1;2;2;2;2;2;2;2;2;1;2;0;2;0;2;1;2;1;2;1;1;1;1;2;2;2;2;2;2;2;2;2;2;2;2;2;2;2;2;2;2
v.b.c.temp.dev.max                       -3.11;2.18;4.89;4.89;4.89;4.89;4.89;4.89;4.89;4.89;-2.11;-2.11;-3.11;2.24;-2.11;2.41;-2.11;2.53;-2.11;3.32;-3.11;2.47;4.73;-2.11;-2.11;-4.11;4.89;4.89;4.89;4.89;4.89;4.89;4.89;4.89;-2.11;-4.11;1.32;-3.11;1.5;-3.11;2.38;-3.11;2.56;-4.11;2.29;-4.11;-4.11;-5.11;4.89;4.89;4.89;4.89;4.89;4.89;-4.11;-4.11;-4.11;-4.11;-3.11;-3.11;-3.11;-3.11;-3.11;-3.11;-4.11;-3.11°C
v.b.c.temp.max                           2;5;10;10;10;10;10;10;10;10;3;3;2;5;3;5;3;5;3;6;2;5;7;3;4;1;10;10;10;10;10;10;10;10;3;1;4;2;4;2;5;2;5;2;5;2;1;1;10;10;10;10;10;10;1;1;1;1;2;2;2;2;2;2;1;2°C
v.b.c.temp.min                           1;2;3;3;3;4;3;3;2;2;2;1;2;2;2;3;3;3;2;3;1;2;5;1;1;0;3;2;3;1;3;1;2;1;1;1;2;1;2;2;3;2;3;1;2;0;0;0;1;1;1;1;1;1;0;1;1;1;1;1;2;2;2;2;1;1°C
v.b.c.voltage                            3.925;3.925;3.93;3.93;3.93;3.93;3.93;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.93;3.925;3.93;3.925;3.93;3.925;3.925;3.93;3.925;3.925;3.925;3.93;3.93;3.93;3.93;3.925;3.93;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.93;3.925;3.93;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.93;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.93;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925V
v.b.c.voltage.alert                      0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0
v.b.c.voltage.dev.max                    -0.00477;-0.00489;0.00545;0.00545;0.00545;0.00545;0.00545;0.00455;-0.0075;-0.0075;-0.00455;0.00489;0.00483;-0.00472;-0.00489;0.00545;-0.00483;0.00545;-0.00955;-0.00955;0.00443;-0.00455;0.00494;-0.00483;-0.00466;0.00466;0.00545;0.00545;0.00545;0.00545;-0.0075;0.00489;-0.0075;-0.0075;0.00472;-0.005;0.00545;-0.00489;0.00545;-0.005;0.00545;0.00545;-0.00483;-0.00955;-0.00477;-0.00483;-0.00489;-0.00483;-0.00489;-0.00489;0.00511;0.00466;0.00545;0.00494;-0.0075;-0.0075;-0.0075;-0.0075;-0.00494;-0.00466;0.00466;-0.00477;-0.00955;-0.00955;-0.00477;0.00483;-0.00455;-0.00489;-0.00477;0.00466;0.00489;-0.005;0.005;-0.0075;-0.0075;-0.00494;-0.0075;-0.0075;-0.005;-0.00523;-0.00472;-0.00511;-0.00955;-0.01455;-0.00477;0.00472;-0.00489;-0.00489V
v.b.c.voltage.max                        3.95;3.95;3.95;3.95;3.95;3.95;3.955;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.955;3.95;3.955;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95;3.95V
v.b.c.voltage.min                        3.92;3.92;3.925;3.925;3.925;3.925;3.925;3.92;3.92;3.92;3.92;3.92;3.92;3.92;3.92;3.92;3.92;3.925;3.915;3.915;3.92;3.92;3.92;3.92;3.92;3.92;3.92;3.92;3.925;3.925;3.925;3.92;3.92;3.92;3.92;3.92;3.92;3.92;3.92;3.92;3.925;3.92;3.92;3.915;3.92;3.92;3.92;3.92;3.92;3.92;3.92;3.92;3.92;3.92;3.92;3.92;3.92;3.92;3.92;3.92;3.92;3.92;3.915;3.915;3.92;3.92;3.92;3.92;3.92;3.92;3.92;3.92;3.92;3.92;3.92;3.92;3.92;3.92;3.92;3.92;3.92;3.92;3.915;3.91;3.92;3.92;3.92;3.92V
v.b.cac                                  2Ah
v.b.consumption                          0Wh/km
v.b.coulomb.recd
v.b.coulomb.used
v.b.current                              -0.64A

metrics list v.b.p

v.b.p.level.avg
v.b.p.level.max
v.b.p.level.min
v.b.p.level.stddev
v.b.p.temp.avg                           2.08°C
v.b.p.temp.max                           5°C
v.b.p.temp.min                           1°C
v.b.p.temp.stddev                        1.11°C
v.b.p.temp.stddev.max                    3.68°C
v.b.p.voltage.avg                        3.92608V
v.b.p.voltage.max                        3.93V
v.b.p.voltage.min                        3.925V
v.b.p.voltage.stddev                     0.00201V
v.b.p.voltage.stddev.max                 0.00409V
v.b.power                                0.217665kW

dexterbg, 2018-11-23:
@KommyKT: looks like there is something wrong with the temperature readings. Distribution and deviations are very unlikely real. Could those 10°C max blocks be some sensor init/exit data?

If these readings are real/normal on i-MiEVs, you need to set much larger default thresholds.

gravtix, 2018-11-23:
Here's some data after charging for 8+ hours overnight:
v.b.p.level.avg
v.b.p.level.max
v.b.p.level.min
v.b.p.level.stddev
v.b.p.temp.avg                           5.59°C
v.b.p.temp.max                           10°C
v.b.p.temp.min                           2°C
v.b.p.temp.stddev                        2.17°C
v.b.p.temp.stddev.max                    4.05°C
v.b.p.voltage.avg                        4.0983V
v.b.p.voltage.max                        4.1V
v.b.p.voltage.min                        4.095V
v.b.p.voltage.stddev                     0.00245V
v.b.p.voltage.stddev.max                 0.03791V
v.b.power                                -0.82961kW

Ive seen big gaps before especially after charging. Temp sensor #64 is pretty notorious for high readings as it's right next to the charger as I recall.

I've never had a reliable means of monitoring the battery temperature so I'll collect some data after DC charging.  I had min/max temps of 42C and 50C this August using DC charging.

KommyKT, 2018-11-25:
@dexterbg all data same as caniOn reading, only the temp positions not same. All temps and volts get from 4 frames 6e1, ..., 6e4. 
6e1 has 24 volt/temps, 
6e2 has 22 temp, 24 volt, 
6e3 has 20 temp,20 volt
6e4 has 0 temp, 20 volt

gravtix, 2018-12-05:
@KommyKT

So I tried DC fast charging and it didn't work.  The car didn't even detect it was charging, unlike the OTA build.

metrics list

m.freeram                                4212592
m.hardware                               OVMS WIFI BLE BT cores=2 rev=ESP32/1
m.monotonic                              248752Sec
m.net.mdm.iccid                          89302720513014237676
m.net.mdm.model                          35316B07SIM5360A
m.net.provider                           ROGERS
m.net.sq                                 -73dBm
m.net.type                               modem
m.serial
m.tasks                                  16
m.time.utc
m.version                                3.1.011-63-ge2fbf81d-dirty/ota_1/main (build idf v3.1-dev-1586-g65355d28 Nov 22 2018 06:53:00)
s.v2.connected                           yes
s.v2.peers                               1
s.v3.connected
s.v3.peers
v.b.12v.current
v.b.12v.voltage                          14.62V
v.b.12v.voltage.alert
v.b.12v.voltage.ref                      14.69V
v.b.c.temp                               10;10;12;11;11;12;11;11;10;10;10;9;10;10;10;11;11;11;11;11;10;10;14;9;10;8;10;10;11;9;10;9;10;8;9;9;9;9;10;9;11;9;10;9;10;9;9;9;10;9;9;9;9;9;8;9;9;9;9;9;10;10;10;10;9;9°C
v.b.c.temp.alert                         2;2;2;2;2;2;2;2;2;2;2;2;2;2;2;2;0;2;0;2;1;2;2;2;2;2;2;2;2;2;2;2;2;2;2;2;2;0;2;0;2;0;2;2;2;2;1;2;2;2;2;2;2;2;2;2;2;2;0;2;0;0;0;0;2;2
v.b.c.temp.dev.max                       -3.91;-3.91;7.09;7.09;7.09;7.09;7.09;7.09;7.09;7.09;-3.95;-3.91;-3.91;-3.91;-3.91;4.09;-2.95;4.7;-2.95;4.12;-3.91;-3.91;5.18;-3.91;-3.91;-3.95;7.09;7.09;7.09;7.09;7.09;7.09;7.09;7.09;-3.95;-3.95;-3.91;-2.95;3.08;-2.95;4.06;-2.95;3.94;-3.91;-3.91;-3.95;-3.91;-4.12;7.09;7.09;7.09;7.09;7.09;7.09;-3.95;-3.95;-3.91;-4.15;-2.95;-3.91;-2.95;-2.95;-2.95;-2.95;-3.91;-3.91°C
v.b.c.temp.max                           12;15;14;16;14;17;14;16;13;16;12;13;13;15;13;16;13;16;13;16;12;15;16;13;14;11;15;13;16;13;15;13;15;11;13;12;14;13;15;13;16;13;16;12;15;12;11;11;13;12;13;12;12;12;11;12;12;11;13;12;13;12;13;12;11;12°C
v.b.c.temp.min                           -1;-1;0;0;-1;0;-1;-1;-1;-1;-1;-1;-1;-1;-1;0;0;0;0;0;-1;-1;-1;-1;-1;-1;0;0;0;0;0;0;-1;-1;-1;-1;-1;0;0;0;0;0;0;-1;-1;-1;-1;-1;0;-1;0;-1;-1;-1;-1;-1;-1;-1;0;-1;0;0;0;0;-1;-1°C
v.b.c.voltage                            4.095;4.095;4.1;4.095;4.095;4.095;4.105;4.095;4.095;4.095;4.095;4.095;4.095;4.1;4.095;4.095;4.095;4.1;4.095;4.09;4.095;4.095;4.1;4.09;4.095;4.1;4.095;4.105;4.1;4.1;4.095;4.095;4.105;4.1;4.1;4.09;4.095;4.09;4.095;4.095;4.09;4.09;4.09;4.09;4.095;4.09;4.09;4.09;4.09;4.095;4.105;4.1;4.105;4.095;4.1;4.1;4.1;4.095;4.09;4.1;4.1;4.095;4.105;4.095;4.095;4.1;4.1;4.095;4.095;4.095;4.1;4.095;4.1;4.095;4.1;4.1;4.095;4.095;4.095;4.095;4.09;4.09;4.1;4.09;4.095;4.095;4.09;4.095V
v.b.c.voltage.alert                      1;1;2;2;2;2;2;2;2;2;2;2;2;2;2;2;2;2;1;2;2;1;2;1;2;2;2;2;2;2;2;2;2;2;2;2;2;2;2;2;1;1;2;2;1;1;1;2;2;2;2;2;2;2;2;2;2;2;2;2;2;2;2;2;2;2;2;1;1;2;2;2;2;2;2;2;2;2;2;2;2;2;2;2;2;1;1;1
v.b.c.voltage.dev.max                    0.02619;-0.02517;0.04619;0.04125;0.05125;0.05625;0.05125;0.06125;0.07625;0.07125;-0.06222;-0.06722;-0.0467;-0.0467;-0.03881;-0.04375;-0.04375;-0.03875;0.02983;0.03483;-0.03375;0.02983;-0.03875;0.02182;-0.03017;-0.03057;0.03625;0.04119;-0.04517;0.04125;0.07125;0.07125;0.0633;0.06625;-0.06722;-0.06222;-0.0467;0.04216;0.03909;-0.04045;0.02983;0.02983;0.0404;0.0304;-0.02875;0.02483;0.02182;0.03438;0.03455;0.03119;-0.04188;0.03625;0.05625;0.07125;0.06125;0.06125;-0.10222;-0.10722;-0.04062;-0.05045;0.03909;-0.03875;-0.06875;0.03409;0.03483;-0.04375;-0.04875;-0.02875;0.02619;0.03119;0.03455;0.04625;0.07625;0.07625;0.07125;0.07125;-0.09722;-0.10222;-0.04062;-0.05045;0.03438;-0.03563;-0.03875;-0.03062;-0.03875;0.02983;-0.02375;0.02483V
v.b.c.voltage.max                        4.105;4.105;4.11;4.105;4.105;4.11;4.11;4.105;4.105;4.105;4.105;4.11;4.11;4.105;4.105;4.11;4.105;4.105;4.105;4.105;4.105;4.11;4.11;4.105;4.11;4.105;4.105;4.11;4.11;4.105;4.11;4.11;4.11;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.11;4.105;4.11;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.11;4.105;4.105;4.105;4.105;4.105;4.11;4.105;4.105;4.11;4.11;4.105;4.11;4.105;4.105;4.105;4.11;4.105;4.105;4.105;4.105;4.105;4.11;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.11;4.105;4.105;4.105;4.105;4.105V
v.b.c.voltage.min                        3.45;3.45;3.445;3.455;3.445;3.46;3.425;3.44;3.44;3.44;3.445;3.445;3.445;3.445;3.445;3.44;3.44;3.445;3.46;3.46;3.45;3.455;3.435;3.475;3.44;3.44;3.45;3.44;3.435;3.43;3.46;3.46;3.425;3.435;3.44;3.475;3.45;3.46;3.445;3.45;3.46;3.46;3.475;3.475;3.455;3.46;3.48;3.49;3.47;3.46;3.44;3.46;3.44;3.455;3.44;3.445;3.43;3.435;3.45;3.435;3.45;3.445;3.415;3.465;3.465;3.44;3.435;3.455;3.465;3.465;3.45;3.46;3.455;3.46;3.45;3.46;3.445;3.455;3.45;3.445;3.475;3.46;3.445;3.455;3.445;3.46;3.46;3.46V
v.b.cac                                  2Ah
v.b.consumption                          0Wh/km
v.b.coulomb.recd
v.b.coulomb.used
v.b.current                              44.62A
v.b.energy.recd
v.b.energy.used
v.b.health
v.b.p.level.avg
v.b.p.level.max
v.b.p.level.min
v.b.p.level.stddev
v.b.p.temp.avg                           9.8°C
v.b.p.temp.max                           14°C
v.b.p.temp.min                           8°C
v.b.p.temp.stddev                        1.03°C
v.b.p.temp.stddev.max                    5.03°C
v.b.p.voltage.avg                        4.09597V
v.b.p.voltage.max                        4.105V
[... cut ...]
xmi.e.heating.watt                       0W
xmi.e.highbeam                           no
xmi.e.leftblinker                        no
xmi.e.lowbeam                            no
xmi.e.rearfog                            no
xmi.e.rightblinker                       no
xmi.e.warninglight                       no

You're picking up the voltage, current and power going into the battery, as 360V, 44A is what was reported on the charger.  It's just not showing up like AC charging is.

Let me know if you need anything else.

KommyKT, 2018-12-09:
i think i repair the problem with quick charge detection, you can download from same location.

gravtix, 2018-12-19:
@KommyKT I installed the update a few days ago.

I only reinstalled OVMS in the car today so I haven't had a chance to do a DC charge yet but I wanted to report some problems with this build:

It crashed this morning and I just did a short charge session up to 80% and it's now reporting an SOC of -1%.

I put some screenshots up here:

https://imgur.com/a/exieayj

The car is just parked right now, not charging.

m.freeram                                4215916
m.hardware                               OVMS WIFI BLE BT cores=2 rev=ESP32/1
m.monotonic                              16652Sec
m.net.mdm.iccid                          89302720513014237676
m.net.mdm.model                          35316B07SIM5360A
m.net.provider                           Stark
m.net.sq                                 -88dBm
m.net.type                               wifi
m.serial
m.tasks                                  16
m.time.utc
m.version                                3.1.011-88-gfe97f0a7-dirty/ota_0/main (build idf v3.1-dev-1586-g65355d28 Dec  8 2018 19:51:37)
s.v2.connected                           yes
s.v2.peers                               1
s.v3.connected
s.v3.peers
v.b.12v.current
v.b.12v.voltage                          12.61V
v.b.12v.voltage.alert
v.b.12v.voltage.ref                      12.6V
v.b.c.temp                               6;7;7;8;7;8;7;8;6;8;6;6;6;7;6;8;6;8;6;8;6;7;10;6;6;4;7;5;7;5;7;5;7;4;6;5;6;5;7;5;8;5;7;5;7;6;4;5;5;5;5;5;5;5;4;5;5;5;5;5;5;6;5;6;5;5°C
v.b.c.temp.alert                         1;1;2;2;2;2;2;2;2;2;1;1;1;1;0;0;0;1;0;1;1;1;1;1;1;1;2;2;2;2;2;2;2;2;1;1;1;1;1;0;1;1;1;1;1;1;1;1;2;2;2;2;2;2;1;1;1;1;1;0;0;0;1;0;1;0
v.b.c.temp.dev.max                       -2.58;-2.58;4.42;4.42;4.42;4.42;4.42;4.42;4.42;4.42;-2.77;-2.91;-2.58;-2.58;-1.91;2.12;-1.77;2.3;-1.91;2.71;-2.58;-2.58;4.38;-2.77;-2.77;-2.91;4.42;4.42;4.42;4.42;4.42;4.42;4.42;4.42;-2.77;-2.91;-2.91;-2.58;-2.58;-1.91;2.11;-2.77;2.23;-2.91;-2.58;-2.77;-2.91;-2.91;4.42;4.42;4.42;4.42;4.42;4.42;-2.91;-2.77;-2.91;-2.58;-2.58;-1.91;-1.91;-1.91;-2.58;-1.91;-2.91;-1.91°C
v.b.c.temp.max                           6;7;10;10;10;10;10;10;10;10;6;6;6;7;6;8;6;8;6;8;6;8;10;6;6;4;10;10;10;10;10;10;10;10;6;5;6;5;7;5;8;5;8;5;7;6;4;5;10;10;10;10;10;10;4;5;5;5;5;6;5;6;5;6;5;5°C
v.b.c.temp.min                           3;3;4;4;4;4;4;4;3;3;3;3;3;3;4;4;4;4;4;4;3;3;5;3;3;3;4;4;4;4;4;3;3;3;3;3;3;3;3;4;4;3;4;3;3;3;3;3;4;3;4;3;3;3;3;3;3;3;3;4;4;4;3;4;3;4°C
v.b.c.voltage                            4.005;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.005;4.005;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.005;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.015;4.01;4.005;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.005;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.005;4.01;4.005;4.01;4.01;4.01;4.01;4.005;4.01;4.01;4.01;4.005;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.005V
v.b.c.voltage.alert                      1;1;1;1;2;2;2;2;2;2;2;2;2;2;1;1;1;0;0;0;0;0;0;0;0;1;2;2;2;2;2;2;2;2;2;2;2;1;1;1;0;0;0;0;1;0;0;1;1;0;1;2;2;2;2;2;2;2;1;1;1;1;0;0;0;0;0;0;0;1;2;2;2;2;2;2;2;2;1;1;1;1;1;0;0;0;0;0
v.b.c.voltage.dev.max                    0.02216;0.02216;0.02716;0.02716;0.03676;0.03676;0.04676;0.04176;-0.04006;-0.03631;-0.03006;-0.03506;-0.03006;-0.03006;-0.02324;0.02324;0.02324;-0.01824;0.01824;0.01824;0.01324;0.01824;0.01858;0.01358;0.01858;0.02216;-0.03176;0.03216;0.03676;0.03676;0.04176;0.04176;-0.04131;-0.04131;-0.03506;-0.03284;-0.03006;-0.02784;-0.02324;-0.02324;-0.01824;-0.01824;0.01824;0.01472;0.02324;0.01824;0.01591;0.02148;0.02176;-0.0192;0.02716;-0.03176;0.03676;0.03676;0.04176;0.04176;-0.04131;-0.04131;-0.02784;-0.02784;0.02324;-0.02324;-0.01824;-0.01324;0.0158;0.01824;-0.01875;0.01358;0.01858;0.02205;-0.03176;-0.03176;0.03676;-0.03676;0.04176;0.04176;-0.04131;-0.03631;-0.02784;-0.02784;-0.02284;-0.02324;0.02324;-0.01824;0.01824;0.01972;0.01358;0.01369V
v.b.c.voltage.max                        4.015;4.015;4.015;4.01;4.015;4.015;4.015;4.01;4.01;4.015;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.015;4.015;4.01;4.015;4.015;4.015;4.01;4.015;4.015;4.01;4.01;4.015;4.015;4.015;4.015;4.015;4.01;4.015;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.015;4.01;4.01;4.01;4.01;4.015;4.015;4.01;4.015;4.01;4.015;4.015;4.015;4.01;4.01;4.015;4.015;4.01;4.015;4.015;4.01;4.01;4.015;4.01;4.01;4.01;4.015;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.01;4.015;4.01;4.01;4.01;4.01;4.01V
v.b.c.voltage.min                        3.72;3.72;3.72;3.725;3.725;3.72;3.705;3.71;3.72;3.715;3.725;3.725;3.72;3.715;3.715;3.725;3.72;3.72;3.72;3.72;3.725;3.72;3.715;3.725;3.715;3.71;3.73;3.72;3.71;3.715;3.715;3.72;3.705;3.71;3.715;3.735;3.715;3.725;3.725;3.725;3.725;3.73;3.73;3.73;3.715;3.72;3.725;3.735;3.725;3.725;3.71;3.72;3.71;3.72;3.715;3.715;3.715;3.72;3.715;3.71;3.73;3.725;3.71;3.725;3.73;3.715;3.705;3.715;3.725;3.73;3.715;3.725;3.73;3.735;3.72;3.73;3.72;3.72;3.715;3.725;3.735;3.73;3.715;3.73;3.715;3.73;3.715;3.725V
v.b.cac
v.b.consumption                          10081.8Wh/km
v.b.coulomb.recd
v.b.coulomb.used
v.b.current                              -120.25A
v.b.energy.recd                          -385012kWh
v.b.energy.used                          295338kWh
v.b.health
v.b.p.level.avg
v.b.p.level.max
v.b.p.level.min
v.b.p.level.stddev
v.b.p.temp.avg                           6.02°C
v.b.p.temp.max                           10°C
v.b.p.temp.min                           4°C
v.b.p.temp.stddev                        1.24°C
[... cut ...]
xmi.e.heating.temp.return                44°C
xmi.e.heating.watt                       0W
xmi.e.heating.watt                       1965.93W
xmi.e.highbeam                           no
xmi.e.leftblinker                        no
xmi.e.lowbeam                            no
xmi.e.rearfog                            no
xmi.e.rightblinker                       no
xmi.e.warninglight                       no
xmi.v.trip.consumption.KWh/100km         3.51267
xmi.v.trip.consumption.km/kWh            28.4684

I will try a DC charge and report back.

gravtix, 2018-12-21:
I noticed it's detecting a quick charge with me just charging on 120V outlet.

m.freeram                                4197372
m.hardware                               OVMS WIFI BLE BT cores=2 rev=ESP32/1
m.monotonic                              31021Sec
m.net.mdm.iccid                          89302720513014237676
m.net.mdm.model                          35316B07SIM5360A
m.net.provider                           Stark
m.net.sq                                 -80dBm
m.net.type                               wifi
m.serial
m.tasks                                  16
m.time.utc
m.version                                3.1.011-88-gfe97f0a7-dirty/ota_0/main (build idf v3.1-dev-1586-g65355d28 Dec  8 2018 19:51:37)
s.v2.connected                           yes
s.v2.peers                               1
s.v3.connected
s.v3.peers
v.b.12v.current
v.b.12v.voltage                          13.77V
v.b.12v.voltage.alert
v.b.12v.voltage.ref                      13.62V
v.b.c.temp                               12;13;14;15;14;15;13;14;12;13;11;12;12;13;13;14;13;15;13;14;11;14;16;12;13;11;14;12;14;12;14;12;13;11;11;11;12;12;14;12;14;12;14;11;13;11;11;11;12;11;12;11;12;11;10;10;11;11;12;12;12;12;12;12;11;11°C
v.b.c.temp.alert                         0;1;1;2;1;2;0;1;0;1;0;0;0;1;0;2;0;1;0;1;0;1;2;0;0;1;1;0;1;0;1;1;0;1;1;1;0;1;1;1;2;1;2;1;1;1;1;1;1;1;1;1;1;1;1;1;1;1;1;1;1;1;1;1;1;1
v.b.c.temp.dev.max                       -1.29;2.59;2.33;4.53;2.3;5.17;1.56;4.29;-1.65;2.48;-1.74;-1.74;-1.74;2.48;1.23;4.09;1.89;4.45;1.35;3.67;-1.73;3.45;5.45;1.45;1.52;-3.29;3.33;-1.59;3.61;-1.62;3.24;-2.29;1.53;-3.29;-2.55;39.71;-1.74;-2.29;2.58;-2.29;3.45;-2.29;3.45;-2.29;2.45;-2.12;-3.29;-3.29;-2.29;-2.38;-2.29;-2.59;-2.29;-2.5;-3.29;-3.29;-3.29;-4.29;-2.29;-3.29;-2.29;-2.29;-2.29;-2.29;-3.29;-3.29°C
v.b.c.temp.max                           12;14;14;16;14;16;13;15;12;14;11;12;12;14;13;15;13;15;13;15;11;14;16;13;13;11;15;12;15;12;15;12;14;11;12;50;13;12;14;12;15;12;15;11;14;11;11;11;12;12;12;12;12;12;11;11;11;11;12;12;12;12;12;12;11;12°C
v.b.c.temp.min                           8;8;9;10;9;10;9;9;7;8;7;7;7;7;8;9;9;9;8;9;7;8;9;8;8;7;9;8;9;8;9;8;7;7;7;7;7;8;8;8;9;8;9;7;8;7;7;7;8;7;8;7;7;7;6;6;6;6;7;7;8;7;7;7;7;7°C
v.b.c.voltage                            3.925;3.925;3.93;3.925;3.93;3.93;3.93;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.93;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.93;3.925;3.925;3.93;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.93;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925;3.925V
v.b.c.voltage.alert                      1;2;2;2;2;2;2;2;2;2;2;2;2;2;2;2;1;1;1;0;1;1;1;2;2;1;2;2;2;2;2;2;2;2;2;2;2;2;2;2;1;1;1;1;0;0;2;2;2;2;2;2;2;2;2;2;2;2;2;2;2;2;2;1;1;1;0;0;2;2;2;2;2;2;2;2;2;2;2;2;2;2;1;1;1;0;1;0
v.b.c.voltage.dev.max                    0.0279;0.0329;0.0379;0.0379;0.0429;0.0429;0.0329;0.0379;0.04159;0.04659;-0.0521;-0.0571;-0.0471;-0.0571;-0.0421;-0.0371;-0.0271;-0.02233;-0.02233;0.01972;0.02307;0.02307;0.0279;0.0329;-0.03193;0.0279;0.0379;0.0329;0.0329;0.0329;0.0429;0.0379;0.04159;0.04159;-0.0621;-0.0521;-0.0471;-0.0421;-0.0371;-0.0421;-0.02233;-0.02233;0.02125;0.02114;0.01807;0.01807;0.0329;0.0329;0.0329;0.0329;0.0329;0.0329;0.0329;0.0379;0.03574;0.0379;-0.0621;-0.0621;-0.0471;-0.0621;-0.0321;-0.0421;-0.0321;-0.0221;0.02307;0.02807;-0.0196;0.01807;0.0329;0.0329;0.0329;0.0329;0.0379;0.0329;0.0379;0.0329;-0.0671;-0.0571;-0.0571;-0.0571;-0.0321;-0.0421;-0.0271;-0.0271;0.02307;0.01807;0.02307;0.01807V
v.b.c.voltage.max                        4.105;4.105;4.105;4.105;4.105;4.105;4.11;4.105;4.105;4.105;4.105;4.11;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.11;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105;4.105V
v.b.c.voltage.min                        3.695;3.7;3.7;3.705;3.705;3.71;3.685;3.7;3.7;3.695;3.69;3.695;3.69;3.69;3.695;3.7;3.7;3.7;3.695;3.705;3.695;3.7;3.69;3.7;3.69;3.69;3.705;3.695;3.695;3.695;3.7;3.71;3.69;3.69;3.685;3.705;3.38;3.705;3.7;3.695;3.705;3.71;3.715;3.71;3.7;3.705;3.705;3.705;3.7;3.695;3.685;3.7;3.695;3.7;3.69;3.7;3.685;3.69;3.69;3.685;3.7;3.695;3.685;3.705;3.7;3.69;3.69;3.695;3.695;3.7;3.69;3.7;3.7;3.7;3.69;3.7;3.69;3.695;3.685;3.69;3.705;3.7;3.7;3.7;3.695;3.7;3.69;3.7V
v.b.cac
v.b.consumption                          0Wh/km
v.b.coulomb.recd
v.b.coulomb.used
v.b.current                              2.38A
v.b.energy.recd                          0kWh
v.b.energy.used                          0kWh
v.b.health
v.b.p.level.avg
v.b.p.level.max
v.b.p.level.min
v.b.p.level.stddev
v.b.p.temp.avg                           12.39°C
v.b.p.temp.max                           16°C
v.b.p.temp.min                           10°C
v.b.p.temp.stddev                        1.32°C
v.b.p.temp.stddev.max                    5.39°C
v.b.p.voltage.avg                        3.92545V
v.b.p.voltage.max                        3.93V
v.b.p.voltage.min                        3.925V
v.b.p.voltage.stddev                     0.00136V
[... cut ...]
xmi.e.heating.amp                        0A
xmi.e.heating.temp.flow                  45.2°C
xmi.e.heating.temp.return                43.4°C
xmi.e.heating.watt                       0W
xmi.e.highbeam                           no
xmi.e.leftblinker                        no
xmi.e.lowbeam                            no
xmi.e.rearfog                            no
xmi.e.rightblinker                       no
xmi.e.warninglight                       no
xmi.v.trip.consumption.KWh/100km         0
xmi.v.trip.consumption.km/kWh            0

gravtix, 2018-12-24:
@KommyKT

I tried DC charging and it works.  

https://i.imgur.com/bDHCwTR.png

Only problem I saw was that when I stopped the session, it still seemed to think the car is charging.

I tried Level 2 AC charging and that works fine.

So it's just that and Level1 AC charging being detected as DC charging.

KommyKT, 2018-12-24:
Now i changed the code for better charge detection.  Now use the old method the charge states. 
Last night i tested level 2 AC charging, and works fine, i think now Level 1 AC is same as Level 2.
DC charging is works i think with new code, and if SOC < 80 then detected interrupted, if SOC <= 80 is completed charge, on AC SOC < 92 interrupted, anything else completed.
The crash i think is my fault, because i don't update esp-idf.

KommyKT, 2018-12-30:
Battery temp, is coming from battery pack avg, but my mistake i set to Voltage, not temp, now show good value. 
The DC charge current is v.b.current 48.78A , i set it to v.c.current, but i don't understand why not show.

The topoff/ballancing only can working on Level 1-2 charging, now not tested properly. 
My car not preheating/cooling, and my searching only after 2013 made Mitsubitshi known this feature. If my car connected to charger (AC), and set heating on only fan working , heating not start.

gravtix, 2018-12-30:
Here's how a DC charge looked for me yesterday

https://i.imgur.com/Kl6zCIR.png

v.c.current is 0 for some reason and the state is balancing.

I thought balancing only happens at 95%, is that something from the CANbus?

I have a remote for my 2012 and I can start cooling/heating remotely.  I can see the status via metrics list xmi so that works.

KommyKT, 2018-12-30:
And what happen when you, plug in level1-2 charger and on middle console power on heater.

Can you create a log file from pre heating with remote control?
You have to connect to OVMS with ssh.

KommyKT, 2018-12-31:
now new version on my NAS.
Corrected DC charging current and voltage measure, and added all main project updates.

KommyKT, 2018-12-31:
Now i updated the code on NAS. Now Energy usage now works fine, i think first charge detection finished.
But i have a bug if connect to own charger, web interface/ssh not working over wi-fi, but i can ping the ovms. My car is connected to my home network, but own network is unreacheable

gravtix, 2018-12-31:
@KommyKT Is this version -151? I installed it this morning and level 1/2 charging seems to work fine.  I'll try DC charging soon.

 It seems to be updating kWh/100km while parked and charging.

When I'm plugged in on I cant turn on the heater or AC, only using the remote.  

I'll connect via SSH and collect some logs.  I just need to find the log commands.

KommyKT, 2019-01-03:
Now i tested on DC charger, i think everything works fine, only mistake current is negative.

gravtix, 2019-01-04:
@KommyKT I'll try out a DC charge.  Sounds like it's all fixed.

I'll open another issue and we can collect more data.

Running nc seems to crash my OVMS module.  I need to reset it as I've lost all connectivity to it.

gravtix, 2019-01-06:
@KommyKT.

I did a test with AC charging today, couldn't test DC as it was in use.

Started a charge at 15%

Standard - Charging
221.0V/5.6A
SOC: 15.0%
Ideal range: 8km
Est. range: 6km
ODO: 49255.0km
SOH: 92%

At 29% the i-MIEV shut charging off for 20 minutes as normal, car reports just a 0.1A current at this time.

Standard - Charging
222.0V/0.1A
SOC: 29.0%
Ideal range: 29km
Est. range: 21km
ODO: 49255.0km
SOH: 92%

But when it resumed it came back as topping off:

Balancing - Topping off
218.0V/14.2A
SOC: 29.0%
Ideal range: 29km
Est. range: 21km
ODO: 49255.0km
SOH: 92%

KommyKT, 2019-01-11:
Added new version to my NAS. 
Now balancing (AC) only show above 92% when DC current at AC charge is negative. No balancing on DC charger.
Now "stat" command show last charge used kwh, on AC and DC side. (if quickcharge only show DC AC=0).
Now CAC computed from SOH value.

Ideal range set to 130km, about 11.5kwh/100km consumption.

Now i test DC charge some time, i think everything works fine. Only "problem" notification show alway 0.0V/0.0A, at start. But if charge started fully, stat command show the real values (notification sent stat command output)

jjlink, 2019-01-18:
It this going to be included in the coming 3.2.000 OTA release?

KommyKT, 2019-01-20:
Updated the fw on my NAS
Updated:
- reworked code for temp and voltage detection
- support 88 cell and 80 cell cars

gravtix, 2019-01-21:
I'll give it a try.  Not having any problems with the 3.2 EAP release so far.

jjlink, 2019-01-24:
I have the 88 cell I-MiEV and was running v3.2.001 this morning and I was getting bombed with these messages wile charging:

[0;33mW (3338709) vehicle: BMS new alerts: 0 voltages, 8 temperatures[0m
[0;32mI (3338719) ovms-server-v2: Send MP-0 PAVoltage: SD=5mV OK|Temperature: SD=1.9C !4:+3.6C !6:+3.9C !8:+3.7C !20:+3.7C !41:+3.6C[0m
[0;33mW (3339709) vehicle: BMS new alerts: 0 voltages, 8 temperatures[0m
[0;32mI (3339719) ovms-server-v2: Send MP-0 PAVoltage: SD=5mV OK|Temperature: SD=1.9C !4:+3.6C !6:+3.9C !8:+3.7C !20:+3.7C !41:+3.6C[0m
[0;33mW (3340709) vehicle: BMS new alerts: 0 voltages, 8 temperatures[0m
[0;32mI (3340719) ovms-server-v2: Send MP-0 PAVoltage: SD=5mV OK|Temperature: SD=1.9C !18:+3.9C !31:+3.5C !43:+3.6C[0m
[0;33mW (3341709) vehicle: BMS new alerts: 0 voltages, 8 temperatures[0m
[0;32mI (3341709) ovms-server-v2: Send MP-0 PAVoltage: SD=5mV OK|Temperature: SD=1.9C !18:+3.9C !31:+3.5C !43:+3.6C[0m
[0;33mW (3342709) vehicle: BMS new alerts: 0 voltages, 8 temperatures[0m
[0;32mI (3342719) ovms-server-v2: Send MP-0 PAVoltage: SD=5mV OK|Temperature: SD=1.9C !18:+3.9C !31:+3.5C !43:+3.6C[0m
[0;33mW (3343709) vehicle: BMS new alerts: 0 voltages, 8 temperatures[0m
[0;32mI (3343719) ovms-server-v2: Send MP-0 PAVoltage: SD=5mV OK|Temperature: SD=1.9C !18:+3.9C !31:+3.5C !43:+3.6C[0m
[0;33mW (3344709) vehicle: BMS new alerts: 0 voltages, 8 temperatures[0m
[0;32mI (3344719) ovms-server-v2: Send MP-0 PAVoltage: SD=5mV OK|Temperature: SD=1.9C !18:+3.9C !31:+3.5C !43:+3.6C[0m
[0;33mW (3345709) vehicle: BMS new alerts: 0 voltages, 8 temperatures[0m
[0;32mI (3345719) ovms-server-v2: Send MP-0 PAVoltage: SD=5mV OK|Temperature: SD=1.9C !18:+3.9C !31:+3.5C !43:+3.6C[0m
[0;33mW (3346709) vehicle: BMS new alerts: 0 voltages, 8 temperatures[0m
[0;32mI (3346719) ovms-server-v2: Send MP-0 PAVoltage: SD=5mV OK|Temperature: SD=1.9C !18:+3.9C !31:+3.5C !43:+3.6C[0m
[0;33mW (3347709) vehicle: BMS new alerts: 0 voltages, 8 temperatures[0m
[0;32mI (3347719) ovms-server-v2: Send MP-0 PAVoltage: SD=5mV OK|Temperature: SD=1.9C !18:+3.9C !31:+3.5C !43:+3.6C[0m
[0;33mW (3348709) vehicle: BMS new alerts: 0 voltages, 8 temperatures[0m
[0;32mI (3348719) ovms-server-v2: Send MP-0 PAVoltage: SD=5mV OK|Temperature: SD=1.9C !18:+3.9C !31:+3.5C !43:+3.6C[0m
[0;33mW (3349709) vehicle: BMS new alerts: 0 voltages, 8 temperatures[0m
[0;32mI (3349719) ovms-server-v2: Send MP-0 PAVoltage: SD=5mV OK|Temperature: SD=1.9C !18:+3.9C !31:+3.5C !43:+3.6C[0m
[0;33mW (3350709) vehicle: BMS new alerts: 0 voltages, 8 temperatures[0m
[0;32mI (3350719) ovms-server-v2: Send MP-0 PAVoltage: SD=5mV OK|Temperature: SD=1.9C !18:+3.9C !31:+3.5C !43:+3.6C[0m
[0;33mW (3351709) vehicle: BMS new alerts: 0 voltages, 8 temperatures[0m
[0;32mI (3351719) ovms-server-v2: Send MP-0 PAVoltage: SD=5mV OK|Temperature: SD=1.9C !18:+3.9C !31:+3.5C !43:+3.6C[0m
[0;33mW (3352709) vehicle: BMS new alerts: 0 voltages, 8 temperatures[0m
[0;32mI (3352719) ovms-server-v2: Send MP-0 PAVoltage: SD=5mV OK|Temperature: SD=1.9C !4:+3.6C !6:+3.9C !8:+3.7C !20:+3.7C !41:+3.6C[0m
[0;33mW (3353709) vehicle: BMS new alerts: 0 voltages, 8 temperatures[0m
[0;32mI (3353719) ovms-server-v2: Send MP-0 PAVoltage: SD=5mV OK|Temperature: SD=1.9C !4:+3.6C !6:+3.9C !8:+3.7C !20:+3.7C !41:+3.6C[0m
[0;33mW (3354709) vehicle: BMS new alerts: 0 voltages, 8 temperatures[0m
[0;32mI (3354709) ovms-server-v2: Send MP-0 PAVoltage: SD=5mV OK|Temperature: SD=1.9C !4:+3.6C !6:+3.9C !8:+3.7C !20:+3.7C !41:+3.6C[0m
[0;33mW (3355709) vehicle: BMS new alerts: 0 voltages, 8 temperatures[0m
[0;32mI (3355709) ovms-server-v2: Send MP-0 PAVoltage: SD=5mV OK|Temperature: SD=1.9C !4:+3.6C !6:+3.9C !8:+3.7C !20:+3.7C !41:+3.6C[0m
[0;33mW (3356709) vehicle: BMS new alerts: 0 voltages, 8 temperatures[0m
[0;32mI (3356709) ovms-server-v2: Send MP-0 PAVoltage: SD=5mV OK|Temperature: SD=1.9C !4:+3.6C !6:+3.9C !8:+3.7C !20:+3.7C !41:+3.6C[0m
[0;33mW (3357709) vehicle: BMS new alerts: 0 voltages, 3 temperatures[0m
[0;32mI (3357709) ovms-server-v2: Send MP-0 PAVoltage: SD=5mV OK|Temperature: SD=1.9C !18:+3.9C !31:+3.5C !43:+3.6C[0m

OI just switched to your updated version: 3.1.011-259-g89e36d82-dirty/ota_1/eap (build idf v3.1-dev-2835-g151269458 Jan 20 2019 13:37:41)

gravtix, 2019-01-24:
I turned off the BMS notifications a while ago as I'm not sure what an abnormal value is.  I've seen pretty big differences in temperature during charging.

jjlink, 2019-01-24:
Thanks for the info, I just turned off this setting: "Enable alert notifications".

gravtix, 2019-01-24:
It's something I hope to figure out over time.

I'd say a voltage drop is more useful.  I had a big voltage drop in one of my cells and the BMS restarted and I lost a bar.  Would have been useful to get a message.

I've seen 10-15C difference in the cells after DC charging.

Maximum/minimum temperature warning might be useful too.  The car will not run or charge or if it's too low and very high temps are bad for the pack.

KommyKT, 2019-02-10:
I found a bug: 
SOC < 10 (car remain km show ---)  and connect to home charger , car show Quick charging.

FilipeF85, 2019-05-23:
Hi, i have Mitsubishi imiev 2015 with ovms V3 installed.
On Main Version Odometer tells me the correct KM on it.
But when i installed your latest firmware version "3.2.001-176-gf580441a-dirty" or the "3.2.002/ota_0/main (build idf v3.2-dev-1420-g151269458 May 12 2019 16:17:52)"
the ODO tells me incorrect KM.
Actual OVMS INFO:
Standard - Charging
224.0V/9.1A
SOC: 85.5%
Est. range: 56km
ODO: 32467.0km
SOH: 100%
AC charge: 1.366kWh kWh
DC charge: 0.973kWh kWh

But on the car real dashboard ODO have: 97747KM

KommyKT, 2019-05-23:
I don't understand this problem because the odometer code from V2 and not modified.

Can you create a can messages log file for me? 

"To collect can message one methode is.
Connect to the OVMS via ssh then the following commands

re start
re mode serve
re serve format crtd
On other terminal, or exit from ovms the following command can save to file:
nc 192.168.4.1 3000 | tee file.crtd"

then set the car to Ready, 10 second log is enough

FilipeF85, 2019-05-23:
> I don't understand this problem because the odometer code from V2 and not modified.
>
> Can you create a can messages log file for me?
>
> "To collect can message one methode is.
> Connect to the OVMS via [...]

I can create log file for you no problem, but i`m having problem with yours instructions.
I have connect with putty to ovms but having problem with last part of yours instructions to save file:
This following part:
 > On other terminal, or exit from ovms the following command can save to file:
> nc 192.168.4.1 3000 | tee file.crtd"

KommyKT, 2019-05-23:
Which operation system do you use?

FilipeF85, 2019-05-23:
I have windows 10 but i can install another one if you want?

dexterbg, 2019-05-23:
Alternatively:
- `can log crtd /sd/file.crtd`
- …record…
- `can log off`
- http://ovmsname.local/file.crtd

FilipeF85, 2019-05-23:
> * http://ovmsname.local/file.crtd

Thank you for your help, i already have the log file.

FilipeF85, 2019-05-23:
> Which operation system do you use?

I have the log file file can i send to you by email or you prefer another service ?

KommyKT, 2019-05-23:
kommykt at gmail.com my e-mail

FilipeF85, 2019-05-23:
> kommykt at gmail.com my e-mail

Email send.

KommyKT, 2019-05-24:
> Email send. 
I uploaded a new version to my nas, all commits in this version, this is a latest version, + fixed the odometer. 
Your odometer is now: 97780

KommyKT, 2019-05-24:
BMS Cell Monitor:
run the following command: **me li v.b**
and show the output

FilipeF85, 2019-05-24:
BMS Cell Monitor output car turned off and car turned ON, attach on email.

OVMS# me li v.b
v.b.12v.current
v.b.12v.voltage                          14.28V
v.b.12v.voltage.alert
v.b.12v.voltage.ref                      12.6V
v.b.c.temp
v.b.c.temp.alert
v.b.c.temp.dev.max
v.b.c.temp.max
v.b.c.temp.min
v.b.c.voltage
v.b.c.voltage.alert
v.b.c.voltage.dev.max
v.b.c.voltage.max
v.b.c.voltage.min
v.b.cac                                  48Ah
v.b.consumption                          0Wh/km
v.b.coulomb.recd
v.b.coulomb.used
v.b.current                              -1.76A
v.b.energy.recd                          0kWh
v.b.energy.used                          0.013476kWh
v.b.health
v.b.p.level.avg
v.b.p.level.max
v.b.p.level.min
v.b.p.level.stddev
v.b.p.temp.avg
v.b.p.temp.max
v.b.p.temp.min
v.b.p.temp.stddev
v.b.p.temp.stddev.max                    0�C
v.b.p.voltage.avg
v.b.p.voltage.max
v.b.p.voltage.min
v.b.p.voltage.stddev
v.b.p.voltage.stddev.max
v.b.power                                0.615825kW
v.b.range.est                            39km
v.b.range.full
v.b.range.ideal                          0km
v.b.soc                                  76%
v.b.soh                                  100%
v.b.temp                                 0�C
v.b.voltage                              351.9V

OVMS# me li v.b
v.b.12v.current
v.b.12v.voltage                          11.97V
v.b.12v.voltage.alert
v.b.12v.voltage.ref                      12.6V
v.b.c.temp
v.b.c.temp.alert
v.b.c.temp.dev.max
v.b.c.temp.max
v.b.c.temp.min
v.b.c.voltage
v.b.c.voltage.alert
v.b.c.voltage.dev.max
v.b.c.voltage.max
v.b.c.voltage.min
v.b.cac                                  48Ah
v.b.consumption                          0Wh/km
v.b.coulomb.recd
v.b.coulomb.used
v.b.current                              0A
v.b.energy.recd                          0kWh
v.b.energy.used                          0kWh
v.b.health
v.b.p.level.avg
v.b.p.level.max
v.b.p.level.min
v.b.p.level.stddev
v.b.p.temp.avg
v.b.p.temp.max
v.b.p.temp.min
v.b.p.temp.stddev
v.b.p.temp.stddev.max                    0�C
v.b.p.voltage.avg
v.b.p.voltage.max
v.b.p.voltage.min
v.b.p.voltage.stddev
v.b.p.voltage.stddev.max
v.b.power                                0kW
v.b.range.est
v.b.range.full
v.b.range.ideal                          0km
v.b.soc
v.b.soh                                  100%
v.b.temp                                 0�C
v.b.voltage

FilipeF85, 2019-05-24:
> BMS Cell Monitor:
> run the following command: **me li v.b**
> and show the output

E-mail send with outputs Car OFF and Car ON.

FilipeF85, 2019-05-25:
> > Email send.
> 
> I uploaded a new version to my nas, all commits in this version, this is a latest version, + fixed the odometer.
> Your odometer is now: 97780

The odometer is working great with your lastest firmware version, the "BMS Cell Monitor" is not working when we open only show blank page and no graphics.
Thanks for your help.

markwj, 2020-02-02:
Is this resolved now? If not, as the issue is so long now, perhaps better to split off each individual remaining problem to it's own GitHub issue.

KommyKT, 2020-02-02:
The main problem is solved. Only problem is the last comment.

FilipeF85, 2020-02-02:
The actual problems is the BMS Cell Monitor only shows blank page.

KommyKT, 2020-02-08:
> The actual problems is the BMS Cell Monitor only shows blank page.

I uploaded a new version to my NAS ( If you want to test my code, go to web interface on OVMS, Config/firmware, on firmware page select Flash from web tab and add this webpage address: http://kommykt-nas.ddns.net: 22289/ovms3.bin I add two whitespace before 22289 the correct is ....net:22289.... If you test it please send me a note of working /not working) My internet is a bit slow but working.

The BMS Cell Monitor only not working for newer cars,(old car my is 2012 is working fine) but no other car is available near me to test the code. 
Car cells only access when the in READY. 

Please check this. Check the  **me li v.b** command, after set the car to ready and after 1-2 minutes if has more values in **v.b.c.voltage** the code is working.

gravtix, 2020-02-09:
The original issues I opened this on were fixed in later versions.

I've only noticed some minor glitches like distance driven in xmi trip command is always 0 even though it calculated the energy consumption correctly.

I will try the new version posted by @KommyKT

FilipeF85, 2020-03-14:
> I uploaded a new version to my NAS ( If you want to test my code, go to web interface on OVMS, Config/firmware, on firmware page select Flash from web tab and add this webpage address: [...]

I have updated today to your new firmware but now the odometer is not working right again.
Shows: 48152,0KM
Real Odometer Value: 113432

KommyKT, 2020-03-14:
> I have updated today to your new firmware but now the odometer is not working right again.
> Shows: 48152,0KM
> Real Odometer Value: 113432

And the main problem is solved? bms cell monitor is working?

FilipeF85, 2020-03-15:
> And the main problem is solved? bms cell monitor is working?

I have test it now but the BMS Cell Monitor  is not working yet, if you need help on on logs for help you just ask.
Thanks for your help.

closed 2020-04-18 by dexterbg