openvehicles/Open-Vehicle-Monitoring-System-3 #178: V3 MQTT SSL support?

glynhudson, 2018-12-17:
Hi,

Does the V3 server support MQTTS encrypted transfer over SSL?

If not is it possible to use the old (V1/V2?) API to access SOC data? I'm developing an integration with OpenEVSE to use OVMS to inform an OpenEVSE charging point of the EV SOC. It's important that user's data is secure. It would also be great to be able to select on the module which metrics are published to MQTT e.g. GPS data is not requited for this application.

dexterbg, 2018-12-22:
Regarding MQTT SSL or topic control: not yet.

The V2 protocol has some builtin encryption, it's old & weak (RC4) but better than nothing.

Regarding SOC via V2 API: of course. See V2 client and Android scripts:
- https://github.com/openvehicles/Open-Vehicle-Monitoring-System/tree/master/client
- https://github.com/openvehicles/Open-Vehicle-Android/wiki/Update-Broadcast-Intent-data

You can also retreive the data via the V2 server REST API (JSON format, no SSL), or if you're on my server, you can use my additional REST API also available via SSL:
- http://lists.openvehicles.com/pipermail/ovmsdev/2018-July/005352.html

M4GNV5, 2018-12-22:
I tried implementing SSL support in the past but stopped.

The problem is that it requires certificate management. The implementation of OpenSSL which comes with the esp-idf does not implement `SSL_CTX_load_verify_locations` so it cannot load certs from the file system (at least not automatically).
Maybe using mbedtls would work, but I have no experience with that.

Alternatively you could simply tell mongoose to accept any certificate, but that's bad security wise and implementing proper CA handling would also help so we can change the OTA updates to be downloaded via https instead of http.

mjuhanne, 2019-08-08:
The certificate management is something which should be worked on. I use mbedTls with Pushover notification support (only in my fork at this time) and the SSL works fine, but at this stage it just accepts any certificate..

dexterbg, 2021-12-30:
Mark solved this in commit 064cf6e5aee971bf75271be9b83eeb9b87b260ab.

closed 2021-12-30 by dexterbg