openvehicles/Open-Vehicle-Monitoring-System-3 #180: Leaf model year needs to be set manually

arsharpe, 2018-12-30:
It appears that the model year for 2016 Leafs needs to be set manually using
     config set xnl modelyear 2016

It would make the user configuration a lot easier (ie plug-n-play) if the code could detect to the model year (eg from status signals that were introduced at that year or any year info in the VIN number)

ltjessem, 2019-02-21:
The 10th character of the VIN tells the model year, starting from A (2010)
My 2015 is F, 2016 is G, 2018 is J . 

The battery pack size is also indicated in the VIN, for the "old" pre-2018 models the 4th character is A = 24kWh and B = 30kWh. On the newer 2018+ then A = 40 kWh, B will presumably be the 62 kWh

ankohanse, 2019-02-22:
Cool, that will allow auto detection on cars that have such a VIN. 

Unfortunately cars made for the Japanese Domestic Market and frequently imported 2nd hand into Australia and New Zealand do not use a VIN but instead have some sort of serial number in the form of ZE0-123456 or AZE0-123456. I have done extensive searches via Google, but have not yet found a reliable relation between such serial numbers and the model-year or battery size...

ltjessem, 2019-02-22:
Found this on a NZ forum, I'll look into this a bit more and see if I can figure something out

> My understanding for a 24 or 30 kw is the chassis number if it is AZE0 2xxxxx then it is a 30kw if it is AZE0 1xxxxx then it is a 24 but I may be wrong

ankohanse, 2019-02-23:
Posted a request on that forum (www.geekzone.co.nz) for people to email me their JDM style VIN combined with year and battery size. Hopefully that will give us some new insights...

I very much hope that can confirm that AZE0-1xxxxx is 24kwh, AZE0-2xxxxx is 30kwh, AZE1-1xxxxx is 40kwh and possibly AZE1-2xxxxx will be 62kwh    :-)

ankohanse, 2019-03-02:
My conclusions after receiving input from users on GeekZone and Facebook:

A. The JDM-style VIN cannot be used to predict battery size. Typically, the 2016 and 2017 models have both 24kwh and 30kwh cars in the same VIN range. See https://www.japan-partner.com/check-manufacture-year.php
 
B The JDM VIN can be used as a rough indication of model year:
 
ZE0-   2011-2012  (24kwh)
AZE0-0 and AZE0-1  2013-2015  (24kwh) 
AZE0-2    2016-2017   (24 and 30kwh) 
ZE1-        2018-           (40 and later possibly 60kwh)

Note that UK based VINs look different again from the USA and Japanese VINs. So we would need to find a relation for those as well.

ankohanse, 2019-03-02:
Other alternatives for determining the model year (or at least distinguish pre-2016 from 2016 and later cars):

A. Analyse traffic on the CAN bus. For pre-2016 models we need to send 1 byte commands, while the 2016 and later use 4 byte command sizes. Do we see similar commands from other units going over the bus?

B. Request versions or part numbers (i.e. poll using pid 0x83 or 0x84) of various components (battery management,TCU, ...). There only seem to be very limited amount of part numbers around per model. 
Part numbers are documented by nissan on pages like this one: https://nissan.7zap.com/en/er/leaf/ze0/1+1011+65535/e/291/

glynhudson, 2019-03-03:
I have a UK Nissan leaf 24kWh 2014 and 2015 24kWh Nissan e-NV200 (OVSM also works on the env200) I can provide my VIN and a number of cars I have access to. However, this auto detection of model year sound tricky to make 100% reliable. I applaud the work you have done so far (very impressive) but how about just use a drop down selection of battery size and model year? Anyone clued-up enough to be using OVMS would certainly know the model year and battery size of their car, it's one-off setup step therefore not much hassle

caederus-ovms, 2019-03-03:
On Sat, Mar 02, 2019 at 04:43:06PM -0800, Glyn Hudson wrote:
> I have a UK Nissan leaf 24kWh 2014 and 2015 24kWh Nissan e-NV200 (OVSM also works on the env200) I can provide my VIN and a number of cars I have access to.

I only have regular access to my own UK 2015 24kWh, so it would be very
helpful to have some annotated logs from these other models.
We're looking at adding range and charge time estimates at the moment,
so getting some early feedback on that development code would be great.

ankohanse, 2019-03-03:
Hi Glyn,

The drop down boxes for year and battery size you describe were implemented by me a couple of weeks ago and will be in the upcoming 3.2.002 release.

What I am looking for is an automatic initialization for these values, so that the system will just work out of the box as well, without going into the configuration. Just a  tiny step to make the system a bit easier to install and allow slightly less techie people to use it as well as the current audience…

For now I have dropped the idea of deriving the modelyear and battery size from the VIN. Indeed, it is too tricky to make it work reliably. Meanwhile, we might have already found a CAN bus message that will provide us the battery size.
As for the modelyear, we only use it to determine in which format to send the commands (aircon on/off, etc) to the CAN bus. We are still looking for a CAN bus message that will be an indicator for that…

Just out of curiosity, would you mind telling us what an UK ENV-200 VIN looks like (with lower digits masked) ?  Are they distinguishable from Leaf VINs or using exactly the same format and range?

OKidoki,
Anko

dalathegreat, 2020-11-28:
A more robust solution would be to listen for specific CAN-messages that are exclusive between the ZE0 / AZE0 / ZE1. That way you can get it to autodetect 100% successfully.

E.g. 
Start by assuming vehicle is a 2011-2013 ZE0
if CAN message 0x393 is identified on the EV-CAN, the vehicle is infact an AZE0 2014-2017
if CAN message 0x1ED is identified on the EV-CAN, the vehicle is infact a ZE0 2018+ vehicle

Then when you have determined the generation of the vehicle, you either identify the battery size from amount of gids when fully charged, or by looking at more CAN-messages that are specific to 24/30/40/62kWh batteries.

Just a thought?

ankohanse, 2020-11-29:
Hi Daniel,

Yes, your approach could work (did not look into the specific CAN messages). At the moment the model year setting is only used to determine how to issue commands to the CAN bus (climate control start/stop, charge start, doors lock/unlock). So we indeed only need to differ between ZE0 / AZE0 / ZE1 and do not need the specific year within those ranges.

So could change the setting from model year onto something more like ‘model generation’.  The setting is specific for Nissan Leaf and not a generic one over all models, so no blocks there from changing it (VoltAmpera and VwUp use their own similar setting).

We already have similar capability to auto detect battery size. There we offer the user to use either autodetected or overridden ‘maxgids’ if needed. Don’t really see the need for an override mechanism here, but someone on this forum might come up with a good reason…