openvehicles/Open-Vehicle-Monitoring-System-3 #181: 30kwh Leaf max gids needs to be set manually

arsharpe, 2018-12-30:
It appears that for 30 kwh (2016 and later) Leafs the maxGids needs to be set manually using
config set xnl maxGids 356

It would make the user configuration a lot easier (ie plug-n-play) if the code could detect to the battery size (eg from available capacity or a component of the VIN number) and set the GIDs accordingly.

caederus-ovms, 2019-02-23:
It should no longer be necessary to set xnl.maxGids to get a useful SOC reading.
The default behaviour is now to report the "instrument" SOC reading, as computed by the car internally and reported on the dash of newer models.
The older alternate method, calculating SOC as a percentage of energy in the battery relative to an arbitrary "new car" value, can be selected by: "config set xnl soc.newcar true" and "config set xnl maxGids ..." to a value of your choice.
To revert to the default behaviour: "config set xnl soc.newcar false".

dalathegreat, 2021-12-26:
@dexterbg please close this issue, no longer relevant

closed 2021-12-26 by dexterbg