openvehicles/Open-Vehicle-Monitoring-System-3 #189: WebSocket stream data corruption

dexterbg, 2019-01-09:
Notes: this is not a new issue, I've seen the effect since working on the websocket connection. It also may not only affect the websocket stream or only wifi transmissions, other protocols can possibly mitigate the data corruption by crc check & retransmit.

The effect replaces 1-3 bytes in a websocket transmission by NUL (0) bytes. The replacement occurs at random positions within the message, but always begins at a 4 byte aligned index.

To reproduce, open the browser console log, then do some full web UI reloads. The effect has a high chance to occur right after the initial websocket open event, as the webserver then does a full metrics transmission. It's not restricted to high volume transfers though. If leaving the log open and just idling the web session, it will occasionally occur even on single metric updates, see examples.

**Examples ('\0' replaced by '_' for readability):**

```
18:19:24.840 script.js?v=1546965931:303 WebSocket OPENED Event
18:19:26.631 script.js?v=1546965931:313 WebSocket msg: SyntaxError: Unexpected token _ in JSON at position 708: {"metrics":{"v.e.handbrake":false,"v.e.headlights":false,"v.e.heating":false,"v.e.hvac":false,"v.e.locked":false,"v.e.on":false,"v.e.parktime":1352,"v.e.temp":0,"v.e.throttle":0,"v.e.valet":false,"v.i.temp":0,"v.m.rpm":0,"v.m.temp":0,"v.p.altitude":314.7,"v.p.direction":170.2,"v.p.gpshdop":1,"v.p.gpslock":true,"v.p.gpsmode":"AA","v.p.gpsspeed":0,"v.p.latitude":51.3024,"v.p.longitude":7.39,"v.p.odometer":0,"v.p.satcount":8,"v.p.speed":0,"v.p.trip":0,"v.tp.fl.p":0,"v.tp.fl.t":0,"v.tp.fr.p":0,"v.tp.fr.t":0,"v.tp.rl.p":0,"v.tp.rl.t":0,"v.tp.rr.p":0,"v.tp.rr.t":0,"v.type":"RT","v.vin":"","xrt.b.c.01.volt.act":0,"xrt.b.c.01.volt.max":0,"xrt.b.c.01.volt.maxdev":0,"xrt.b.c.01.volt.min":0,"xrt.b.c.02.volt.ac___0,"xrt.b.c.02.volt.max":0,"xrt.b.c.02.volt.maxdev":0,"xrt.b.c.02.volt.min":0,"xrt.b.c.03.volt.act":0,"xrt.b.c.03.volt.max":0,"xrt.b.c.03.volt.maxdev":0,"xrt.b.c.03.volt.min":0,"xrt.b.c.04.volt.act":0,"xrt.b.c.04.volt.max":0,"xrt.b.c.04.volt.maxdev":0,"xrt.b.c.04.volt.min":0,"xrt.b.c.05.volt.act":0,"xrt.b.c.05.volt.max":0}}
ws.onmessage @ script.js?v=1546965931:313
18:19:26.631 script.js?v=1546965931:313 WebSocket msg: SyntaxError: Unexpected token _ in JSON at position 304: {"metrics":{"xrt.b.c.05.volt.maxdev":0,"xrt.b.c.05.volt.min":0,"xrt.b.c.06.volt.act":0,"xrt.b.c.06.volt.max":0,"xrt.b.c.06.volt.maxdev":0,"xrt.b.c.06.volt.min":0,"xrt.b.c.07.volt.act":0,"xrt.b.c.07.volt.max":0,"xrt.b.c.07.volt.maxdev":0,"xrt.b.c.07.volt.min":0,"xrt.b.c.08.volt.act":0,"xrt.b.c.08.volt.ma___0,"xrt.b.c.08.volt.maxdev":0,"xrt.b.c.08.volt.min":0,"xrt.b.c.09.volt.act":0,"xrt.b.c.09.volt.max":0,"xrt.b.c.09.volt.maxdev":0,"xrt.b.c.09.volt.min":0,"xrt.b.c.10.volt.act":0,"xrt.b.c.10.volt.max":0,"xrt.b.c.10.volt.maxdev":0,"xrt.b.c.10.volt.min":0,"xrt.b.c.11.volt.act":0,"xrt.b.c.11.volt.max":0,"xrt.b.c.11.volt.maxdev":0,"xrt.b.c.11.volt.min":0,"xrt.b.c.12.volt.act":0,"xrt.b.c.12.volt.max":0,"xrt.b.c.12.volt.maxdev":0,"xrt.b.c.12.volt.min":0,"xrt.b.c.13.volt.act":0,"xrt.b.c.13.volt.max":0,"xrt.b.c.13.volt.maxdev":0,"xrt.b.c.13.volt.min":0,"xrt.b.c.14.volt.act":0,"xrt.b.c.14.volt.max":0,"xrt.b.c.14.volt.maxdev":0,"xrt.b.c.14.volt.min":0,"xrt.b.c.15.volt.act":0,"xrt.b.c.15.volt.max":0,"xrt.b.c.15.volt.maxdev":0}}
ws.onmessage @ script.js?v=1546965931:313
18:24:27.327 script.js?v=1546965931:313 WebSocket msg: SyntaxError: Unexpected token _ in JSON at position 20: {"metrics":{"v.i.tem___9}}
ws.onmessage @ script.js?v=1546965931:313
18:24:58.816 script.js?v=1546965931:313 WebSocket msg: SyntaxError: Unexpected token _ in JSON at position 20: {"metrics":{"v.i.tem___9}}
ws.onmessage @ script.js?v=1546965931:313
18:25:03.521 script.js?v=1546965931:313 WebSocket msg: SyntaxError: Unexpected token _ in JSON at position 20: {"metrics":{"v.i.tem___9}}
ws.onmessage @ script.js?v=1546965931:313
18:40:26.256 script.js?v=1546965931:313 WebSocket msg: SyntaxError: Unexpected token _ in JSON at position 100: {"metrics":{"m.monotonic":2613,"m.net.sq":-83,"m.time.utc":1546969226,"v.e.parktime":902,"v.p.gpshdo___1,"v.p.satcount":7}}
ws.onmessage @ script.js?v=1546965931:313
18:44:21.178 script.js?v=1546965931:313 WebSocket msg: SyntaxError: Unexpected token _ in JSON at position 88: {"metrics":{"m.monotonic":2848,"m.time.utc":1546969461,"v.e.parktime":1137,"v.p.gpshdop"_1,"v.p.satcount":6}}
…
10:51:43.479 script.js?v=1546965931:313 WebSocket msg: SyntaxError: Unexpected token _ in JSON at position 48: {"metrics":{"m.monotonic":60883,"v.b.12v.current__1,"v.b.coulomb.used":0.72551,"v.b.current":0.5,"v.b.energy.used":0.0373578,"v.b.power":0.0256,"v.c.temp":10,"v.e.parktime":59172,"xrt.p.stats.cst.used":0.0373578}}
```

**Analysis:**

```
SyntaxError: Unexpected token '\0' in JSON at position 708:
  22 78 72 74 2e 62 2e 63  2e 30 32 2e 76 6f 6c 74  |"xrt.b.c.02.volt|
  2e 61 63 00 00 00 30 2c  22 78 72 74 2e 62 2e 63  |.ac...0,"xrt.b.c|
  2e 30 32 2e 76 6f 6c 74  2e 6d 61 78 22 3a 30 2c  |.02.volt.max":0,|

SyntaxError: Unexpected token '\0' in JSON at position 304:
  30 2c 22 78 72 74 2e 62  2e 63 2e 30 38 2e 76 6f  |0,"xrt.b.c.08.vo|
  6c 74 2e 6d 61 00 00 00  30 2c 22 78 72 74 2e 62  |lt.ma...0,"xrt.b|
  2e 63 2e 30 38 2e 76 6f  6c 74 2e 6d 61 78 64 65  |.c.08.volt.maxde|

SyntaxError: Unexpected token '\0' in JSON at position 20:
  20 32 30 3a 20 7b 22 6d  65 74 72 69 63 73 22 3a  | 20: {"metrics":|
  7b 22 76 2e 69 2e 74 65  6d 00 00 00 39 7d 7d 0a  |{"v.i.tem...9}}.|

SyntaxError: Unexpected token '\0' in JSON at position 100:
  6b 74 69 6d 65 22 3a 39  30 32 2c 22 76 2e 70 2e  |ktime":902,"v.p.|
  67 70 73 68 64 6f 00 00  00 31 2c 22 76 2e 70 2e  |gpshdo...1,"v.p.|
  73 61 74 63 6f 75 6e 74  22 3a 37 7d 7d 0a 77 73  |satcount":7}}.ws|

SyntaxError: Unexpected token '\0' in JSON at position 88:
  2e 65 2e 70 61 72 6b 74  69 6d 65 22 3a 31 31 33  |.e.parktime":113|
  37 2c 22 76 2e 70 2e 67  70 73 68 64 6f 70 22 00  |7,"v.p.gpshdop".|
  31 2c 22 76 2e 70 2e 73  61 74 63 6f 75 6e 74 22  |1,"v.p.satcount"|

SyntaxError: Unexpected token '\0' in JSON at position 48:
  74 6f 6e 69 63 22 3a 36  30 38 38 33 2c 22 76 2e  |tonic":60883,"v.|
  62 2e 31 32 76 2e 63 75  72 72 65 6e 74 00 00 31  |b.12v.current..1|
  2c 22 76 2e 62 2e 63 6f  75 6c 6f 6d 62 2e 75 73  |,"v.b.coulomb.us|
```

**Message generation:**

The websocket message is created here:
https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/blob/master/vehicle/OVMS.V3/components/ovms_webserver/src/ovms_websockethandler.cpp#L132

I don't think anything can go wrong here. Also, as the message is given to mongoose in one piece, no mongoose mbuf or realloc issue can possibly cause this effect.

dexterbg, 2019-01-09:
The effect is still present when building with the latest esp-idf upstream/master (commit e931fe9f578a7ce6b6aa28261750bb1344509e15).

dexterbg, 2019-01-24:
[string-memory-corruption-2.txt](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/files/2793355/string-memory-corruption-2.txt)
This is an example test session with my workbench module.

Please verify this effect with your modules and report your results.

```
Synopsis: test string <loopcnt> <mode>

The command builds a std::string of at least 1024 chars repeatedly by
appending small pieces from different sources. If it detects a NUL
byte corruption it stops and outputs the string so far.

Source can be selected by mode:
1 = walk through metrics, add AsJSON()
2 = walk through metrics, add AsString()
3 = walk through metrics, add m_name
4 = add constant MyConfig.GetParamValue() result
5 = add constant local c string
6 = add constant local std::string

Loop counts of 500 will do for modes 1 & 2, these usually run into
corruptions very frequently. Modes 3-6 need to be called with high
loop count, e.g. 10000, and possibly need multiple runs to encounter
a corruption.
```

slcasner, 2019-01-25:
On my bench OVMS 3.0 module there were no crashes for any of the tests.  Perhaps you've made a similar observation.  That module has no SPIRAM and is configured with modem off and a pared-down configuration.

My in-car OVMS 3.1 module the first test 1 with 500 only had a length change.  Repeating that finished with success.  A third try with 1000 hit enough errors to abort.  A test 2 with 1000 hit several errors but not enough to abort.  Output attached.
[string-memory-corruption.txt](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/files/2794507/string-memory-corruption.txt)

dexterbg, 2019-01-25:
That looks like your 3.1 module also has the corruptions. Both my 3.1 modules show the effect. Length changes are expected for metrics changes during the run with modes 1 & 2.

I can confirm that my 3.0 module, as yours, seems to don't have the issue.

I have just disabled SPIRAM for my 3.1 build and can now confirm my 3.1 module running without SPIRAM also does not show the effect.

That points strongly to the SPIRAM implementation now.

markwj, 2019-01-25:
I tried with a minimal (autoinit disabled), and still get this:

> OVMS# test string 500 2
> #1: stdlen = 1038
> 4187292OVMS WIFI BLE BT cores=2 rev=ESP32/114143.2.001-18-gac0ae29b/factory/edge (build idf v3.1-dev-2835-g151269458-dirty Jan 25 2019 20:43:09)012.64187292OVMS WIFI BLE BT cores=2 rev=ESP32/114143.2.001-18-gac0ae29b/factory/edge (build idf v3.1-dev-2835-g151269458-dirty Jan 25 2019 20:43:09)012.64187292OVMS WIFI BLE BT cores=2 rev=ESP32/114143.2.001-18-gac0ae29b/factory/edge (build idf v3.1-dev-2835-g151269458-dirty Jan 25 2019 20:43:09)012.64187292OVMS WIFI BLE BT cores=2 rev=ESP32/114143.2.001-18-gac0ae29b/factory/edge (build idf v3.1-dev-2835-g151269458-dirty Jan 25 2019 20:43:09)012.64187292OVMS WIFI BLE BT cores=2 rev=ESP32/114143.2.001-18-gac0ae29b/factory/edge (build idf v3.1-dev-2835-g151269458-dirty Jan 25 2019 20:43:09)012.64187292OVMS WIFI BLE BT cores=2 rev=ESP32/114143.2.001-18-gac0ae29b/factory/edge (build idf v3.1-dev-2835-g151269458-dirty Jan 25 2019 20:43:09)012.64187292OVMS WIFI BLE BT cores=2 rev=ESP32/114143.2.001-18-gac0ae29b/factory/edge (build idf v3.1-dev-2835-g151269458-dirty Jan 25 2019 20:43:09)
> 
> loop #155: corruption detected at pos=446 size=447
> 4187292OVMS WIFI BLE BT cores=2 rev=ESP32/118143.2.001-18-gac0ae29b/factory/edge (build idf v3.1-dev-2835-g151269458-dirty Jan 25 2019 20:43:09)012.64187292OVMS WIFI BLE BT cores=2 rev=ESP32/118143.2.001-18-gac0ae29b/factory/edge (build idf v3.1-dev-2835-g151269458-dirty Jan 25 2019 20:43:09)012.64187292OVMS WIFI BLE BT cores=2 rev=ESP32/118143.2.001-18-gac0ae29b/factory/edge (build idf v3.1-dev-2835-g151269458-dirty Jan 25 2019 20:43:09)012.
> 
> loop #228: corruption detected at pos=893 size=894
> 4186940OVMS WIFI BLE BT cores=2 rev=ESP32/120143.2.001-18-gac0ae29b/factory/edge (build idf v3.1-dev-2835-g151269458-dirty Jan 25 2019 20:43:09)012.64186940OVMS WIFI BLE BT cores=2 rev=ESP32/120143.2.001-18-gac0ae29b/factory/edge (build idf v3.1-dev-2835-g151269458-dirty Jan 25 2019 20:43:09)012.64186940OVMS WIFI BLE BT cores=2 rev=ESP32/120143.2.001-18-gac0ae29b/factory/edge (build idf v3.1-dev-2835-g151269458-dirty Jan 25 2019 20:43:09)012.64186940OVMS WIFI BLE BT cores=2 rev=ESP32/120143.2.001-18-gac0ae29b/factory/edge (build idf v3.1-dev-2835-g151269458-dirty Jan 25 2019 20:43:09)012.64186940OVMS WIFI BLE BT cores=2 rev=ESP32/120143.2.001-18-gac0ae29b/factory/edge (build idf v3.1-dev-2835-g151269458-dirty Jan 25 2019 20:43:09)012.64186940OVMS WIFI BLE BT cores=2 rev=ESP32/120143.2.001-18-gac0ae29b/factory/edge (build idf v3.1-dev-2835-g151269458-dirty Jan 25 2019 20:43:09)012.
> 
> finished

Corruption for me ALWAYS happens at the decimal point of those float to string conversions. Different places, but always a float to string. Those use std::ostringstream which is presumably memory allocation intensive.

If I change the for() loop to be for (m = StandardMetrics.ms_m_version ; msg.size() < 1024 ;), I can’t get the problem to appear.

I know our std::string implementation has small string support (so strings < 16 bytes or so are stored directly, and bigger than that requires a dynamic allocation).

dexterbg, 2019-01-25:
See my log especially on modes 5 & 6 -- no conversion or allocations involved, just appending to the string having enough capacity.

I currently try to reproduce this with a vanilla project to create a test case for Espressif.

dexterbg, 2019-01-25:
No luck so far reproducing this outside our framework. My test project:
[test-memory-corruption.zip](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/files/2797634/test-memory-corruption.zip)

Any ideas on what to test next?

slcasner, 2019-01-25:
Do we need to have (some of) the strings be in SPIRAM?

slcasner, 2019-01-26:
Would it be feasible to try maximally removing components from the OVMS configuration and turning off configurable functions to see if that stops the errors, then add them back in incrementally to find what is significant?

dexterbg, 2019-01-26:
I'm currently trying to go the other direction, disabling options one by one. It seems just disabling the "aggressive SPIRAM" option is sufficient to stop the error. Can you confirm that?

dexterbg, 2019-01-26:
I've pretty much narrowed it down to wifi.

I now have a build with just the bare wifi component, no mongoose based components. Running this build I can start other components, load a vehicle etc. without any issues. The bug does not show when I keep wifi off. As soon as I start wifi, the bug begins to show. The wifi mode is irrelevant, and starting just wifi is sufficient.

Stopping wifi again after starting reduces the bug frequency, but it still shows. A reboot is necessary to get rid of it.

As our wifi class is already InternalRamAllocated, it seems the wifi blob contains some C++ code that uses our "aggressive SPIRAM" new() overrides and runs into trouble with that. I will now try to isolate the wifi blob from our overrides.

slcasner, 2019-01-26:
Good progress.

slcasner, 2019-01-26:
As you say, if other projects are overriding std::new to take advantage of SPIRAM, then it is surprising that none have discovered this problem.  There must be something unsafe in how we are doing it?

dexterbg, 2019-01-27:
Adding wifi with APSTA mode to the test project didn't produce the bug, and I just got a corruption without starting wifi on the reduced OVMS build, so it was just a correlation by chance.

dexterbg, 2019-01-27:
I've nailed it :-)

Wifi is indeed a factor, but only in terms of stress / load. The bug didn't show in my test project because I was running the test on core #0. Apparently, the bug only shows on core #1. And it's definitely in esp-idf, not in our code.

Mark, Steve, please try the new version of my test project:
[test-memory-corruption.zip](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/files/2800915/test-memory-corruption.zip)

A test run log (also included in the ZIP):
[test-log-20190128.txt](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/files/2800916/test-log-20190128.txt)

If you can confirm the bug shows with this test project on your modules as well, I'll open an issue with Espressif.

dexterbg, 2019-01-27:
Oops, take care not to share your logs without removing your wifi passwords…

markwj, 2019-01-29:
I had some problems with:

```
test-memory-corruption/main/wifi_client.cpp:120:52: error: 'ESP_ERR_NVS_NEW_VERSION_FOUND' was not declared in this scope
     if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
                                                    ^
```

Anyway, commenting that out let it build.

> I (10306) test_task: testing mode 1 loopcnt 1000
> E (10316) test_task: loop #1: corruption detected at pos=497 size=498
> 4348864OVMS WIFI BLE BT cores=2 rev=ESP32/1WLAN-214677-52wifi103.2.001-22-g14d16854-dirty/factory/edge (build idf v3.2-dev-1419-g5afc1b0cb-dirty Jan 27 2019 15:08:23)4348864OVMS WIFI BLE BT cores=2 rev=ESP32/1WLAN-214677-52wifi103.2.001-22-g14d16854-dirty/factory/edge (build idf v3.2-dev-1419-g5afc1b0cb-dirty Jan 27 2019 15:08:23)4348864OVMS WIFI BLE BT cores=2 rev=ESP32/1WLAN-214677-52wifi103.2.001-22-g14d16854-dirty/factory/edge (build idf v3.2-dev-1419-g5afc1b0cb-dirty Jan 27 2019 15:08:23
> 
> #1: size=498, data @0x3f901ca0
> 4348864OVMS WIFI BLE BT cores=2 rev=ESP32/1WLAN-214677-52wifi103.2.001-22-g14d16854-dirty/factory/edge (build idf v3.2-dev-1419-g5afc1b0cb-dirty Jan 27 2019 15:08:23)4348864OVMS WIFI BLE BT cores=2 rev=ESP32/1WLAN-214677-52wifi103.2.001-22-g14d16854-dirty/factory/edge (build idf v3.2-dev-1419-g5afc1b0cb-dirty Jan 27 2019 15:08:23)4348864OVMS WIFI BLE BT cores=2 rev=ESP32/1WLAN-214677-52wifi103.2.001-22-g14d16854-dirty/factory/edge (build idf v3.2-dev-1419-g5afc1b0cb-dirty Jan 27 2019 15:08:23
> 
> E (10456) test_task: loop #3: corruption detected at pos=497 size=498
> 4348864OVMS WIFI BLE BT cores=2 rev=ESP32/1WLAN-214677-52wifi103.2.001-22-g14d16854-dirty/factory/edge (build idf v3.2-dev-1419-g5afc1b0cb-dirty Jan 27 2019 15:08:23)4348864OVMS WIFI BLE BT cores=2 rev=ESP32/1WLAN-214677-52wifi103.2.001-22-g14d16854-dirty/factory/edge (build idf v3.2-dev-1419-g5afc1b0cb-dirty Jan 27 2019 15:08:23)4348864OVMS WIFI BLE BT cores=2 rev=ESP32/1WLAN-214677-52wifi103.2.001-22-g14d16854-dirty/factory/edge (build idf v3.2-dev-1419-g5afc1b0cb-dirty Jan 27 2019 15:08:23
> 
> E (10596) test_task: loop #6: corruption detected at pos=497 size=498
> 4348864OVMS WIFI BLE BT cores=2 rev=ESP32/1WLAN-214677-52wifi103.2.001-22-g14d16854-dirty/factory/edge (build idf v3.2-dev-1419-g5afc1b0cb-dirty Jan 27 2019 15:08:23)4348864OVMS WIFI BLE BT cores=2 rev=ESP32/1WLAN-214677-52wifi103.2.001-22-g14d16854-dirty/factory/edge (build idf v3.2-dev-1419-g5afc1b0cb-dirty Jan 27 2019 15:08:23)4348864OVMS WIFI BLE BT cores=2 rev=ESP32/1WLAN-214677-52wifi103.2.001-22-g14d16854-dirty/factory/edge (build idf v3.2-dev-1419-g5afc1b0cb-dirty Jan 27 2019 15:08:23
> 
> E (10666) test_task: loop #7: corruption detected at pos=829 size=830
> 4348864OVMS WIFI BLE BT cores=2 rev=ESP32/1WLAN-214677-52wifi103.2.001-22-g14d16854-dirty/factory/edge (build idf v3.2-dev-1419-g5afc1b0cb-dirty Jan 27 2019 15:08:23)4348864OVMS WIFI BLE BT cores=2 rev=ESP32/1WLAN-214677-52wifi103.2.001-22-g14d16854-dirty/factory/edge (build idf v3.2-dev-1419-g5afc1b0cb-dirty Jan 27 2019 15:08:23)4348864OVMS WIFI BLE BT cores=2 rev=ESP32/1WLAN-214677-52wifi103.2.001-22-g14d16854-dirty/factory/edge (build idf v3.2-dev-1419-g5afc1b0cb-dirty Jan 27 2019 15:08:23)4348864OVMS WIFI BLE BT cores=2 rev=ESP32/1WLAN-214677-52wifi103.2.001-22-g14d16854-dirty/factory/edge (build idf v3.2-dev-1419-g5afc1b0cb-dirty Jan 27 2019 15:08:23)4348864OVMS WIFI BLE BT cores=2 rev=ESP32/1WLAN-214677-52wifi103.2.001-22-g14d16854-dirty/factory/edge (build idf v3.2-dev-1419-g5afc1b0cb-dirty Jan 27 2019 15:08:23
> 
> E (10846) test_task: loop #10: corruption detected at pos=497 size=498
> 4348864OVMS WIFI BLE BT cores=2 rev=ESP32/1WLAN-214677-52wifi103.2.001-22-g14d16854-dirty/factory/edge (build idf v3.2-dev-1419-g5afc1b0cb-dirty Jan 27 2019 15:08:23)4348864OVMS WIFI BLE BT cores=2 rev=ESP32/1WLAN-214677-52wifi103.2.001-22-g14d16854-dirty/factory/edge (build idf v3.2-dev-1419-g5afc1b0cb-dirty Jan 27 2019 15:08:23)4348864OVMS WIFI BLE BT cores=2 rev=ESP32/1WLAN-214677-52wifi103.2.001-22-g14d16854-dirty/factory/edge (build idf v3.2-dev-1419-g5afc1b0cb-dirty Jan 27 2019 15:08:23
> 
> too many errors, abort.

Is that what you need?

dexterbg, 2019-01-29:
Yes, perfect.

It still puzzles me we're the first to notice this. I think I'll add another test using a directly allocated memory buffer instead of the std::string buffer to see if the bug only occurs with std::string.

dexterbg, 2019-01-29:
The test using a simple char buffer does not show the bug, so this seems to be only affecting std::string, but possibly other libstdc++ components as well.

markwj, 2019-01-30:
Test changing CONFIG_SPIRAM_SIZE to 2MB had no impact.

However, adding a 2MB SPIRAM allocation right at the start seems to workaround the problem (both in the test replication code test-memory-corruption, and in the main OVMS head):

`void* big2mb = heap_caps_malloc(1024*1024*2, MALLOC_CAP_SPIRAM);`

That allocation is at 0x3f800028.

dexterbg, 2019-01-30:
Confirmed, and running the test project with that 2 MB allocation and also starting the test task on core 0 shows the corruptions again. It seems core 0 can only work reliably with the lower 2 MB and core 1 only with the higher 2 MB.

dexterbg, 2020-07-16:
Issue solved, PSRAM fix is now included in the official toolchain & esp-idf release, and changes are merged.
https://github.com/espressif/esp-idf/issues/2892#issuecomment-658327913

closed 2020-07-16 by dexterbg