openvehicles/Open-Vehicle-Monitoring-System-3 #201: Nissan LEAF SOH shows 108%

imperialjoy, 2019-02-07:


imperialjoy, 2019-02-07:
After updating to (3.2.001/ota_1/main (build idf v3.1-dev-2835-g1512694 Jan 19 2019 13:27:14)) SOH shows ovms//LEAF/metric/v/b/soh 108.502.  How is this possible?  LeafSpy reports 90.71% SOH which seems more reasonable.  This is a 2016 30kwh model, with the required parameters set in the config.

caederus-ovms, 2019-02-23:
Could try running these shell commands and post the results?
> config list xnl
> metrics list soh

glynhudson, 2019-03-08:
Leafspy also returns over 100% SOH values for the 30kwh leaf. Do you know what SOC leafspy returns for your car? Have you had the latest BMS battery update? This is supposed to fix SOH calculation.

imperialjoy, 2019-03-26:
This seems to have come almost right by itself.  I didn't change anything.  Leafspy reports SOC 42.3% and the dash says 40%.  Leafspy SOH is 88.16% and ovms reports 88.6958%.  The output of those two commands is: 

OVMS# config list xnl
xnl (readable writeable)
  kmPerKWh: 6.1
  maxGids: 356
  modelyear: 2016
  newCarAh: 79

OVMS# metrics list soh
v.b.soh                                  88.6958%

dalathegreat, 2021-12-26:
@dexterbg please close this issue, no longer an issue

closed 2021-12-26 by dexterbg