openvehicles/Open-Vehicle-Monitoring-System-3 #213: OVMS V2 Protocol SOC Python Example?

glynhudson, 2019-03-11:
Hi, I'm developing a smart charging scheduling interface for [OpenEVSE](https://openevse.com) based charging stations. I would like to integrate with OVMS to obtain SOC data from a connected EV. 

I've taken a look at the [OVMS V2 protocol pdf](https://github.com/openvehicles/Open-Vehicle-Monitoring-System/blob/master/docs/OVMS_Protocol.pdf), it does not look easy to implement the V2 protocol. Is there a simple python example available for how to retrieve a metric like SOC from OVMS? Either locally via Wi-Fi or remotely via OVMS server? 

I'm happy to give discount on an OpenEVSE charging station via OpenEnergyMonitor.org store if anyone can help with this. Thank you.

caederus-ovms, 2019-03-11:
Have you considered using MQTT with the V3 server?  That's really easy to use; just use any MQTT client to subscribe to api.openvehicles.com (or your own server), topic ovms/$user/$vehicle_id/metric/v/b/soc

glynhudson, 2019-03-11:
We're currently using V3 MQTT for testing, however we figured it would be more secure and easier to keep the different users separate if the used V2 server. We also wanted to make the OVSM setup process as easy as possible since there will be non-technical users of the system.

glynhudson, 2019-03-13:
Ah just found a very helpful post by @dexterbg on one of my previous issues: https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/issues/178

His dexters-web OVMS server has a really great REST API which works over SSL 👍 

http://lists.openvehicles.com/pipermail/ovmsdev/2018-July/005352.html

This works really great, I've been able to obtain SOC data for our application: 
https://github.com/emoncms/demandshaper/issues/7

closed 2019-03-13 by glynhudson