openvehicles/Open-Vehicle-Monitoring-System-3 #221: SOC Fail Opel Ampera

liebefuroscar, 2019-04-08:
I have last versión of Edge OTA and I see 0% on my SOC everytime.
Since it has few functions, please keep the ones that work.
My versión is 3.2.001-176
Thanks

dexterbg, 2019-05-07:
Here's another user report:

> The latest EDGE versions for Opel Ampera are not showing any vehicle data. GPS position only

@goev I think this is related to your latest changes to the OBD processing. Can you have a look?

liebefuroscar, 2019-05-07:
How can I check it?

goev, 2019-05-08:
Ah... I'm sorry! Fortunately, I'm pretty sure I have a solution. 

The Read Data By Identifier (VEHICLE_POLL_TYPE_OBDIIEXTENDED / $22) seems to give both single frame and multi-frame response, unlike VEHICLE_POLL_TYPE_OBDIIVEHICLE and VEHICLE_POLL_TYPE_OBDIIGROUP.

This is what I have found: 

Single frames have 0 in the most significant nibble on byte 0 and the frame size in least significant nibble of that byte. Response Id($62) in byte 1. These are the ones used by the Ampera and Think, and I've also found responses like that from at least one ECU on the Kia Niro too.

Multi-frame responses have 1 or 2 in the most significant nibble and frame size (12 bit) in the least significant nibble of byte 0 AND the whole byte 1.  Response id($62) in byte 2. These are used heavily on the Kia Niro (and Hyundai Kona and Ioniq).

Unfortunately I don't have time to do more until next week, but I believe it is just a matter of "else-ing" the old code after the my new code after "case VEHICLE_POLL_TYPE_OBDIIEXTENDED" in vehicle.PollerReceive.

There are both types of responses available on the Kia Niro, so I should be able to test both.

dexterbg, 2019-05-08:
A potential fix is in 3.2.001-206-g0f4c05b8, please test & report.

liebefuroscar, 2019-05-09:
It´s working again with 3.2.001-206-g0f4c05b8.
Thanks so much 
[image]
[image]
[image]

closed 2019-05-09 by dexterbg