openvehicles/Open-Vehicle-Monitoring-System-3 #229: Updated Vehicle Support Table

glynhudson, 2019-05-25:
I'm working on making an updated vehicle support table for OVMS. I'm aware there is a table at the bottom of the Google Doc Advanced User Guide. However, that table is very technical and more suited to developers interesting in knowing which metrics are available. There is also an old compatibility table on the [OVMS website here](https://www.openvehicles.com/vehiclesupport) which could do with updating.

[image]

I only have experience using OVMS with Nissan LEAF / e-NV200, I've tried to compile the table based on the info available to me. Please could you help me fill in the gaps and verify the features. 

The table can be edited using the following google doc link, once complete I'm happy to donate the table to OVMS for use on website etc: https://docs.google.com/document/d/1ZPC8Rh2OVGgT-EPuN3xdZsomhk1JA0sQ_WTTxTIfS9I/edit?usp=sharing

glynhudson, 2019-05-29:
@markwj has created a per vehicle supported support table in the 'Vehicle Documentation' section of the user guide. These tables should be completed instead of my table. https://docs.google.com/document/d/16JrXR7rybp-18DrEoeh1rg6GqQT_jVBvhaXh2oEWRHw/edit#heading=h.u42it5lxi2lo

Currently, the following compatibility tables needs completing Fiat 500e, Kia e-niro, Kia Soul, Mitsubishi i-Miev, Renault Twizy, SmartED, ThinkCity, Renault Twizy, ZEVA BMS,  Volt/Ampera

@markwj has completed tables for Tesla Roadster, Models S and ODB2
I've completed the Nissan LEAF table.

KommyKT, 2019-05-30:
I stearted creating the Mitsubishi i-MiEV documnetation:
https://docs.google.com/document/d/1YLk4rDLQjfmrNN2RyDifydsb1I9Y2E8pVKulJuatqao/edit?usp=sharing

Glyn Hudson <notifications@github.com> ezt írta (időpont: 2019. máj. 30.,
Cs, 0:47):

> @markwj <https://github.com/markwj> has created a per vehicle supported
> support table in the 'Vehicle Documentation' section of the user guide.
> These tables should be completed instead of my table. [...]

glynhudson, 2019-06-03:
>I stearted creating the Mitsubishi i-MiEV documnetation:

Great. I hope you don't mind I copied your documentation into the advanced user guide: 

https://docs.google.com/document/d/16JrXR7rybp-18DrEoeh1rg6GqQT_jVBvhaXh2oEWRHw/edit#heading=h.p7xmrzdc2v8l

closed 2019-12-10 by markwj

glynhudson, 2019-12-11:
Docs for the new excellent Read the Docs site are hosted and revision controlled on github. Chances to the docs should be made as pull requests if possible. As an example here is the roadster docs:

http://docs.openvehicles.com/en/latest/components/vehicle_teslaroadster/docs/index.html

Source: 
https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/blob/master/vehicle/OVMS.V3/components/vehicle_teslaroadster/docs/index.rst