openvehicles/Open-Vehicle-Monitoring-System-3 #231: Nissan Leaf: support LOCK, UNLOCK

amralomari, 2019-05-27:
Support Lock/Unlock for Nissan leaf

glynhudson, 2019-05-27:
I'm almost certain that the Nissan leaf does not support remote lock / unlock via CAN bus commands therefore this is impossible to implement. I believe the key authentication and lock / unlock is a separate hardware module.

dalathegreat, 2019-07-31:
glynhudson is correct, this issue should be closed

glynhudson, 2020-01-16:
A user has reported that lock / unlock works with a 30kWh Nissan LEAF: https://twitter.com/smallrougeone/status/1217554916513927168

I didn't think this was possible. Could anyone else test unlock on a 30kwh leaf? It doesnt work on my 2014 24kwh leaf or env200.

glynhudson, 2020-01-16:
**UPDATE:** 

It's been confirmed that OVMS remote Lock/Unlock works great on a 30kWh LEAF, I've seen a video of it working :+1: 

It still does not work on my 2014 24kWh LEAF or 2015 24kWh e-NV200.

ltjessem, 2020-05-05:
Do we know why it works on the 30? Might be worth looking into, could be a simple fix.

Alenux55, 2020-07-21:
> **UPDATE:**
> 
> It's been confirmed that OVMS remote Lock/Unlock works great on a 30kWh LEAF, I've seen a video of it working 👍
> 
> It still does not work on my 2014 24kWh LEAF or 2015 24kWh e-NV200.

Seen some others on the forum claiming the same. To your knowledge, the code has not been modified to make this work correctt?

mjkapkan, 2020-07-21:
Have not seen any recent modifications to lock/unlock part. It migh be worth to take a deeper look. The Leaf has automatic lock/unlock when reaching certain speed or engaging parking break. Might be worth sniffing the traffic on the older gen models to check what’s sent.

Alenux55, 2020-07-31:
Supposedly it should only work if the CAR-CAN is active. At the end of the second page [here](https://www.mynissanleaf.com/viewtopic.php?f=44&t=16358&start=10):

>I have a test app I put together a few weeks ago that locks and unlocks the doors through the CAN bus.
>However, as has been stated above the CAN door lock function also enables the alarm. But the CAN unlock function only unlocks the doors and leaves the alarm active. So when I open the door the alarm goes off.
>Another short coming is that the CAR-CAN must be active before the lock/unlock CAN messages are sent for the messages to be seen. There is a short window after the Leaf is turned off and when you plug the charge cord in when the CAN is active. So in most cases the Leaf is not in a position to even hear the command.
>Having wires go to the PTS (Push to Start) button would allow you to wake up the CAN. Not sure if that disables the alarm too.

I tried a quick test: turned on the car, drove around a bit, then turned off the car and ran both the door lock and unlock commands to see what would happen, which was unfortunately nothing. Haven't done any traffic sniffing yet, that will be the next step.

dalathegreat, 2020-08-01:
What about sending the wakeup CAN messages on the bus before sending the unlock messages? This should wake the modules so that they can understand what's going on.

Eg. on EV-CAN, you can send an empty 0x603 to simulate a wakeup from VCM, and an empty 0x679 to simulate the charger being plugged in.

Alenux55, 2020-08-02:
So based on the data cable pinout ([1779000](https://www.fasttech.com/product/1779000-nissan-leaf-obd-ii-to-db9-data-cable-for-ovms)), the [schematic](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/blob/master/vehicle/hardware/v3.1/Mainboard_SCH_v3.1_20180305.pdf), and the Leaf CANbus decoding [forum post](https://www.mynissanleaf.com/viewtopic.php?t=4131), I gather that:

- CAN0 (can1 in firmware): EV-CAN
- CAN1 (can2 in firmware): Primary/CAR-CAN
- CAN2 (can3 in firmware): Unused

Is that correct? In that case then, I ran:

```
can can1 tx standard 0x603 0x0 
can can1 tx standard 0x679 0x0
lock [my pin]
```

Here's the response:

```
OVMS# can can1 tx standard 0x603 0x0
E (511903) can: can1: intr=11438 rxpkt=11114 txpkt=52 errflags=0x8040d9 rxerr=0 txerr=116 rxovr=0 txovr=0 txdelay=0 wdgreset=0 errreset=0
E (511903) can: can1: intr=11448 rxpkt=11114 txpkt=52 errflags=0x8040d9 rxerr=0 txerr=132 rxovr=0 txovr=0 txdelay=0 wdgreset=0 errreset=0
I (516223) webserver: HTTP POST /api/execute
I (516233) webcommand: HttpCommandStream[0x3f858394]: 3939776 bytes free, executing: can can1 tx standard 0x679 0x0
OVMS# can can1 tx standard 0x679 0x0
E (516233) can: can1: intr=11517 rxpkt=11114 txpkt=53 errflags=0x8040d9 rxerr=0 txerr=131 rxovr=0 txovr=0 txdelay=0 wdgreset=0 errreset=0
I (519513) webserver: HTTP POST /api/execute
I (519523) webcommand: HttpCommandStream[0x3f852394]: 3939976 bytes free, executing: lock 22518954
OVMS# lock [my pin]
Vehicle locked
I (519523) v-nissanleaf: RemoteCommandHandler
I (519523) v-nissanleaf: Sending Gen 2 Wakeup Frame
E (519533) can: can1: intr=11595 rxpkt=11114 txpkt=54 errflags=0x8040d9 rxerr=0 txerr=130 rxovr=0 txovr=0 txdelay=0 wdgreset=0 errreset=0
I (519633) v-nissanleaf: RemoteCommandTimer 24
I (519633) v-nissanleaf: OLD TCU on EV Bus
I (519633) v-nissanleaf: Lock Doors
I (519733) v-nissanleaf: RemoteCommandTimer 23
I (519733) v-nissanleaf: OLD TCU on EV Bus
I (519733) v-nissanleaf: Lock Doors
E (519733) can: can1: intr=11712 rxpkt=11152 txpkt=57 errflags=0x204c00 rxerr=0 txerr=127 rxovr=0 txovr=0 txdelay=0 wdgreset=0 errreset=0
I (519833) v-nissanleaf: RemoteCommandTimer 22
I (519833) v-nissanleaf: OLD TCU on EV Bus
I (519833) v-nissanleaf: Lock Doors
I (519933) v-nissanleaf: RemoteCommandTimer 21
I (519933) v-nissanleaf: OLD TCU on EV Bus
I (519933) v-nissanleaf: Lock Doors
I (520033) v-nissanleaf: RemoteCommandTimer 20
I (520033) v-nissanleaf: OLD TCU on EV Bus
I (520033) v-nissanleaf: Lock Doors
I (520133) v-nissanleaf: RemoteCommandTimer 19
I (520133) v-nissanleaf: OLD TCU on EV Bus
I (520133) v-nissanleaf: Lock Doors
I (520233) v-nissanleaf: RemoteCommandTimer 18
I (520233) v-nissanleaf: OLD TCU on EV Bus
I (520233) v-nissanleaf: Lock Doors
I (520333) v-nissanleaf: RemoteCommandTimer 17
I (520333) v-nissanleaf: OLD TCU on EV Bus
I (520333) v-nissanleaf: Lock Doors
I (520433) v-nissanleaf: RemoteCommandTimer 16
I (520433) v-nissanleaf: OLD TCU on EV Bus
I (520433) v-nissanleaf: Lock Doors
I (520533) v-nissanleaf: RemoteCommandTimer 15
I (520533) v-nissanleaf: OLD TCU on EV Bus
I (520533) v-nissanleaf: Lock Doors
I (520633) v-nissanleaf: RemoteCommandTimer 14
I (520633) v-nissanleaf: OLD TCU on EV Bus
I (520633) v-nissanleaf: Lock Doors
I (520733) v-nissanleaf: RemoteCommandTimer 13
I (520733) v-nissanleaf: OLD TCU on EV Bus
I (520733) v-nissanleaf: Lock Doors
I (520833) v-nissanleaf: RemoteCommandTimer 12
I (520833) v-nissanleaf: OLD TCU on EV Bus
I (520833) v-nissanleaf: Lock Doors
I (520933) v-nissanleaf: RemoteCommandTimer 11
I (520933) v-nissanleaf: OLD TCU on EV Bus
I (520933) v-nissanleaf: Lock Doors
I (521033) v-nissanleaf: RemoteCommandTimer 10
I (521033) v-nissanleaf: OLD TCU on EV Bus
I (521033) v-nissanleaf: Lock Doors
I (521133) v-nissanleaf: RemoteCommandTimer 9
I (521133) v-nissanleaf: OLD TCU on EV Bus
I (521133) v-nissanleaf: Lock Doors
I (521233) v-nissanleaf: RemoteCommandTimer 8
I (521233) v-nissanleaf: OLD TCU on EV Bus
I (521233) v-nissanleaf: Lock Doors
I (521333) v-nissanleaf: RemoteCommandTimer 7
I (521333) v-nissanleaf: OLD TCU on EV Bus
I (521333) v-nissanleaf: Lock Doors
I (521433) v-nissanleaf: RemoteCommandTimer 6
I (521433) v-nissanleaf: OLD TCU on EV Bus
I (521433) v-nissanleaf: Lock Doors
I (521533) v-nissanleaf: RemoteCommandTimer 5
I (521533) v-nissanleaf: OLD TCU on EV Bus
I (521533) v-nissanleaf: Lock Doors
I (521633) v-nissanleaf: RemoteCommandTimer 4
I (521633) v-nissanleaf: OLD TCU on EV Bus
I (521633) v-nissanleaf: Lock Doors
I (521733) v-nissanleaf: RemoteCommandTimer 3
I (521733) v-nissanleaf: OLD TCU on EV Bus
I (521733) v-nissanleaf: Lock Doors
I (521833) v-nissanleaf: RemoteCommandTimer 2
I (521833) v-nissanleaf: OLD TCU on EV Bus
I (521833) v-nissanleaf: Lock Doors
I (521933) v-nissanleaf: RemoteCommandTimer 1
I (521933) v-nissanleaf: OLD TCU on EV Bus
I (521933) v-nissanleaf: Lock Doors
I (522033) v-nissanleaf: RemoteCommandTimer 0
```

I notice in the Nissan Leaf [component](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/blob/master/vehicle/OVMS.V3/components/vehicle_nissanleaf/src/vehicle_nissanleaf.cpp), the Gen2 wakeup command is also used on the Gen1 models (i.e. it sends 0x68c regardless of the model), so that might explain the errors I'm seeing. Is there any sort of guide for debugging error messages anywhere?

Also looking at the "lock doors" command in the same file, I believe I should be recreating this command rather than typing "lock [my pin]" into the console. In that case, would the lock door command look like:

`can can1 tx standard 0x56e 0x60 0x80 0x00 0x00`

And the unlock door command look like:

`can can1 tx standard 0x56e 0x11 0x00 0x00 0x00`

Sorry for the confusion on my end, still wrapping my head around the command syntax and all that; this is my first time actually working with the CANbus.

Alenux55, 2020-08-03:
Realized I was typing the can commands incorrectly (i.e. there should be no 0x prefix). So then to lock the doors, I should be doing:

```
can can1 tx standard 603 00 
can can1 tx standard 679 00
can can1 tx standard 56e 60 80 00 00
```

And to unlock the doors:

```
can can1 tx standard 603 00 
can can1 tx standard 679 00
can can1 tx standard 56e 11 00 00 00
```

Neither of these command sets changed the door lock state

dalathegreat, 2020-08-03:
You are sending can0 wakeup commands on can1. So the CAN probably doesn't wake up.

Take a log of what can1 messages happen on your Leaf when it starts. In my database file I have these three

- 0x603
- 0x682
- 0x68C

But I am unsure which ones are really going on, since I haven't looked much into CAR-CAN yet.

glynhudson, 2021-06-17:
Just to add some more info to this thread;

2015 - 24kWh e-NV200 door unlocking using OVMS does NOT work 
2018 - 40kWh e-NV200 door unlocking works perfectly using OVMS 👍

meelash, 2022-11-30:
@dexterbg This issue was never reopened when new information was discovered and so it seems to have fallen by the way side. Subject to time constraints I will look into it further for my 2012 leaf, but I'm a newbie (albeit experience hardware and software engineer in general).

Regardless of progress or not on figuring out why it works, the documentation for leaf states with no reservations that it does work, so shouldn't that be updated?

dexterbg, 2022-11-30:
You're welcome to take care of that. Adding to the documentation is simple, and we're glad to help new developers get into the system.

andyorkney, 2025-09-13:
So brand new to OVMS had a mare of a time setting things up. Using apps commands are not working for me but in the console Door Commands for e-NV200:
	•	Lock: can can2 tx standard 56e 60 80 00 00
	•	Unlock: can can2 tx standard 56e 11 00 00 00
I know:
	1.	CAN ID: 56e (door control)
	2.	CAN Bus: CAN2 (not CAN1)
	3.	Ignition: Must be on for CAN commands to work
	4.	Commands work perfectly when sent directly
My OVMS issue is:
	•	The vehicle module is sending commands on CAN1 instead of CAN2
	•	The cfg_ev_request_port: 2 setting is being ignored
	•	is definitely a firmware bug that needs fixing or am I missing something?
	•	Vehicle: 2018 e-NV200
	•	Issue: cfg_ev_request_port: 2 ignored, commands sent on CAN1
	•	Working solution: Direct CAN2 commands work perfectly
	•	Lock: 56e 60 80 00 00
	•	Unlock: 56e 11 00 00 00

dexterbg, 2025-09-13:
No Leaf/e-NV owner, but from the code & docs:
- Did you set your model year accordingly? Model years >= 2016 tell the module to use CAN2.
- `cfg_ev_request_port` isn't the CAN port, it's the I/O port for the system activation request.

andyorkney, 2025-09-16:
OK I switched server hosts and updated the date of vehicle to 2018. Now all works except SoC in ABRP script. FYI change of date did not work until moved to turboserv host.