openvehicles/Open-Vehicle-Monitoring-System-3 #238: Kia Niro remote climate control?

glynhudson, 2019-06-26:
Currently, Kia Niro (and Kia Soul) do not support remote climate control. Is this due to a limitation of the car or could remote climate control feasibly be added in the future?

goev, 2019-06-26:
> What do you think @goev?

This is not supported because we simply don’t know how. It might be possible. 

Best regards,
Geir

Sendt fra min iPhone

closed 2019-08-05 by glynhudson

DrewCyber, 2021-03-09:
As I understood, Kia UVO is located inside multimedia unit (it has embedded e-sim).
I digged around and found that multimedia pinout has own CAN pins. Pinout details in both links below:
https://pinoutguide.com/Car-Stereo-Hyundai-Kia/Hyundai_Kona_2017-2019_He.shtml
https://pinoutguide.com/Car-Stereo-Hyundai-Kia/Kia_Niro_Audio_Video_Navi.shtml

(Connector B, pins 7 and 20)

@goev  Is it possible that preheating signals can be sent from the multimedia?
Can we try to trace OBD signals from multimedia side?  Does it make sense?
Thank you!