openvehicles/Open-Vehicle-Monitoring-System-3 #245: Vehicle::Ticker may be called before Vehicle is constructed

mjuhanne, 2019-08-07:
I was playing around and switching between vehicles using web UI, when crash occured. It seems Ticker1 of inheriting module can be called even before the Vehicle itself is properly initialized. In this cased **m_v_door_lock_fr of Kia Niro module** was referred even before it was constructed. It seems Ticker1 is registered in OvmsVehicle::OvmsVehicle(), but the OvmsVehicleKiaNiroEv::OvmsVehicleKiaNiroEv() constructor was not yet finished, when  OvmsVehicleKiaNiroEv::Ticker1 was being called via OvmsVehicle::VehicleTicker1...

This can be easily replicated with adding
`  vTaskDelay(1100 / portTICK_PERIOD_MS);  `
to the end of OvmsVehicle::OvmsVehicle() and switching to Kia Niro module.

We need some kind of lock mechanism in place to prevent Tickers before Vehicle is initialized. Seems pretty rare, but could happen.  

Debug trace:
``` I (674178) webserver: HTTP GET /assets/zones.json
I (681058) webserver: HTTP POST /cfg/vehicle
I (681058) v-twizy: Shutdown Renault Twizy vehicle module
I (681068) v-kianiroev: Kia Niro EV v1.0 vehicle module
OVMS# Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x402cf522  PS      : 0x00060130  A0      : 0x80194784  A1      : 0x3ffc0d50  
0x402cf522: OvmsMetricBool::AsBool(bool) at /Users/markojuhanne/esp/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/main/ovms_metrics.cpp:656

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000004  A5      : 0x00000004  
A6      : 0xffffffff  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x00000001  
A10     : 0xfffffff7  A11     : 0x3f40fb78  A12     : 0x3ffc0e74  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400930dc  LEND    : 0x4009310a  LCOUNT  : 0x00000000  
0x400930dc: memcpy at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memcpy.S:168

0x4009310a: memcpy at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memcpy.S:201

ELF file SHA256: 92b7d3ed0b11eb13addcfe759de797551659f1a3bcb9f0f98a10bf6ab797b51d

Backtrace: 0x402cf522:0x3ffc0d50 0x40194781:0x3ffc0d70 0x40189b0e:0x3ffc0da0 0x4018a1e5:0x3ffc0ec0 0x400fd0d1:0x3ffc0f00 0x400fdac5:0x3ffc0f30 0x400fdbc9:0x3ffc0fe0 0x400fdbd9:0x3ffc1010 0x40097461:0x3ffc1030
0x402cf522: OvmsMetricBool::AsBool(bool) at /Users/markojuhanne/esp/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/main/ovms_metrics.cpp:656

0x40194781: OvmsVehicleKiaNiroEv::Ticker1(unsigned int) at /Users/markojuhanne/esp/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/components/vehicle_kianiroev/src/vehicle_kianiroev.cpp:784

0x40189b0e: OvmsVehicle::VehicleTicker1(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, void*) at /Users/markojuhanne/esp/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/components/vehicle/vehicle.cpp:1210

0x4018a1e5: void std::_Mem_fn_base<void (OvmsVehicle::*)(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, void*), true>::operator()<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, void*, void>(OvmsVehicle*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&, void*&&) const at /Users/markojuhanne/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/functional:600
 (inlined by) void std::_Bind<std::_Mem_fn<void (OvmsVehicle::*)(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, void*)> (OvmsVehicle*, std::_Placeholder<1>, std::_Placeholder<2>)>::__call<void, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&, void*&&, 0u, 1u, 2u>(std::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&, void*&&>&&, std::_Index_tuple<0u, 1u, 2u>) at /Users/markojuhanne/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/functional:1074
 (inlined by) void std::_Bind<std::_Mem_fn<void (OvmsVehicle::*)(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, void*)> (OvmsVehicle*, std::_Placeholder<1>, std::_Placeholder<2>)>::operator()<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, void*, void>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&, void*&&) at /Users/markojuhanne/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/functional:1133
 (inlined by) std::_Function_handler<void (std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, void*), std::_Bind<std::_Mem_fn<void (OvmsVehicle::*)(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, void*)> (OvmsVehicle*, std::_Placeholder<1>, std::_Placeholder<2>)> >::_M_invoke(std::_Any_data const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&, void*&&) at /Users/markojuhanne/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/functional:1871

0x400fd0d1: std::function<void (std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, void*)>::operator()(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, void*) const at /Users/markojuhanne/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/functional:2271

0x400fdac5: OvmsEvents::HandleQueueSignalEvent(event_queue_t*) at /Users/markojuhanne/esp/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/main/ovms_events.cpp:180

0x400fdbc9: OvmsEvents::EventTask() at /Users/markojuhanne/esp/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/main/ovms_events.cpp:148

0x400fdbd9: EventLaunchTask(void*) at /Users/markojuhanne/esp/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/main/ovms_events.cpp:56

0x40097461: vPortTaskWrapper at /Users/markojuhanne/esp/esp-idf/components/freertos/port.c:435

Entering gdb stub now.
$T0b#e6GNU gdb (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_apple-darwin16.3.0 --target=xtensa-esp32-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /Users/markojuhanne/esp/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/build/ovms3.elf...done.
Remote debugging using /dev/cu.SLAB_USBtoUART
OvmsMetricBool::AsBool (this=0x0, defvalue=false) at /Users/markojuhanne/esp/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/main/ovms_metrics.cpp:656
656       if (IsDefined())
[... 21 lines cut ...]
#1  0x40194784 in OvmsVehicleKiaNiroEv::Ticker1 (this=0x3ffdd270, ticker=<optimized out>)
    at /Users/markojuhanne/esp/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/components/vehicle_kianiroev/src/vehicle_kianiroev.cpp:379
379                             m_v_door_lock_fr->AsBool() & m_v_door_lock_fl->AsBool() & m_v_door_lock_rr->AsBool() & m_v_door_lock_rl->AsBool() & !StdMetrics.ms_v_door_trunk->AsBool()
(gdb) info locals
isCharging = 160        
(gdb) p m_v_
m_v_cruise_control          m_v_door_lock_rr            m_v_heated_handle           m_v_preheating              m_v_seat_belt_back_right    m_v_traction_control
m_v_door_lock_fl            m_v_emergency_lights        m_v_power_usage             m_v_rear_defogger           m_v_seat_belt_driver        m_v_trip_consumption1
m_v_door_lock_fr            m_v_env_highbeam            m_v_preheat_timer1_enabled  m_v_seat_belt_back_left     m_v_seat_belt_passenger     m_v_trip_consumption2
m_v_door_lock_rl            m_v_env_lowbeam             m_v_preheat_timer2_enabled  m_v_seat_belt_back_middle   m_v_steering_mode           
(gdb) p m_v_door_lock_fr
$1 = (OvmsMetricBool *) 0x0
(gdb) l
374             {
375             //ESP_LOGD(TAG,"Pollstate: %d sec with no client: %d ",m_poll_state, kn_secs_with_no_client);
376     
377             // Register car as locked only if all doors are locked
378             StdMetrics.ms_v_env_locked->SetValue(
379                             m_v_door_lock_fr->AsBool() & m_v_door_lock_fl->AsBool() & m_v_door_lock_rr->AsBool() & m_v_door_lock_rl->AsBool() & !StdMetrics.ms_v_door_trunk->AsBool()
380             );
381     
382             if(kn_shift_bits.CarOn!=StdMetrics.ms_v_env_on->AsBool())
383                     {
(gdb) 
```

dexterbg, 2019-08-07:
Marko,

sorry – I had this on the Twizy but didn't recognize it as a potential general problem.

I use a bool m_ready member to inhibit execution of critical code (tickers & CAN processing) until the init has been finished.

We can generalize that pattern by moving the variable & checks into the vehicle base class. Vehicles then just need to set it to true at the end of their
constructor.

Objections?

Regards,
Michael

mjuhanne, 2019-08-07:
That sounds like a nice solution. No need for actual mutexes.

markwj, 2019-08-08:
Seems messy to do this in every vehicle implementation.

Can we have the boolean, initialised to false in the OvmsVehicle constructor, as described. And then have SetVehicle() set the flag (as the vehicle must have been initialised by that time). Then, honour the flag in the OvmsVehicle functions for things like timers and CAN messages?

Regards, Mark

dexterbg, 2019-08-10:
Implemented in 90835455cff1d27254322f09a968c61424c19f57

closed 2019-08-10 by dexterbg