openvehicles/Open-Vehicle-Monitoring-System-3 #252: MG eZS support?

glynhudson, 2019-08-19:
I'm keen to help with adding support for the MG eZS. Is anyone else interested? Has anyone got access or an order placed for an MG eZS? Getting access to the car for testing will be the main issue. I may be able to help with this. Please contact me if interested, financial support may be available to help with development.

It looks like a rather interesting EV, Saic Chinese built with a decent battery size and very competitive price.

There is a lot of excitement regarding the launch of this EV in the UK, pre-orders have sold out. In Europe (unlike in Asia) the MG eZS is not due to come with mobile app support, there is a lot of people keen to add third party OVMS app support.

https://mg.co.uk/mg-zs-electric/

glynhudson, 2019-10-18:
Mike Abbott has been making good progress obtaining CAN access and starting the process of reverse decoding the MG eZS working towards adding OVMS support. Here is his progress so far, help with decoding is much appreciated, please get involved if you're interested: 

Write up of work done so far: 
https://www.ruralelectricsystems.com/morris-reverse-engineering-mg

CAN ID Decoding spreadsheet 
https://docs.google.com/spreadsheets/d/1cs_8YQVGERspQiPMMY5yFpfLYoyH7UEP30oKzrMk35g/edit#gid=120754786

Captured logs: 
https://drive.google.com/drive/folders/1kDKXQ16qzaEfDEueMquFHN56qHOnI3_j

Gateway module pinout:
https://docs.google.com/document/d/1G00gL3riW5w8HCd-Nwp30EphJW_JDmI382TbkeD2U94/edit

glynhudson, 2020-08-29:
Good news, Mike has made good progress reverse decoding and identifying PID's for the MG ZS EV. It should be possible to now add support to OVMS 👍 

Is anyone with OVMS coding experience able to help? 

See spreadsheet with decoded data: https://docs.google.com/spreadsheets/d/1cs_8YQVGERspQiPMMY5yFpfLYoyH7UEP30oKzrMk35g/edit?fbclid=IwAR2NBmoZdVB-i8IBUW2QRfGpnfLUQKXGGQQcDv4tah93q8HXM2fxuszIAF4#gid=4521783

It should be possible to add support for the following:

Item | Support Status
-- | --
SOC | OBD PID is Known
Range | --
GPS Location | Yes (from modem module GPS)
Speed | OBD PID is Known
Cabin Temperature | Multiple PIDs available
Ambient Temperature | OBD PID is Known
SetPoint Temperature | OBD PID is Known
HVAC Fan Speed | OBD PID is Known
HVAC Heating/Cooling Status | Can infer from PTC output and Blower temp values
HVAC On Status | OBD PID is Known
HVAC Temperature Setpoint | OBD PID is Known
HVAC Ventilation Mode | OBD PID is Known
BMS v+t | Voltage and Currrent PIDs Known
TPMS | ECU is available not yet decoded
Charge Status | OBD PID is Known
Charge Interruption Alerts | Can be inferred

Item | Support Status
-- | --
SOC | OBD PID is Known
Range | --
GPS Location | Yes (from modem module GPS)
Speed | OBD PID is Known
Cabin Temperature | Multiple PIDs available
Ambient Temperature | OBD PID is Known
SetPoint Temperature | OBD PID is Known
HVAC Fan Speed | OBD PID is Known
HVAC Heating/Cooling Status | Can infer from PTC output and Blower temp values
HVAC On Status | OBD PID is Known
HVAC Temperature Setpoint | OBD PID is Known
HVAC Ventilation Mode | OBD PID is Known
BMS v+t | Voltage and Currrent PIDs Known
TPMS | ECU is available not yet decoded
Charge Status | OBD PID is Known
Charge Interruption Alerts | Can be inferred

https://www.openvehicles.com/node/2581

chrisstaite, 2020-08-31:
Initial support added on https://github.com/MG-EV-Hackers/Open-Vehicle-Monitoring-System-3-MGEV/tree/feature/mg-zs-ev

dexterbg, 2020-09-06:
Android App prerelease 3.17.1: https://github.com/openvehicles/Open-Vehicle-Android/raw/master/release/OpenVehicleApp-release-3.17.1.apk

Your clone looks like having more than initial support already. Waiting for your pull request.

Regards,
Michael

chrisstaite, 2020-09-06:
Bit worried about setting off the alarm and keeping too much awake at the moment. Will hopefully get those sorted and make a request.

closed 2020-09-17 by dexterbg

dclarkbell, 2020-12-01:
Any progress on this MG ZS EV code? Do it now fully work? Thank You in advance.

glynhudson, 2020-12-01:
The MG ZS is now fully functional, the code is still in beta but it's fully working

commit ef124c37 (2020-09-17): Merge pull request #431 from MG-EV-Hackers/feature/mg-zs-ev

Add MG ZS EV support (fix #252)