openvehicles/Open-Vehicle-Monitoring-System-3 #253: unable to save script if directory name contains a space

gbonadio, 2019-08-22:
Unable to save a file from the text editor into a directory with a space. (my location did contain a space in its description 

When saving all seems fine:
File /store/events/location.leave.Ile Falcon/bt95pct.sh saved.
but the diectory is empty. As a result no script is executed at entering/leaving the location.

For sake of debug, I tried to save with a directory with space but putting space in the filename:
/store/events/location.leave.IleFalcon/empty with space.sh
Message:
File /store/events/location.leave.IleFalcon/empty with space.sh saved. and it turns out it working so the issue is only when having a space in the directory. Seems the vfs command can't handle spaces, I tried to remove the old directories from the location with a space but not working. Tried with vfs rmdir "path/with space", vfs rmdir path/with\ space

WorkAround: rename the location description without space, create the new directories and recreate the scripts without spaces. Works well

slcasner, 2019-08-22:
To include the space in the directory or file name, did you quote the
whole pathname?  Otherwise the vfs commands would see the path as
multiple tokens.

                                                        -- Steve

gbonadio, 2019-08-23:
yes I tried to quote both with "directory path" and 'directory path'

dexterbg, 2019-08-25:
Fixed in commit 8d532fbbadd6cda190f34fa9f2d430f943f9ad0c

closed 2019-08-25 by dexterbg