openvehicles/Open-Vehicle-Monitoring-System-3 #255: Hyundai Ioniq BEV support...curretly partially working using Soul

glynhudson, 2019-08-25:
Yesterday I tested OVMS V3 running V3.2.002 on a Hyundai Ioniq BEV. I assumed the Ioniq would share the eNiro / Kona platform added in https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/pull/219. However, with the eNiro vehicle selected no metrics where read. 

However, when the Kia Soul vehicle was selected most metrics (apart from SoC) seem to be correctly read, see below. I used the Generic Left OBD2 cable. I also tried the Kia Soul cable but that resulted in no readings. 

How to read Ioniq SoC is documented in EVnotify code: https://github.com/EVNotify/EVNotify/blob/master/app/www/components/cars/IONIQ_BEV.vue#L123

A list of PIDS for Ioniq, Kona, Niro and Soul seem to be documented here: https://github.com/JejuSoul/OBD-PIDs-for-HKMC-EVs

I have a contact with Ioniq BEV, therfore am able to do more testing if required 

```
m.freeram                                2037628
m.hardware                               OVMS WIFI BLE BT cores=2 rev=ESP32/1
m.monotonic                              182Sec
m.net.mdm.iccid                          8944500301190489276
m.net.mdm.model                          35316B12SIM5360E
m.net.provider                           vodafone UK Hologram
m.net.sq                                 -93dBm
m.net.type                               modem
m.serial
m.tasks                                  17
m.time.utc                               1566653556Sec
m.version                                3.2.002/factory/main (build idf v3.1-dev-2835-g1512694 May 12 2019 22:10:06)
s.v2.connected                           yes
s.v2.peers                               0
s.v3.connected
s.v3.peers
v.b.12v.current
v.b.12v.voltage                          15.2V
v.b.12v.voltage.alert
v.b.12v.voltage.ref                      12.6V
v.b.c.temp
v.b.c.temp.alert
v.b.c.temp.dev.max
v.b.c.temp.max
v.b.c.temp.min
v.b.c.voltage
v.b.c.voltage.alert
v.b.c.voltage.dev.max
v.b.c.voltage.max
v.b.c.voltage.min
v.b.cac                                  74.25Ah
v.b.consumption                          0Wh/km
v.b.coulomb.recd
v.b.coulomb.used
v.b.current
v.b.energy.recd
v.b.energy.used
v.b.health
v.b.p.level.avg
v.b.p.level.max
v.b.p.level.min
v.b.p.level.stddev
v.b.p.temp.avg
v.b.p.temp.max
v.b.p.temp.min
v.b.p.temp.stddev
v.b.p.temp.stddev.max                    0°C
v.b.p.voltage.avg
v.b.p.voltage.max
v.b.p.voltage.min
v.b.p.voltage.stddev
v.b.p.voltage.stddev.max
v.b.power
v.b.range.est
v.b.range.full                           160km
v.b.range.ideal                          160km
v.b.soc
v.b.soh                                  110%
v.b.temp                                 20°C
v.b.voltage
v.c.charging                             no
v.c.climit
v.c.current
v.c.duration.full
v.c.duration.range
v.c.duration.soc
v.c.kwh
v.c.limit.range                          0km
v.c.limit.soc                            0%
v.c.mode
v.c.pilot
v.c.state
v.c.substate
v.c.temp
v.c.time                                 0Sec
v.c.timermode
v.c.timerstart
v.c.type
v.c.voltage
v.d.cp
v.d.fl
v.d.fr
v.d.hood
v.d.rl
v.d.rr
v.d.trunk
v.e.alarm
v.e.awake
v.e.c.config
v.e.c.login
v.e.cabintemp
v.e.charging12v
v.e.cooling
v.e.drivemode
v.e.drivetime                            0Sec
v.e.footbrake
v.e.gear
v.e.handbrake
v.e.headlights
v.e.heating
v.e.hvac
v.e.locked
v.e.on                                   no
v.e.parktime                             182Sec
v.e.regenbrake
v.e.temp
v.e.throttle
v.e.valet
v.i.temp
v.m.rpm
v.m.temp
v.p.acceleration
v.p.altitude                             104.1m
v.p.direction                            0°
v.p.gpshdop                              0.7
v.p.gpslock                              yes
v.p.gpsmode                              AA
v.p.gpsspeed                             0km/h
v.p.latitude                             52.3922
v.p.longitude                            -1.52626
v.p.odometer
v.p.satcount                             17
v.p.speed
v.p.trip
v.tp.fl.p
v.tp.fl.t
v.tp.fr.p
v.tp.fr.t
v.tp.rl.p
v.tp.rl.t
v.tp.rr.p
v.tp.rr.t
v.type                                   KS
v.vin
xks.b.aux.soc                            100%
xks.b.bms.soc                            0%
xks.b.cell.det.max                       0%
xks.b.cell.det.max.no                    0
xks.b.cell.det.min                       0%
xks.b.cell.det.min.no                    0
xks.b.cell.volt.max                      0V
xks.b.cell.volt.max.no                   0
xks.b.cell.volt.min                      0V
xks.b.cell.volt.min.no                   0
xks.b.heat1.temp                         0°C
xks.b.heat2.temp                         0°C
xks.b.inlet.temp                         0°C
xks.b.max.temp                           0°C
xks.b.min.temp                           0°C
xks.c.power                              0kW
xks.c.speed                              0km/h
xks.e.climate.ac                         no
xks.e.climate.auto                       no
xks.e.climate.driver.only                no
xks.e.climate.fan.speed                  0
xks.e.climate.mode                       0
xks.e.climate.resirc                     no
xks.e.climate.temp                       0°C
xks.e.highbeam                           no
xks.e.lowbeam                            no
xks.e.pos.arrival.hour                   0
xks.e.pos.arrival.minute                 0
xks.e.pos.dist.to.dest                   0km
xks.e.pos.street
xks.e.preheat.timer1.enabled             no
xks.e.preheat.timer2.enabled             no
xks.e.preheating                         no
xks.e.trip                               0km
xks.e.trip.energy.recuperated            0kWh
xks.e.trip.energy.used                   0kWh
xks.ldc.in.volt                          12V
xks.ldc.out.amps                         0A
xks.ldc.out.volt                         12V
xks.ldc.temp                             0°C
xks.obc.pilot.duty                       0%
xks.obc.timer.enabled                    no
xks.v.cruise.control.enabled             no
xks.v.emergency.lights                   no
xks.v.power.usage                        0kW
xks.v.seat.belt.back.left                no
xks.v.seat.belt.back.middle              no
xks.v.seat.belt.back.right               no
xks.v.seat.belt.driver                   no
xks.v.seat.belt.passenger                no
xks.v.steering.mode                      Unknown
xks.v.traction.control                   no
xks.v.trip.consumption.KWh/100km         0
xks.v.trip.consumption.km/kWh            0
xks.version                              0.4.7 May  8 2019 20:10:17
```

glynhudson, 2019-09-23:
Lots of PIDS and decoding info for Kia and Hyundai car are documented in the source code for SoulEVSpy app https://github.com/langemand/SoulEVSpy

dalathegreat, 2020-01-01:
I added the polarwhite Ioniq model the the android app here https://github.com/openvehicles/Open-Vehicle-Android/pull/103

glynhudson, 2020-01-16:
Have you tested the Ioniq with OVMS? I've not had any luck getting any metrics from an Ioniq using OVMS. I don't think it's a good idea to add android app images unless some sort of support is available.

Was the Ioniq screenshot on your AndroidApp PR taken with OVMS installed in an Ioniq? If so it looks like most metrics are supported?! https://github.com/openvehicles/Open-Vehicle-Android/pull/103

What version FW and cable and vehicle settings did you use? It would be good to document what features are supported and possibly create a seperate vehicle entry.

glynhudson, 2020-11-24:
Ioniq is not currently supported by OVMS

sixtybi, 2020-12-14:
> Ioniq is not currently supported by OVMS

Is there any plan for Ioniq to be supported? EVNotify works with Ioniq, so I guess technically it should be possible

glynhudson, 2020-12-14:
Most of the work should already have been done since the Kona is supported, some metrics may already work. All the developers for OVMS are volunteers, support for vehicles is usually limited to what vehicles developers own or have easy access to. It may be a good idea to post on some Ioniq forums or contact EVNotify to see if anyone with knowledge of the Ioniq has interested in adding support to OVMS.

sixtybi, 2020-12-15:
> Most of the work should already have been done since the Kona is supported, some metrics may already work. All the developers for OVMS are volunteers, support for vehicles is usually limited to what [...]

I just about to order OVMS hardware if at least SoC can be extracted for Ioniq. But if it is totally unusable I wouldn't waste my money on this. But if it can be used somehow, I would maybe help improving the support for this car. I am just about to order OpenEVSE and this OVMS would be a great addition to this. Both of these, I would like to integrate to my Home assistant to control charging.

henke2, 2020-12-15:
> I just about to order OVMS hardware if at least SoC can be extracted for Ioniq. But if it is totally unusable I wouldn't waste my money on this. But if it can be used somehow, I would maybe help [...]

I have the OVMS HW3(?!) and the Ioniq 28. From what I understand SOC is to be there but I have not yet got it up and running.

I also run EV-notify and that one is fine but unstable for me. Needs an extra android device in the car kept alive as a server.

OVMS HW sould be able to do this easy but I've not had time to dig in to it.

Running EV notify I learnt that CAN is only active when the car is either in running state or charging. When I did my initial tests the car was off.

I'm 100% sure OVMS can do it. I don't have the knowledge to program a new profile nor understand how to use the PIDs. For any developer it's probably very east to have it up in beta state.

sixtybi, 2020-12-15:
> I have the OVMS HW3(?!) and the Ioniq 28. From what I understand SOC is to be there but I have not yet got it up and running.
>
> I also run EV-notify and that one is fine but unstable for me. Needs an [...]

Thanks! I guess I will give it a chance. I don't afraid playing with it. I just don't want to waste money on this if it not working for sure, but as I can imagine, if it works for Kona and eNiro, it should be fine also for Ioniq.

GitSole, 2020-12-15:
I Just bought a Ioniq 28 and am realising i would really like something like OVMS, if it is confirmed i can at least get SOC, battery temps and basic info like that, let alone control AC, then i would be willing to order OVMS hardware, and although i'm not a developer or anything, i'll gladly help as much as possible if anyone wants to use my car and hardware for testing, even from a distance!

dexterbg, 2021-03-17:
By request of Henri Bachmann on the german GoingElectric forum, I've added the Hyundai Ioniq vFL (28 kWh):

https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/commit/797e1aad569a64d5ff711013c8251ba201713b7b

The implementation is based on the reverse engineering data published by the EVNotify project:
https://github.com/EVNotify/EVNotiPi/blob/master/car/ioniq_bev.py

Module Documentation:
https://docs.openvehicles.com/en/latest/components/vehicle_hyundai_ioniqvfl/docs/index.html

Henri has tested the module, but more tests from other Ioniq drivers are welcome.

One of the PIDs needs more data/logs: the "driveMotorSpeed" didn't match the actual speed on Henri's tests. That PID isn't used by the EVNotify system so may be wrong, or could be RPM.

If you can help with logs on this, enable file logging and do a test drive with level verbose on module v-hyundaivfl. Try to maintain a fixed maximum speed for 20 seconds for reference. Send me the logs and the max speed.

Regards,
Michael

LolDeibid, 2021-05-07:
> 
> 
> By request of Henri Bachmann on the german GoingElectric forum, I've added the Hyundai Ioniq vFL (28 kWh):
> 
> [797e1aa](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/commit/797e1aad569a64d5ff711013c8251ba201713b7b)
> 
> The implementation is based on the reverse engineering data published by the EVNotify project:
> https://github.com/EVNotify/EVNotiPi/blob/master/car/ioniq_bev.py
> 
> Module Documentation:
> https://docs.openvehicles.com/en/latest/components/vehicle_hyundai_ioniqvfl/docs/index.html
> 
> Henri has tested the module, but more tests from other Ioniq drivers are welcome.
> 
> One of the PIDs needs more data/logs: the "driveMotorSpeed" didn't match the actual speed on Henri's tests. That PID isn't used by the EVNotify system so may be wrong, or could be RPM.
> 
> If you can help with logs on this, enable file logging and do a test drive with level verbose on module v-hyundaivfl. Try to maintain a fixed maximum speed for 20 seconds for reference. Send me the logs and the max speed.
> 
> Regards,
> Michael

I'm about to purchase an Ioniq mk1. How can I learn to "read" all PID's or just the work that we need to improve OVMS for Ioniqs? 

Still without remote climate control, chargue control.... Is it completely impossible to make it works?, or it's just that there is no people working on it?

Regards,
David

LolDeibid, 2021-05-07:
> I'm not familiar with the Hyundai model family… what's a "mk1"? What are the differences to the models already supported?

Mk1 means old 28kwh Ioniq, so mk2 it's new 38kwh.

dexterbg, 2021-05-07:
So that's the Hyundai Ioniq vFL (28 kWh) and you'd like to improve on the module? Go ahead. I didn't implement all PIDs known, just a first set. And you'll probably find more info on the web. You can also use the OVMS RE toolkit to discover more PIDs.

dexterbg, 2021-05-08:
Some merged mails I wrote on this on the developer list:

First of all, query the search engines and open source hubs (github, sourceforge etc.) for existing publications on your car. Try searching for "<car model> OBD PIDs".

If there is no open info but some diagnostic tool available, consider buying that tool and an OBD split cable to enable you to connect both the tool and the OVMS at the same time. That way you can record & analyze the communication between the tool and the car.

Please read: https://docs.openvehicles.com/en/latest/crtd/can_logging.html

The OBD2 port has only one standard CAN bus (on pins 6+14) which is connected as can1. Manufacturer specific additional buses may be present in your car, it depends on their purposes and your goals if you need to use them.

The standard OVMS OBD2 cable connects can2 additionally to pins 12+13, which are used by many manufacturers for a secondary CAN bus. If you need the third CAN bus as well, you'll need to use the DA26 connector.

If you need to connect to the K-Line or need a fourth CAN bus, you'll need to add the K-Line or SWCAN extension boards.

If you don't know anything about the car, you'll first need to find the right CAN bus and baud rate to access the vehicle (see above).

Be aware there may be no regular transmissions on the buses you can access via the OBD port, so simply listening may not be enough.

If you can see frames on some buses, using the RE tool is an option. Another is to stream the packets to SavvyCAN for analysis. Both will help you to identify related bits & bytes. Next step is then to find out which metrics and indicators can be read from the frames. Standard procedure is to record changes you see on the bus from actions you do with the car, and to correlate changes you see in the car's displays to those on the bus. Then you need to decode the values, i.e. how to convert the raw values into their respective readings.

RE tool UI: https://docs.openvehicles.com/en/latest/plugin/retools/README.html

Next step is checking for OBD2 or CANopen devices on the bus(es). These devices need to be queried actively, so you need to send requests on the bus.

For CANopen, read: https://docs.openvehicles.com/en/latest/components/canopen/docs/Howto-detect-CANopen-nodes.html

We don't have a similar guide for OBD2 yet. Begin by some simple standard broadcasts to see if you get any reaction, for example "obdii can1 request broadcast 0100" asks all broadcast listeners to tell which of the service 01 PIDs in the range 01…20 they support. The responses should give you a hint about the device addresses, so you can investigate specifically.

Standard OBD PIDs: https://en.wikipedia.org/wiki/OBD-II_PIDs

Most OBD2 devices won't respond to broadcasts though, and many EVs do not respond to standard OBD modes & PIDs. To identify them, you need to send a test request to all CAN IDs in your assumed ID range. The scan tools for this are below "re obdii scan":

OVMS# re obdii scan start ?
Usage: re obdii scan start <bus> <ecu> <start_pid> <end_pid> [-r<rxid>[-<rxid>]] [-t<poll_type>] [-x<timeout>]
Give all values except bus and timeout hexadecimal. Options can be positioned anywhere.
Default <rxid> is <ecu>+8, try 0-7ff if you don't know the responding ID.
Default <poll_type> is 22 (ReadDataByIdentifier, 16 bit PID).
Default <timeout> is 3 seconds.

RE PID scanner UI: https://docs.openvehicles.com/en/latest/plugin/repidscan/README.html

Most newer vehicles will support reading 16 bit PIDs by UDS poll type 22. If you're lucky, poll type 24 is supported and gives you info on how to translate the raw values.

If not you once again need to monitor the response data, correlate changes to car data updates and actions and find out how to convert the values.

Be aware you may need to activate a diagnostic session with a device before actually be able to read data from it. Diagnostic sessions need to be kept alive by either sending the session control request once per minute or activating the "tester present" signal ("re obdii tester"). Once again, you need to find out which session mode you need for a device. You can tell if the device supports a session mode by it's response.

More on UDS request types: https://en.wikipedia.org/wiki/Unified_Diagnostic_Services

You'll find more info on CAN reverse engineering and OBD hacking on the web. A good starting point is SavvyCAN.

You'll also find more on all of this by browsing the list archives: http://lists.openvehicles.com/pipermail/ovmsdev/

Regards,
Michael

bassmaster187, 2022-02-16:
@dexterbg I sent you an email. 
Can all others confirm there is no speed on OVMS at all?

dexterbg, 2022-02-19:
Thanks to PID inspection & decoding by @bassmaster187 the vehicle speed reading is now working.

dexterbg, 2022-02-19:
Note: there is some difference to the speed displayed, which is perfectly normal for most cars. I recommend testing if the speed reading matches the GPS speed measured e.g. by some App. If you'd rather have the metric matching the dashboard closely, we can add a user configurable scaling.

@bassmaster187 codewise, that could simply be:
```
      // Read vehicle speed:
      float speed =
        UnitConvert(Mph, Kph, RXB_INT16(53) / 100.0f) * 
        MyConfig.GetParamValueFloat("xhi", "speedfactor", 1.0f);
      StdMetrics.ms_v_pos_speed->SetValue(fabs(speed));
```

…and you would need to add this to the vehicle constructor:
```
  // Init config:
  MyConfig.RegisterParam("xhi", "Hyundai Ioniq vFL", true, true);
```

A user can then do `config set xhi speedfactor 1.1` to scale up the speed by 10%. You can also decide to choose a default higher than `1.0f`.

Regards,
Michael

DobbyNator94, 2023-03-06:
Hey everybody,

is there ever be the chance that OVMS can control the cabin temperature (pre heat / cool) and lock/unlock the car?

These are the things that I really missed about the Ioniq 28KWh, because of the lack of BlueLink connectivity here in Europe.

Thank you all for your work.
When these things work someday, I would by the OVMS instantly.