openvehicles/Open-Vehicle-Monitoring-System-3 #264: Kia eNiro Stability Issues

glynhudson, 2019-09-23:
See forum thread with a number of users reporting the issue: https://www.openvehicles.com/node/2260

Are you able to replicate the same stability issue @goev?

Timopen, 2019-09-30:
Glyn
I have a fix for this.
What happens is that there are too many errors in the obd2 pulling when the car is off. The esp shuts off after a number off errors. 2 pulls too close to each other are causing this in the e-niro module. If someone is willing to help me with updating to git, I can send the changed files.

Timopen, 2019-09-30:
Managed to create a pull request myself, first time ever. :)

glynhudson, 2019-10-20:
Nice work! Your PR has been merged: https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/pull/267

Feedback from users has been positive https://www.openvehicles.com/node/2260#comment-5994

@dexterbg please could this fix be included in the next stable release

closed 2019-10-20 by glynhudson