openvehicles/Open-Vehicle-Monitoring-System-3 #266: V3 MQTT topic prefix is ignored

glynhudson, 2019-09-25:
Whatever custom topic prefix is entered in the V3 topic prefix field is ignore and the default topic is always used e.g `ovms/<username>/<vehicle id>`

dexterbg, 2019-09-26:
The V3 server only reads the topic.prefix config on connecting. Did you do a reconnect after changing it?

glynhudson, 2019-09-30:
Just checked again and even after restarting V3 server (stop/start), the topic prefix is ignored. I tested with and without a leading `\` 

[image]

markwj, 2020-02-02:
I can't repeat this issue. I did add some more diagnostics as 1d1819a2, to log the topic used when displaying the 'connect' log.

> OVMS# config set server.v3 topic.prefix ovms/markwj/DEVBENCH2/
> Parameter has been set.
>
> OVMS# config list server.v3
> server.v3 (readable writeable)
>   server: api.openvehicles.com
>   topic.prefix: ovms/<redacted>/DEVBENCH2/
>   user: <redacted>
>
> OVMS# server v3 start
> I (4436162) ovms-server-v3: Status: Server has been started
> I (4436162) ovms-server-v3: OVMS Server v3 running
> I (4436172) ovms-server-v3: Connection is api.openvehicles.com:1883 DEVBENCH/<redacted>
> I (4436172) ovms-server-v3: Status: Connecting...
> V (4436332) ovms-server-v3: OvmsServerV3MongooseCallback(MG_EV_CONNECT=0)
> I (4436332) ovms-server-v3: Connection successful
> V (4436342) ovms-server-v3: OvmsServerV3MongooseCallback(MG_EV_MQTT_CONNACK)
> I (4436342) ovms-server-v3: Status: OVMS V3 MQTT login successful
> I (4437022) ovms-server-v3: Subscribe to MQTT topics
> I (4437022) ovms-server-v3: Transmit all metrics
> I (4437022) ovms-server-v3: Tx metric ovms/<redacted>/DEVBENCH2/metric/m/egpio/input=0,1,2,3,4,5,6,7,8,9

I also tested the web interface, and it seems to set the parameter correctly. The prefix entered should not start with a '/', but should end with one.

glynhudson, 2020-02-08:
Just tested on `
3.2.010-3-g2fced46/ota_0/edge (build idf v3.3-beta3-775-gdc1ca69 Feb 7 2020 00:01:09)` and I can confirm this issue has been fixed 👍

closed 2020-02-08 by glynhudson