openvehicles/Open-Vehicle-Monitoring-System-3 #269: Compiling OVMS

glynhudson, 2019-10-04:
I'm trying to set up to ESP toolchain to be able to compile OVMS V3. I followed the [Espressif Setup Guide](https://docs.espressif.com/projects/esp-idf/en/stable/get-started/) to install the latest tool chain and have been able to successfully compile the 'hello-word' example.  

I'm aware I now need to switch to the [OVMS specific older version of esp-idf](https://github.com/openvehicles/esp-idf) in order to compile OVMS V3. I've tried changing the `$IDF_PATH` to the OVMS esp-idf but I get a `Toolchain version is not supported` warning and some errors (see below). Any trips on how to make this work, I've been trying to follow the guide in the old OVMS developers gdoc. Once I get this working would you like me to write a guide for the new docs page? 

```
WARNING: Toolchain version is not supported: esp32-2019r1
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 8.2.0
Expected to see version(s): 5.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Toolchain version is not supported: esp32-2019r1
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 8.2.0
Expected to see version(s): 5.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Missing submodule components/bootloader/subproject/components/micro-ecc/micro-ecc...
Attempting 'git submodule update --init components/bootloader/subproject/components/micro-ecc/micro-ecc' in esp-idf root directory...
error: pathspec 'components/bootloader/subproject/components/micro-ecc/micro-ecc' did not match any file(s) known to git.
/home/bailey/Downloads/dev/esp-idf/make/project.mk:611: recipe for target '/home/bailey/Downloads/dev/esp-idf/components/bootloader/subproject/components/micro-ecc/micro-ecc/.git' failed
make[1]: *** [/home/bailey/Downloads/dev/esp-idf/components/bootloader/subproject/components/micro-ecc/micro-ecc/.git] Error 1
/home/bailey/Downloads/dev/esp-idf/components/bootloader/Makefile.projbuild:41: recipe for target '/home/bailey/Downloads/dev/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/build/bootloader/bootloader.bin' failed
make: *** [/home/bailey/Downloads/dev/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/build/bootloader/bootloader.bin] Error 2
```

dexterbg, 2019-10-04:
Glyn,

you currently also need the old toolchain version, or the one adding the PSRAM fix to that. See my previous posts.

Regards,
Michael

glynhudson, 2019-10-04:
Thanks, where can I find your previous post?

dexterbg, 2019-10-05:
http://lists.openvehicles.com/pipermail/ovmsdev/2019-September/006285.html

glynhudson, 2019-10-05:
Thanks, I've switched to using [toolkit -98](https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-98-g4638c4f-5.2.0-20190827.tar.gz) along with the modded libs and from https://github.com/espressif/esp-idf/issues/2892

Trying to compile `spiram-fix-test` looks like good progress, the build starts but then fails with the following error, have you got any ideas?

`xtensa-esp32-elf-gcc: fatal error: -fuse-linker-plugin, but liblto_plugin.so not found`

Full output:

```
~/Downloads/dev/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3$ make
Toolchain path: /home/bailey/Downloads/dev/old-esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-98-g4638c4f
Expected to see version: esp32-2019r1
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 5.2.0
Expected to see version(s): 8.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Python requirements from /home/bailey/Downloads/dev/esp/esp-idf/requirements.txt are satisfied.
GENCONFIG
App "ovms3" version: 3.2.002-281-g2867065
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-98-g4638c4f
Expected to see version: esp32-2019r1
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 5.2.0
Expected to see version(s): 8.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
CC build/bootloader/bootloader_support/src/bootloader_flash.o
CC build/bootloader/bootloader_support/src/flash_qio_mode.o
CC build/bootloader/bootloader_support/src/bootloader_init.o
CC build/bootloader/bootloader_support/src/bootloader_utility.o
CC build/bootloader/bootloader_support/src/bootloader_common.o
CC build/bootloader/bootloader_support/src/bootloader_clock.o
CC build/bootloader/bootloader_support/src/bootloader_flash_config.o
CC build/bootloader/bootloader_support/src/flash_partitions.o
CC build/bootloader/bootloader_support/src/esp_image_format.o
CC build/bootloader/bootloader_support/src/flash_encrypt.o
CC build/bootloader/bootloader_support/src/bootloader_random.o
CC build/bootloader/bootloader_support/src/esp32/secure_boot_signatures.o
CC build/bootloader/bootloader_support/src/esp32/secure_boot.o
CC build/bootloader/bootloader_support/src/esp32/bootloader_sha.o
CC build/bootloader/bootloader_support/src/esp32/flash_encrypt.o
AR build/bootloader/bootloader_support/libbootloader_support.a
CC build/bootloader/log/log.o
AR build/bootloader/log/liblog.a
CC build/bootloader/spi_flash/spi_flash_rom_patch.o
AR build/bootloader/spi_flash/libspi_flash.a
CC build/bootloader/soc/esp32/cpu_util.o
CC build/bootloader/soc/esp32/rtc_init.o
CC build/bootloader/soc/esp32/rtc_clk_init.o
CC build/bootloader/soc/esp32/rtc_sleep.o
CC build/bootloader/soc/esp32/rtc_time.o
CC build/bootloader/soc/esp32/rtc_clk.o
CC build/bootloader/soc/esp32/emac_hal.o
CC build/bootloader/soc/src/memory_layout_utils.o
CC build/bootloader/soc/src/soc_include_legacy_warn.o
CC build/bootloader/soc/src/hal/spi_slave_hal_iram.o
CC build/bootloader/soc/src/hal/spi_slave_hal.o
CC build/bootloader/soc/src/hal/spi_flash_hal_iram.o
CC build/bootloader/soc/src/hal/spi_flash_hal.o
CC build/bootloader/soc/src/hal/spi_hal_iram.o
CC build/bootloader/soc/src/hal/spi_hal.o
AR build/bootloader/soc/libsoc.a
CC build/bootloader/main/bootloader_start.o
AR build/bootloader/main/libmain.a
CC build/bootloader/efuse/esp32/esp_efuse_table.o
CC build/bootloader/efuse/src/esp_efuse_api.o
CC build/bootloader/efuse/src/esp_efuse_utility.o
CC build/bootloader/efuse/src/esp_efuse_fields.o
AR build/bootloader/efuse/libefuse.a
LD build/bootloader/bootloader.elf
**xtensa-esp32-elf-gcc: fatal error: -fuse-linker-plugin, but liblto_plugin.so not found**
compilation terminated.
/home/bailey/Downloads/dev/esp/esp-idf/make/project.mk:538: recipe for target '/home/bailey/Downloads/dev/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/build/bootloader/bootloader.elf' failed
make[1]: *** [/home/bailey/Downloads/dev/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/build/bootloader/bootloader.elf] Error 1
/home/bailey/Downloads/dev/esp/esp-idf/components/bootloader/Makefile.projbuild:41: recipe for target '/home/bailey/Downloads/dev/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/build/bootloader/bootloader.bin' failed
make: *** [/home/bailey/Downloads/dev/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/build/bootloader/bootloader.bin] Error 2
```

Output of `xtensa-esp32-elf-gcc -v`
```
Using built-in specs.
COLLECT_GCC=xtensa-esp32-elf-gcc
COLLECT_LTO_WRAPPER=/home/bailey/Downloads/dev/old-esp/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/5.2.0/lto-wrapper
Target: xtensa-esp32-elf
Configured with: /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/configure --build=x86_64-build_pc-linux-gnu --host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf --prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf --with-local-prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf/sysroot --with-sysroot=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf/sysroot --with-newlib --enable-threads=no --disable-shared --with-pkgversion='crosstool-NG crosstool-ng-1.22.0-98-g4638c4f' --disable-__cxa_atexit --enable-cxx-flags='-fno-rtti -ffunction-sections' --with-gmp=/builds/idf/crosstool-NG/.build/xtensa-es [... cut ...] /crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-libelf=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --enable-lto --enable-target-optspace --without-long-double-128 --disable-libgomp --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libquadmath-support --disable-nls --enable-languages=c,c++ --disable-libstdcxx-verbose --enable-threads=posix --enable-gcov-custom-rtio
Thread model: posix
gcc version 5.2.0 (crosstool-NG crosstool-ng-1.22.0-98-g4638c4f) 
```

dexterbg, 2019-10-06:
I'd say if you don't have that file, your toolkit installation or is flawed.

```
balzer@leela:~/esp/xtensa-esp32-elf> find . -name liblto_plugin.so
./libexec/gcc/xtensa-esp32-elf/5.2.0/liblto_plugin.so
```

```
balzer@leela:~/esp/xtensa-esp32-elf> xtensa-esp32-elf-gcc -v
Using built-in specs.
COLLECT_GCC=xtensa-esp32-elf-gcc
COLLECT_LTO_WRAPPER=/home/balzer/esp/xtensa-esp32-elf-1.22.0-98/bin/../libexec/gcc/xtensa-esp32-elf/5.2.0/lto-wrapper
Target: xtensa-esp32-elf
Configured with: /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/configure --build=x86_64-build_pc-linux-gnu --host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf --prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf --with-local-prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf/sysroot --with-sysroot=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf/sysroot --with-newlib --enable-threads=no --disable-shared --with-pkgversion='crosstool-NG crosstool-ng-1.22.0-98-g4638c4f' --disable-__cxa_atexit --enable-cxx-flags='-fno-rtti -ffunction-sections' --with-gmp=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpfr=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpc=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-isl=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-cloog=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-libelf=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --enable-lto --enable-target-optspace --without-long-double-128 --disable-libgomp --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libquadmath-support --disable-nls --enable-languages=c,c++ --disable-libstdcxx-verbose --enable-threads=posix --enable-gcov-custom-rtio
Thread model: posix
gcc version 5.2.0 (crosstool-NG crosstool-ng-1.22.0-98-g4638c4f) 
```

glynhudson, 2019-10-07:
Strangely I had a file called `liblto_plugin.so.0.0.0` at the same location, I just renamed it to `liblto_plugin.so` and it now seems to be compiling.

However, the build now fails with 

```
In member function 'virtual void esp32wifi::SetPowerMode(PowerMode)':
error: 'WIFI_PS_MODEM' was not declared in this scope
       esp_wifi_set_ps(WIFI_PS_MODEM);
                       ^
```

I'm on the latest `spiram-fix-test` branch: 30ffa4ff8a7c9fcdfc708f80089230a9a4677ba5

I assume this branch is comping ok for you? I wonder what the issue is on my system? I should now be running the exact same toolkit and edf-idf as yourself.

dexterbg, 2019-10-07:
I'd suggest doing a fresh toolkit & idf setup from scratch.

glynhudson, 2019-10-08:
Can you see any issue with how I'm installing the various components:

***
## Install Requirements

`sudo apt-get install git wget libncurses-dev flex bison gperf python python-pip python-setuptools python-serial python-click python-cryptography python-future python-pyparsing python-pyelftools cmake ninja-build ccache`

## Install Espressif IDF

```
cd ~/Downloads/dev/esp
git clone --recursive https://github.com/espressif/esp-idf.git
cd esp-idf
./install.sh
. ./export.sh
```

## Install toolkit -98

From: https://github.com/espressif/esp-idf/issues/2892#issuecomment-525286663

```
cd ~/Downloads/dev/old-esp
wget https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-98-g4638c4f-5.2.0-20190827.tar.gz
tar -xzf xtensa-esp32-elf-linux64-1.22.0-98-g4638c4f-5.2.0-20190827.tar.gz
export PATH=$PATH:$HOME/Downloads/dev/old-esp/xtensa-esp32-elf/bin
```

## Apply modded libs

From: https://github.com/espressif/esp-idf/issues/2892#issuecomment-525286663

```
wget https://dl.espressif.com/dl/newlib-psram-fix-ldstld-14aug.zip
unzip newlib-psram-fix-ldstld-14aug.zip

cp libc-psram-workaround.a /xtensa-esp32-elf/xtensa-esp32-elf/sysroot/lib/esp32-psram/libc.a
cp libg-psram-workaround.a /xtensa-esp32-elf/xtensa-esp32-elf/sysroot/lib/esp32-psram/libg.a
cp libm-psram-workaround.a /xtensa-esp32-elf/xtensa-esp32-elf/sysroot/lib/esp32-psram/libm.a
```

# Clone OVMS spiram-fix-test branch

```
cd Downloads/dev
git clone https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3
cd Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3
git submodule update --init --recursive
cd vehicle/OVMS.V3
git checkout spiram-fix-test
$ cp support/sdkconfig.default.hw31 sdkconfig
```
# Compile
$ make

### Error

```
In member function 'virtual void esp32wifi::SetPowerMode(PowerMode)':
error: 'WIFI_PS_MODEM' was not declared in this scope
       esp_wifi_set_ps(WIFI_PS_MODEM);
                       ^
```

dexterbg, 2019-10-08:
You're cloning the original esp-idf, you need to clone our version and check out branch "spiram-fix-test" on that as well:
https://github.com/openvehicles/esp-idf/tree/spiram-fix-test

dexterbg, 2019-10-08:
…and the esp-idf also needs a `git submodule update`

glynhudson, 2019-10-10:
Whoo! It now builds successfully using OVMS esp-idf. 

Thanks so much for your help. I've updated the steps I posted above to reflect the final process. It feels like now is not a good time to write a compiling OVMS guide since it's currently quite messy due to the spiram mods required. Hopefully this will be resolved soon

closed 2019-10-10 by glynhudson

glynhudson, 2019-10-11:
It seems when I flash OVMS with the build: `spiram-fix-test` branch compiled using modded toolchain as described above OMVS goes into a crashed reboot loop. Are then any flashing or build settings I could be missing? 

```
rst:0xc (SW_CPU_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5556
load:0x40078000,len:11940
ho 0 tail 12 room 4
load:0x40080400,len:6888
entry 0x40080724
I (984) psram: This chip is ESP32-D0WD
I (985) spiram: Found 32MBit SPI RAM device
I (985) spiram: SPI RAM mode: flash 40m sram 40m
I (987) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (994) cpu_start: Pro cpu up.
I (998) cpu_start: Application information:
I (1003) cpu_start: Project name:     ovms3
I (1008) cpu_start: App version:      3.2.002-284-g30ffa4f-dirty
I (1014) cpu_start: Compile time:     Oct 10 2019 02:14:17
I (1021) cpu_start: ELF file SHA256:  291f355d34611f1f...
I (1027) cpu_start: ESP-IDF:          v3.3-beta3-772-gc3465e0
I (1033) cpu_start: Starting app cpu, entry point is 0x400818f4
I (1025) cpu_start: App cpu up.
I (1933) spiram: SPI SRAM memory test OK
I (1934) heap_init: Initializing. RAM available for dynamic allocation:
I (1934) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1941) heap_init: At 3FFBCD50 len 000232B0 (140 KiB): DRAM
I (1946) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1953) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1960) heap_init: At 4009BA04 len 000045FC (17 KiB): IRAM
I (1966) cpu_start: Pro cpu start user code
I (1971) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (99) ovms_main: Set default logging level for * to INFO
I (100) command: Initialising COMMAND (1000)
I (101) boot: Initialising BOOT (1100)
I (104) boot: Boot #228479890 reasons for CPU0=12 and CPU1=12
I (111) boot: Reset reason Interrupt watchdog (5)
E (116) boot: Early crash #256 detected
I (121) events: Initialising EVENTS (1200)
I (126) config: Initialising CONFIG (1400)
I (131) time: Initialising TIME (1500)
I (135) script: Initialising SCRIPTS (1600)
I (139) script: Using DUKTAPE javascript engine
I (145) metrics: Initialising METRICS (1810)
I (150) metrics: Expanding DUKTAPE javascript engine
I (159) notify: Initialising NOTIFICATIONS (1820)
I (161) notify: Registered notification type info
I (166) notify: Registered notification type error
I (171) notify: Registered notification type alert
I (177) notify: Registered notification type data
I (182) notify: Registered notification type stream
I (188) location: Initialising LOCATIONS (1900)
I (195) location: Expanding DUKTAPE javascript engine
I (199) vehicle: Initialising VEHICLE Factory (2000)
I (206) pcp: Initialising POWER (4000)
I (209) max7317: Initialising MAX7317 EGPIO (4200)
I (214) sdcard: Initialising SD CARD (4400)
I (219) ota: Initialising OTA (4400)
I (224) canformat: Initialising CAN Format Factory (4500)
I (229) canformat-crtd: Registering CAN Format: CRTD (4505)
I (236) canformat-pcap: Registering CAN Format: PCAP (4505)
I (242) canformat-gvret: Registering CAN Format: GVRET (4505)
I (248) canformat-lawricel: Registering CAN Format: LAWRICEL (4505)
I (255) canformat-raw: Registering CAN Format: RAW (4505)
I (261) can: Initialising CAN (4510)
I (267) dbc-app: Initialising DBC (4520)
I (271) canlog: Initialising CAN logging (4550)
[... 57 lines cut ...]
PC      : 0x400976bc  PS      : 0x00060c34  A0      : 0x80098ad5  A1      : 0x3ffc6720  
A2      : 0x3ffb0098  A3      : 0x0000abab  A4      : 0xb33fffff  A5      : 0x00000001  
A6      : 0x00060c23  A7      : 0x0000cdcd  A8      : 0x0000abab  A9      : 0x3ffc6730  
A10     : 0x3ffca798  A11     : 0x3ffca798  A12     : 0x00060423  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x3ffc6de8  SAR     : 0x00000017  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

ELF file SHA256: 291f355d34611f1f57a2043650abcde4b8e550465324b5431c1ad75bf8e008ab

Backtrace: 0x400976bc:0x3ffc6720 0x40098ad2:0x3ffc6750 0x40096156:0x3ffc6770 0x400964e8:0x3ffc6790 0x400965cd:0x3ffc67d0 0x400965e5:0x3ffc67f0 0x40085a3e:0x3ffc6810 0x40085a6c:0x3ffc6830 0x40085bcd:0x3ffc6860 0x40129cb5:0x3ffc6880 0x4011ec31:0x3ffc6b40 0x4011e72d:0x3ffc6b90 0x40096cdf:0x3ffc6bc0 0x4009853d:0x3ffc6be0 0x40099660:0x3ffc6c00 0x40099616:0xa5a5a5a5

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5556
load:0x40078000,len:11940
ho 0 tail 12 room 4
load:0x40080400,len:6888
entry 0x40080724
```

dexterbg, 2019-10-12:
Your git checkout status doesn't seem to be right:
```
I (1008) cpu_start: App version:      3.2.002-284-g30ffa4f-dirty
I (1027) cpu_start: ESP-IDF:          v3.3-beta3-772-gc3465e0
```

When building from the current spiram-fix-test branches, these should be:
```
I (892) cpu_start: App version:      3.2.005-51-g30ffa4ff
I (910) cpu_start: ESP-IDF:          v3.3-beta3-774-g6652269e1
```

Did you create local branches? Not sure if the version system works on a remote branch.

My git branch status:
```
balzer@leela:~/esp/esp-idf> git branch --list --verbose |cat
  master          c3465e05c Merge pull request #4 from leres/master
* spiram-fix-test 6652269e1 Merge branch 'master' into spiram-fix-test

balzer@leela:~/esp/Open-Vehicle-Monitoring-System-3> git branch --list --verbose |cat
  master          a8833a7e Twizy: immediate tuning metrics updates (was delegated to ticker before),   remember profile key loaded into working set,   remember tuning applied state
* spiram-fix-test 30ffa4ff Merge branch 'master' into spiram-fix-test
```

Also… did you still install the espressif/esp-idf first? You shouldn't install that at all, just the openvehicles version. Check what the install and export scripts did, you probably need to reverse that or at least the env variables set.

A corruption on `/store` could also be possible, but you should first check the above.

glynhudson, 2019-10-14:
Ah, I did not realise the openvehicles/esp-idf also needed to be on `spiram-fix-test` branch e.g https://github.com/openvehicles/esp-idf/tree/spiram-fix-test

My compiled build now seems to upload and run successfully :-) 

I'm now getting `build idf v3.3-beta3-774-g6652269 Oct 14 2019 20:30:04`

KommyKT, 2019-11-05:
I have a problem with my new computer. A Macbook Air With Catalina. If I
compile and update the code from openvehicles GitHub repo everything ok
version show 3.2.005-163-g25862f1b, but is use my forked repo and merged
from openvehicle show 3.1.010-906-g83eb230d, this is the small problem, the
big problem is my forked "app" always crash.
My toolkit is: crosstool-ng-1.22.0-98-g4638c4f with spiral-fix modifications
esp-idf version: v3.3-beta3-774-g6652269e1
kommykt@Kovacss-Air esp-idf % git branch --list --verbose |cat

  master          c3465e05c Merge pull request #4 from leres/master
* spiram-fix-test 6652269e1 Merge branch 'master' into spiram-fix-test

OVMS3:

kommykt@Kovacss-Air Open-Vehicle-Monitoring-System-3_stable % git branch
--list --verbose |cat

  master          c32f8000 Webserver: add logging via websocket stream /
web shell

* spiram-fix-test 25862f1b Merge branch 'master' into spiram-fix-test

Glyn Hudson <notifications@github.com> ezt írta (időpont: 2019. okt. 14.,
H, 22:26):

> Ah, I did not realise the openvehicles/esp-idf also needed to be on
> spiram-fix-test branch e.g
> https://github.com/openvehicles/esp-idf/tree/spiram-fix-test
>
> My compiled build now seems to upload and run successfully :-)
>
> I'm now getting build idf v3.3-beta3-774-g6652269 Oct 14 2019 20:30:04

dexterbg, 2019-11-05:
Tamás,

your tags are out of sync. 906 is the commit distance to the latest version tag, which seems to be 3.1.010 on your fork.

That wouldn't produce crashes though. I guess if your tags are out of sync you did something wrong on the merge as well.

Maybe you should reset your branch to a clean previous commit and reapply the merge.

Regards,
Michael