openvehicles/Open-Vehicle-Monitoring-System-3 #274: SIMCOM UART / Mux dropping characters

dexterbg, 2019-10-20:
Investigating the frequent `gsm-mux: Frame overflow` warnings, I found this:
- The frame detection gets out of sync because the actual data length received is smaller than signaled by the mux protocol
- The incidents observed lost 1-3 characters, always at position 47 of the frame (observed communication was NMEA, so larger PPP frames may lose characters further in)
- Neither placing all buffers in internal RAM, nor raising the UART queue size or task priority had any effect

This may be related to issue #104.

Typical example:
```
W (244304) gsm-mux: Frame overflow (2048 bytes)
V (244304) gsm-mux: Frame head: f9 05 ff 8f 24 47 50 52 4d 43 2c 31 37 35 39 34 | ....$GPRMC,17594
V (244314) gsm-mux: Frame head: 33 2e 30 2c 41 2c 35 31 31 38 2e 31 34 30 36 30 | 3.0,A,5118.14060
V (244314) gsm-mux: Frame head: 33 2c 4e 2c 30 30 37 32 33 2e 33 39 32 34 32 35 | 3,N,00723.392425
V (244314) gsm-mux: Frame head: 2c 45 30 2c 31 38 2e 32 2c 32 30 31 30 31 39 2c | ,E0,18.2,201019,
V (244314) gsm-mux: Frame head: 2c 2c 41 2a 35 33 0d 0a 49 f9 f9 05 ff 91 24 47 | ,,A*53..I.....$G
V (244314) gsm-mux: Frame head: 4e 47 4e 53 2c 31 37 35 39 34 34 2e 30 2c 35 31 | NGNS,175944.0,51
V (244314) gsm-mux: Frame head: 31 38 2e 31 34 30 36 30 33 2c 4e 2c 30 30 37 32 | 18.140603,N,0072
V (244314) gsm-mux: Frame head: 33 2e 33 39 32 34 32 35 2c 45 2c 41 41 2c 31 32 | 3.392425,E,AA,12

First frame in bufer:
f9 05 ff 8f 24 47 50 52 4d 43 2c 31 37 35 39 34 | ....$GPRMC,17594
33 2e 30 2c 41 2c 35 31 31 38 2e 31 34 30 36 30 | 3.0,A,5118.14060
33 2c 4e 2c 30 30 37 32 33 2e 33 39 32 34 32 35 | 3,N,00723.392425
2c 45 30 2c 31 38 2e 32 2c 32 30 31 30 31 39 2c | ,E0,18.2,201019,
2c 2c 41 2a 35 33 0d 0a 49 f9                   | ,,A*53..I.      

Payload length signaled: 0x8f >> 1 = 71 bytes
Actual payload length: 68 bytes (24 47 … 0d 0a)

Payload:
$GPRMC,175943.0,A,5118.140603,N,00723.392425,E0,18.2,201019,,,A*53
                                              ^ lost bytes here

Standard $GPRMC sentence:
$GPRMC,133329.0,A,5118.140106,N,00723.393834,E,0.0,169.3,201019,,,A*6E
```

dexterbg, 2019-10-21:
Some new findings, no solution yet:
- The issue is caused by a UART hardware FIFO overflow
- Running the SIMCOM task on core 0 with priority 21 had no effect
- The UART fifo overflow is triggered less often when the Wifi modem is switched off (i.e. although PPP communication is running via modem then additional to the NMEA data)

The UART driver installs an ISR that transfers data from the HW FIFO to a ring buffer. The FIFO has 128 bytes, the ring buffer 2048 bytes. So it seems the ISR gets sometimes blocked too long by other (higher level?) interrupts or enforced hardware waits.

The example code says:
> If fifo overflow happened, you should consider adding flow control for your application.

That doesn't seem to be an option for us… other ideas?

slcasner, 2019-10-21:
You mentioned adjusting the SIMCOM task priority, but could the priority of the UART interrupt be adjusted higher?

dexterbg, 2019-10-22:
Good idea, I'll try that. The default seems to be flags=0 → level 1-3. Also just saw the UART driver ISR is not placed in IRAM, that could also be worth trying:
https://github.com/openvehicles/esp-idf/blob/master/components/driver/uart.c#L1324

dexterbg, 2019-10-22:
Requesting level 3 and placing the default UART ISR in IRAM reduces the frequency but doesn't fix the overflows.

Higher level ISR need to be written in asm:
https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/hlinterrupts.html
ESP32 asm is currently out of my scope.

dexterbg, 2019-10-31:
Excerpt from developer mailing list post:

Analyzing our current core usage I found we've been running nearly all interrupt handlers on
core 1. That's because interrupts are bound to the core that allocates them, and after system
init, the housekeeper created all peripheral components on core 1.

I had some advances on the FIFO overflows by allocating a level 3 interrupt for the SIMCOM UART,
but not much. I then tried moving the SIMCOM interrupt handler to core #0, and that immediately
reduced the overflows drastically -- most of the time.

I read through the Espressif forums and issues on interrupt latency optimization and found a
precious hint here: https://www.esp32.com/viewtopic.php?t=11462#p46437

> Finally, by default the LWIP task is allowed to run on either core, and will sometimes take a
> spinlock which requires disabling interrupts. Try pinning this task to CPU 0, only

I did so, and the FIFO overflows disappeared completely on my bench. In my car there still
are some, but very few. I also tested pinning LWIP to core 1, that also worked and may be
an alternative.

I then searched for more config candidates affecting interrupt performance and also disabled
the interrupt backtrace feature. That saves just 4 instructions on interrupt entry, so has
little effect and can be re-enabled in case we need it.

I think that's as much as we can achieve without adding hardware flow control or writing our own asm UART ISR, so I'll close this issue for now.

closed 2019-10-31 by dexterbg