openvehicles/Open-Vehicle-Monitoring-System-3 #275: Possible Kia e-Niro 12V drain

glynhudson, 2019-10-20:
I've had reports of high 12V drain from a Kia e-niro owner running latest edge build. 

Possibly the car is being kept aware? The drain rate seems to be higher than just the OVMS module. It's unclear if this became an issue after https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/pull/267. Hopefully I'll get some more details for the owner soon

@Timopen have you noticed any issues with 12V drain using latest OVMS build?

Timopen, 2019-10-21:
Hi Glyn
The drain is probably a bit higher because the issue in the previous version made the OVMS canbus crash and shut down after a couple of minutes when the car was off and not charging.
We have to poll the canbus to know if the car is started or if it started charging.
But I have been running the latest version for a couple of months and sometimes had my car standing for a couple of days without problems with 12 v being completely drained.

glynhudson, 2019-10-21:
Some users have been reporting 12V voltage as low as 11.4V after the car has been sitting for a while (even as little as 10mins of sitting). What 12V voltage does you OVMS report after the car has been sitting for a while?

Timopen, 2019-10-21:
I have been monitoring my car today as it has been sitting all day, it dropped from 13.3 to 11.6 v after 24 hrs.
The car is using the main battery to charge up the 12v when it drops to around 11.5 v.
However, if you are connected with the app it will draw more power. My android phone disconnects automatically when it goes to sleep. I don’t know how Iphone handles this as I don’t own one.

[image]

Timopen, 2019-10-21:
Made a test with the app connected for 15 mins, but it only dropped 0.1 v .

glynhudson, 2019-10-21:
That's good to hear the car charges the 12V automatically. when it reaches 11.5V. It would be interesting to see how the battery voltage drops witthout OVMS connected. There could be other systems in the e-Niro which cause the battery to drain. The drain rate seem to be much higher than my Nissan e-NV200 with OVNS connected, battery dropped from 12.5V - 12.1 over 19hrs:

[image]

I've just measured the power consumption of just the OVMS module, it consumed 60mA = 0.72W with WiFi on and GSM/GPS modem off and 100mA = 1.2W with everything switched on.

toreblomhaug, 2019-11-17:
My Car has been total dead 2 times. But not any warnings on 12v. Charge 12v battery and all good again... But not do happy wife.

plord12, 2019-11-19:
I've also been seeing high drain on my Kona ( once the battery was dead and had to be jump started ).

Is there a technique to measure current use in the car itself ?  I know the 12v battery is low capacity but ...

Timopen, 2019-11-20:
Hi
I also had to jumpstart my car once.
But this was when the simcom was stuck in PoweringOn state, maybe a coincident but I have not had it since. Maybe we should pop the question to Marc/Michael if the simcom is consuming a lot when it’s stuck.

Best Regards

TIMO PENTTILÄ
System Engineer/Developer

glynhudson, 2019-11-20:
Another user has reported 12V going flat overnight when using OVMS with the eNiro, see voltage graph: 

[image]

OVMS only consumes approx 1.2W therefore the issue could be OVMS keeping the car awake which kills the battery. User was running FW `3.2.005 Nov 20th 2019 build`

dexterbg, 2019-11-20:
Glyn, why was no 12V alert triggered on that event?

I'm currently working on improving the 12V monitor, also thinking about changing the alert threshold to an absolute voltage level. But the current monitor should have triggered, unless the reference voltage wasn't correct. Can you provide a chart of the reference voltage on that event? It's stored in the D records as well.

plord12, 2019-11-23:
One thing I noticed that Niro/Kona stops polling when there are no clients - ie v2 & v3 server connections are stopped.

However, the code looks at ms_s_v2_peers/ms_s_v3_peers which is not reset to 0 when the server connection is disconnected.  So if the server is started then stopped, polling never stops.

In my tree I've set ms_s_v2_peers=0/ms_s_v3_peers=0 when the server is disconnected.

plord12, 2019-11-23:
Today I left my kone on the drive for a few yours with OVMS plugged in.  I just started getting some low battery alerts such as -

```
2019-11-23 20:09:52,49520,'#54 C rx msg P A12V Battery restored: 12.6V (ref=12.6V)'
2019-11-23 20:10:03,49579,'#54 C rx msg D 0,8,4,0,0,9,0,90710,0,21,0,0,1,0,9.25,0,12.6,0,0,34.6,0'
2019-11-23 20:10:22,49692,'#54 C rx msg S 92.0,M,0,0,stopped,standard,226,260,0,0,0,0,0,21,0,0,0,0,147.20,0,0,0,0,-1,0,0,0,0,0,246,0,0.00,396.60,100'
2019-11-23 20:10:22,49693,'#54 C rx msg D 0,8,4,0,0,9,0,90710,0,41,0,0,1,0,11.95,0,12.6,0,0,34.6,0'
2019-11-23 20:10:22,49694,'#54 C rx msg L 51.422745,-0.855867,0,59.2,1,1,0,0,0,0.833,0,0'
2019-11-23 20:10:22,49695,'#54 C rx msg F 3.2.005-127-gfd6952cc-dirty/ota_0/edge (build idf v3.3-beta3-772-gc3465e05c Nov 23 2019 12:51:43),KMHK581GUKU002988,13,1,KN,Petes wan'
2019-11-23 20:10:52,49874,'#54 C rx msg P A12V Battery critical: 9.6V (ref=12.6V)'     
```

Seemed to be very quick to move from 12.6V to 9.6V.

This is current master + my updates for Kona.  Car is connected to home wifi and cell.  Web console is very sluggish / unusable.  

Only odd thing I noticed in the logs is -

```
2019-11-23 19:08:11,25662,'#58 C got error: Broken pipe'
2019-11-23 19:08:31,25807,'#31 C got login'
2019-11-23 19:08:59,26029,'#12 C got login'
2019-11-23 19:08:59,26030,'#31 C error - duplicate car login - clearing first connection'
```

plord12, 2019-11-24:
I did try connecting a watt meter in-line in the car ( pin 16 for 12v and 5 for ground ) but didn't manage to measure any current -

[image]

Also tried pin 4 for ground.

plord12, 2019-11-24:
I see that when a app is connected, we go into POLLSTATE_CHARGING ( OvmsVehicleKiaNiroEv::EventListener ) regardless if we are actually charging or not.  

As I understand it this triggers polling of several PID's even though many are offline.

Is this excessive ?

goev, 2019-11-27:
The main problem with OVMS on the e-Niro, Kona (and partially on the Soul) is that setting the correct pollstate has been, eh, a "challenge". If the pollstate is set to running or charging, OVMS will keep some ECUs awake, which eventually will drain the 12V battery. But without polling, the car won't tell you if it wakes up, and if it's charging or running. Therefore I thought I could use ms_s_v2_peers+ms_s_v3_peers>0 to set pollstate when a client connects. This however, didn't work, as plord12 figured out,  and now hopefully have fixed 👍  
When it comes to which pollstate to use when a client connects when the car is off, I don't remember if I had a good reason to choose POLLSTATE_CHARGING.  However, I think that can be sorted out by firing off a CAN-command to the BMS (7e4h 8 03 22 01 01 00 00 00 00). If the BMS responds the car is on and it should be fairly easy to see if it is charging, based on the response. Then setting the correct pollstate should be easy.

plord12, 2019-11-27:
> This however, didn't work, as plord12 figured out, and now hopefully have fixed 👍

Oops, sorry, I've only just subitted pull request for this.

goev, 2019-11-27:
No, I meant that my old code didn’t work, and that you have found a fix by resetting those two values when clients disconnects 🙂
Your pull request is probably just fine!

BR
Geir

Sendt fra min iPhone

Timopen, 2019-12-04:
There is a setting in the menu for e-Niro under system settings -> other -> Aux. Battery saver+
When this is ticked the 12 v battery is recharged by the main battery. But only 10 times in a row for some reason, so leaving the car for longer periods could cause a dead battery anyway.
Manual page 60:
https://di-uploads-pod4.dealerinspire.com/lawrencekia/uploads/2019/01/2019-Niro-EV-OM-sm.pdf

Timopen, 2019-12-04:
I did some measuring of the cars 12 v consumption.
Without OVMS module it takes about 0,16-0,17 Amps when completely sleeping.
Same figures with OVMS in pullstate off 0,25-0,37 Amps. This is about 1.44 watts for the OVMS.
With OVMS in pullstate charging 1,81-2,40 Amps (car still sleeping, app connected). This is about 26,76 watts for the OVMS.
In pullstate charging it would take less than 20 hours to completely drain the battery, probably a lot less if the car has some kind of security system to avoid complete drain.
In pullstate off it takes 3-4 days to drain the battery ( from my own experience).
If you plan to leave your car for a longer time, disconnect OVMS. If you leave it on your driveway and don't drive it for a couple of days, you can just go out and start the car and then shut it off. This will reset the 10 times charging counter in the Aux. Battery Save+ function.
The car probably has some kind of deep sleep as well, because I had my car standing in the airport for 15 days without problems once. I cold not capture it this time though. One thing I could se was that the car did not go to sleep if the keyfob was nearby, so keep it away from the car.

happyjaxx, 2019-12-04:
> On 4 Dec 2019, at 22:04, Timopen <notifications@github.com> wrote:
> 
> ...
> With OVMS in pullstate charging 1,81-2,40 Amps (car still sleeping, app connected). This is about 26,76 watts for the OVMS
> ....

Hi,

26 watts ? Are you sure ? That seems to be a lot to dissipate for the OVMS.

JB./.

Timopen, 2019-12-04:
Not actually just for the OVMS, it’s the overhead generated compared to the car sleeping.
But I’m sure. Tested to connect and disconnect the app multiple times.

Med Vänlig Hälsning

TIMO PENTTILÄ
Systemtekniker/Utvecklare

happyjaxx, 2019-12-04:
Yeah, makes more sense, the phrasing mislead me a little bit (and not a fire hazard ;) )
I'm paying close attention to power issues, the VW Up! doesn't seem to bear winter on it's own, even plugged to the mains, it won't trigger heating over CarNet, and complains about the 12V battery, so with the OVMS, would have been worse !

plord12, 2019-12-07:
> However, I think that can be sorted out by firing off a CAN-command to the BMS (7e4h 8 03 22 01 01 00 00 00 00). If the BMS responds the car is on and it should be fairly easy to see if it is charging, based on the response. 

I added -

```		{ 0x7e4, 0x803, VEHICLE_POLL_TYPE_OBDIIEXTENDED,  	0x0101, 		{      9,    9,   9 } }, 	// CAN command
```

I didn't get a response when charging on my Kona.

toreblomhaug, 2019-12-31:
Any update on this issue?

plord12, 2020-01-03:
> Any update on this issue?

Today I've been tinkering a bit and thought I would make sure we are always polling the minimum whilst the car is off regardless of any client connected.  Ie stay in poll state 0.

With my local changes, after 4 hours sat on my drive I see 0.2v drop in the 12v battery.  So expected 1.2v over 24hours.

I see the "Aux. Battery Saver+"  mode will try to charge the 12v battery for 20mins up to 10 times when the car is off ... hopefully this is sufficient.

I'll continue to test over the weekend.

plord12, 2020-01-04:
> Today I've been tinkering a bit and thought I would make sure we are always polling the minimum whilst the car is off regardless of any client connected. Ie stay in poll state 0.

For the record, with pool state 0 and the following polling configuration -

```
		{ 0x7e4, 0x7ec, VEHICLE_POLL_TYPE_OBDIIEXTENDED,  	0x0101, 		{      9,    9,   9 } }, 	// BMC Diag page 01 - Must be called when off to detect when charging
		{ 0x770, 0x778, VEHICLE_POLL_TYPE_OBDIIEXTENDED,  	0xbc03, 		{      7,    7,   7 } },  // IGMP Door status + IGN1 & IGN2 - Detects when car is turned on
		{ 0x770, 0x778, VEHICLE_POLL_TYPE_OBDIIEXTENDED,  	0xbc04, 		{       11,   11,  11 } },  // IGMP Door status
```

I see -

[image]

Timopen, 2020-01-04:
Hi Peter
Why are you polling:
{ 0x770, 0x778, VEHICLE_POLL_TYPE_OBDIIEXTENDED,  	0xbc04, 		{       11,   11,  11 } },  // IGMP Door status
Is this PID really giving any response when the car is off?

plord12, 2020-01-04:
> Why are you polling:
> { 0x770, 0x778, VEHICLE_POLL_TYPE_OBDIIEXTENDED, 0xbc04, { 11, 11, 11 } }, // IGMP Door status
> Is this PID really giving any response when the car is off?

Yes - this ECU is always active and enables OVMS to determine if the doors are closed and car is locked after the car is turned off ( otherwise the app shows the car is unlocked ).

plord12, 2020-01-04:
A couple of times now I've seen a sudden battery drop -

[image]

plord12, 2020-01-05:
I also sometimes see repeating -

`E (751364) can: can1 watchdog inactivity timeout - resetting bus`

OVMS doesn't recover from this.

Not (yet) sure if this correlates to battery drain, but a reboot does resolve it.

plord12, 2020-01-11:
I've noticed that the 12v battery drains slowly when ovms is not polling.  Indeed, this morning I noticed my Kona topping up the 12v battery without ovms connected at all ( light is on at the front of the car ) ... so I'm wondering if I'm are chasing the wrong thing here.

Perhaps the more urgent issue is the sudden battery drop reported 7 days ago.

Timopen, 2020-02-09:
I have some input from the Swedish FB forum for e-Niro. There seems to be a bug in the cars software, when you schedule the charging for your leaving time. While the car is waiting to start the charging some people have experienced a flat battery. This is probably because the car is semi-awake but not charging the 12v battery when needed.

Marco521, 2020-02-15:
What about connecting OVMS to a fuse where the 12V is only present when the car is on or charging? So you don't discharge the 12V battery.

Timopen, 2020-03-16:
Yesterday my battery was flat again. So I took it inside to charge it, meanwhile I replaced it with the battery from my caravan trailer. The caravan battery only drained 0,1 volts during the last 24 hours with the OVMS connected. It's an old 70 Ah battery. I think the original battery in the KIA is a really crappy one.

glynhudson, 2020-03-17:
Has anyone experienced this issue with a Kona EV? I believe the Kona is very similar to the E-Niro. What's the best advice for a user who has a Kona with OVMS and experiencing a battery drain? Does Kona have the battery save + feature?

Wil63, 2020-05-01:
Last week I had my 12v battery fully drained (Kia Niro EV). I did not understand, as it is supposed to be recharged by the main battery. Also because I don't really use it (covid), it is true that night recharged is programmed though I don't plug it. On top, the day before I played a lot with trying javascript module, using SSH/SCP and web client. Also, my car constantly tries to connect to my house wifi, with a poor signal level, I guess altogether it does not help.

Marco521, 2020-05-08:
The Aux battery saver+ can not prevent discharge when the aux battery is used as a power supply for unauthorized electronic devices.  The aux battery saver+ function only operates for max 20 minutes.  But when it's finished it takes 20 minutes for the car to go dark. In that period still consuming energy. So it doesn't seem worth using in the first place. 
I have connected the 12V for OVMS to the heated steering wheel fuse. There is 12V also while charging. I recommend to put your car in utility mode when you want to play with OVMS.

hubaksis, 2020-06-08:
Is it possible to send a request to CanBus to some module that won't wake the car up and will not keep it awake? 
I mean that it can be some not-important request, like 'is radio on?', but the vehicle will respond only when it's awake and it will do nothing when it's asleep. 
Is there a module that does not wake up when the request is sent when the vehicle is off?

In this case it will be possible to keep the car in the original state as it supposed to be and send any further requests only when we get a response to that non-important request.

jheredianet, 2020-07-14:
Hi guys, after trying a lot of suggestions you made. What I do with my KonaEV is to power off the CAN bus, then I start it manually when needed. These are the commands I use:
power can1 off

power can1 on

Controlling this manually, the aux battery does not drain a lot, it could last days without problems. 
So I guess it's something with that module that need to be fixed.

RowdyDoc, 2020-11-17:
Has anyone found a workaround for this that is somewhat automated?
I guess I am suggesting something along the lines of:
1. unit does the equiv of the "power can1 off" console command after x minutes (eg 20min)  of CarOff&DoorsClosed&NoCharging
2. unit is able to be triggered somehow to "power can1 on" when DoorOpen|CarOn|Charging (Wired signal from 12V supply -   someone suggested steering wheel heater).
3. do a duty cycle of power can1 on/off when CarOff&DoorsClosed&NoCharging to check status (12V aux battery etc).  Maybe for 1 in 20 (eg 30s every 10 minutes)?

For my specific use case I would probably re-enable can1 polling a minute or so before I start a charge when garaged (EVSE start & SP managed by home automation).

Wireheadbe, 2020-12-18:
> The Aux battery saver+ can not prevent discharge when the aux battery is used as a power supply for unauthorized electronic devices. The aux battery saver+ function only operates for max 20 minutes. [...]

Did you just splice the cable for this? I'm thinking about this to. Battery drops way to fast when OVMS connected..

Marco521, 2020-12-19:
> Did you just splice the cable for this? I'm thinking about this to. Battery drops way to fast when OVMS connected..

[image]

With this splitter it's easy to make a slice.

Wireheadbe, 2020-12-31:
Bought this one: https://www.amazon.de/dp/B073Y519FG (not affiliated). Was able to push out the original connector. The ones from the splitter snap into place just like the stock one. Looks just like it came from the factory :) Spliced #16 wire to the switched 12V of the heated steering module. Awesome!

blomhaugen1, 2021-01-05:
@Wireheadbe  would you care to explain this in a way a "none hardware" guy can do this? have not use OVMS in years due to 12v drain if this can fix this i will use it again :)

toreblomhaug, 2021-01-05:
@Wireheadbe would you care to explain this in a way a "none hardware" guy can do this? have not use OVMS in years due to 12v drain if this can fix this i will use it again :)

Wireheadbe, 2021-01-05:
[image]

In the splitter I referenced, on the side where you plug the splitter into the original outlet, I cut the Nr16 wire - which is purple and a permanent 12V (don't cut into the original socket! :D )

I then connected this 12V wire, going to the socket where the OVMS plugs in; to a Micro2 Fuse "splitter" like this:
[image]

I plugged this in where the steering wheel heating resides. This is a switched 12V that's on when:

> - driving
> - charging

So perfect for our case.

Timopen, 2021-01-06:
@Wireheadbe Can you see if the car is locked in the app with this solution?
The main problem is the poor 12v battery in e-Niro. It's the same starter battery that's shipped with the KIA Rio for south European market. I replaced my battery with one that is made for caravans (not a starter battery), and it hasn't gone flat since that. This is almost one year ago.

Wireheadbe, 2021-01-06:
> @Wireheadbe Can you see if the car is locked in the app with this solution?
> The main problem is the poor 12v battery in e-Niro. It's the same starter battery that's shipped with the KIA Rio for [...]

Ah no - that you can't. Unless it's charging :) I wouldn't rely on an 3rd party app anyway to be really sure a car is locked :)

Wireheadbe, 2021-01-06:
One nice side-effect of this, is that you can actually see the car wake-up daily to top-off the 12V battery. So you could see the "last-known" lock-state by this as well.

happyjaxx, 2021-01-06:
@Timopen : Do you have any ref to your battery while we're at it ? Did dealers complain about your tweak? 
I had my e-Soul 64kWh go X-Mas Tree on me a few months back because of near depletion (dashboard blinking, UVO rebooting, steering without any resistance, total loss of regen, actually scary) by the time I decided to go to the dealer it was charged enough to not have any issue when starting it up anymore : https://www.youtube.com/watch?v=WV0oqe83-bo

Wireheadbe, 2021-01-06:
> 
> @Timopen : Do you have any ref to your battery while we're at it ? Did dealers complain about your tweak? 
My 12V battery was actually replaced a while ago because of another issue that KIA is troubleshooting on my e-Niro (but it didn't fix it). No UVO here, so I don't have any parasitic drain -at all- anymore. Actually the dealer was quite positive to my changes, as it helps them troubleshoot the issue I'm having. (Something related to either the HV battery or the BMS). But in your case; it seems like it's an interference issue. Quite odd!

Timopen, 2021-01-06:
This is the battery i use:
https://www.jula.se/catalog/bil-och-garage/biltillbehor/bilbatterier-och-tillbehor/bilbatterier/bilbatteri-003967/
I'm sure you can find similar batteries at your local stores.
It's a bit longer than the original and doesn't fit in the tray, but it's possible to fasten it very well anyway.
I change back to the original battery and remove the OVMS when I have service done.

Wireheadbe, 2021-01-06:
They should just put 12V Lithium instead of lead-acid in these cars. It's a shame really..
[image]
They've been available for years afaik..

RowdyDoc, 2021-01-07:
@Wireheadbe - very impressed by your very neat (and unhacked) solution for the 12V supply to the unit.
I have been holding out for some indication that it was feasible to cycle the power can1 on/off solution on a timer to allow for some limited monitoring of the vehicle when not charging|driving but it looks like that may not be as easy as I hoped (and I would need to spend a fair bit of time to get up to speed with the code to be confident of trying it myself).
Because of this and the availability issues with OVMS via the Fasttech source (I am in Oz), I have been dragging my feet on ordering the unit.
Another thought I had was to use the same 12V that you used and interface it to one of the GPIO inputs and use that to control the power can1 on/off  on a timer - but that is both coding AND extra wiring (but maybe more reliable that trying to detect car status via the CAN).
@tibcoplord seemed to be close to having a good result - but since he has not commented for about a year I suspect he may have not had satisfaction.  It would be really good to hear from him to see what his experience has been since.
In the meantime - I think I will purchase the OVMS anyway and worst case use your solution to at least just see the car charge status while I am charging and collect the trip data.  It is not exactly what I was hoping for but it is certainly better than nothing (which is what I currently have).

happyjaxx, 2021-01-08:
@Timopen : thx for the ref... probably won't be able to find the same but I'll try to find a close replacement
(Found Varta's C22 and C30 that should fit for 52 and 54Ah respectively, with good specs and which is still more than the original 45Ah one)

@Wireheadbe : that would be a wet dream, unfortunately, they cost a kidney

Marco521, 2021-01-12:
[image]

@happyjaxx In Korea someone replaced the battery with a AGM 60Ah. They did cut 1 cm of the right side of the base plate to fit this bigger battery.  When i need to buy a new one i go for Varta D52 60AH AGM. AGM could be discharged a bit more then lead accid.

happyjaxx, 2021-01-13:
@Marco521 D52 in my amazon basket ;) Will give it some thought before going any further 👍

happyjaxx, 2021-01-26:
Well... the battery died on me when I was about to take off for a medical appointment last thursday... ordered a Varta C30 (Original size but 54Ah instead of 45Ah) over Amazon, received it yesterday, topped it up using a charger and fit it back in (Didn't want to start trimming the battery plate to fit a bigger one yet, next one maybe ;-) )
And charging limits are different between classic SMF's and AGM's, didn't want to play with that since I'm having the 30000km service soon, the dealer wasn't really helpful up to now, I'm not in a mood for a fight ;) 

Happy nothing browned out, and underground, my Soul thought it was at Hyundai's Design Center in Germany instead of the mustard city of France

Turning off SIMCOM (via Power Management) changes dramatically the discharge rate (values are hardly measurable but it's between 2 and 8 mV per hour instead of roughly double)... 

[image]
[image]

RowdyDoc, 2021-02-06:
Interesting -  I am in Oz and the Kona steering wheel circuit only comes on with the "accessory" state of the car (ie not energised with the car off but charging).  Must be different here than in northern hemisphere...  Tested all of the fuse circuits and none of them behaved in that way.  In fact none of them energised at all with charging - just static with OFF/ACC/ON state.
Some my plan to emulate @Wireheadbe has hit a bit of a roadblock :(
Guess I will have to look at a bit of a code hack to try to produce some sort of periodic CAN sleep-off/wake-on cycle behaviour when the unit sees that the car is not on or charging.
If anybody has any high level tips on where to start on that journey it would be much appreciated.

Even if I do that though, I still really wanted an immediate wake-up when charging starts - maybe I could send a remote cmd to the unit from home automation to enable CAN when I am about start charging... (via shell/console)?

Wireheadbe, 2021-02-21:
> 
> 
> Interesting - I am in Oz and the Kona steering wheel circuit only comes on with the "accessory" state of the car (ie not energised with the car off but charging). Must be different here than in northern hemisphere... 

Wait, that's exactly like it is here: live when either charging or "driving".

[4 more replies]