openvehicles/Open-Vehicle-Monitoring-System-3 #278: Add support for kona/e-niro odometer

plord12, 2019-10-26:
The kona odometer can be found on PID 22B002 ECU 7C6 ( cluster module ).  Hex dump on my kona is -

```
pid 22b002 @ 2019-09-28T00:12:22.593056
CLU 7CE 21 00 00 B6 00 00 00 00 [?=0] [?=0] [?=182] [?=0] [?=0] [?=0] [?=0]
CLU 7CE 22 1E 80 AA AA AA AA AA [odometer=7808] [PAD] [PAD] [PAD] [PAD] [PAD]
```

On my car (UK) this is 7808 *miles* - I've read other cars report this in km.  

I believe data is 3 bytes long.

plord12, 2019-10-27:
Ah.  I see we already have -

{code}
			if (m_poll_ml_frame == 1)
				{
				StdMetrics.ms_v_pos_odometer->SetValue(CAN_UINT24(3), Kilometers );
				}
{code}

but this doesn't work with my kone.  The following does work though -

{code}
			// UK Kona
			if (m_poll_ml_frame == 1)
				{
				odo=CAN_BYTE(6)<<16;
				}
			if (m_poll_ml_frame == 2)
				{
				odo+=CAN_UINT(0);
				StdMetrics.ms_v_pos_odometer->SetValue(odo, Miles);
				}
{code}

car shows 8263 miles on the dash, and I see -

{code}
v.p.odometer                             13298km
{code}

So assuming the checked in code is correct for e-neros displaying km, looks like we need to test for e-nero/kona and km/miles.

glynhudson, 2020-01-16:
Can this now be closed? Nice work @plord12 :+1:

RowdyDoc, 2021-07-25:
Looks like the following is incorrect for the Kona:
					// ms_v_bat_12v_current doesn't seem to be right
					StdMetrics.ms_v_bat_12v_current->SetValue((((int16_t)CAN_BYTE(4)<<8)+CAN_BYTE(3))/1000.0, Amps);
Based on the data at:
https://github.com/JejuSoul/OBD-PIDs-for-HKMC-EVs/tree/master/Hyundai%20Kona%20EV%20%26%20Kia%20Niro%20EV
The correct formula for 12V charge amps should be:
(32700 - [CAN_BYTE(3)*256+CAN_BYTE(4)] )/100.0
However it clearly needs some checking against measured values - which I will do over the next few days.
Did a quick back-calc tho and it looks more like sensible values.

plord12, 2021-07-25:
> On 25 Jul 2021, at 11:18, RowdyDoc ***@***.***> wrote:
> 
> 
> Looks like the following is incorrect for the Kona:
> // ms_v_bat_12v_current doesn't seem to be right
> StdMetrics.ms_v_bat_12v_current->SetValue((((int16_t)CAN_BYTE(4)<<8)+CAN_BYTE(3))/1000.0, Amps);
> Based on the data at:
> https://github.com/JejuSoul/OBD-PIDs-for-HKMC-EVs/tree/master/Hyundai%20Kona%20EV%20%26%20Kia%20Niro%20EV <https://github.com/JejuSoul/OBD-PIDs-for-HKMC-EVs/tree/master/Hyundai%20Kona%20EV%20%26%20Kia%20Niro%20EV>
> The correct formula for 12V charge amps should be:
> (32700 - [CAN_BYTE(3)*256+CAN_BYTE(4)] )/100.0
> However it clearly needs some checking against measured values - which I will do over the next few days.
> Did a quick back-calc tho and it looks more like sensible values.
> 
> 

Yes - some confirmation here http://techno-fandom.org/~hobbit/cars/ev/auxbat.html

RowdyDoc, 2021-07-25:
Cool - hadn't found that page previously.  Will digest.
Also looking at the odometer data.  The Kona mod is returning zero km for me.  Need to pull the raw data and have a closer look.
My Kona is a Sep 2020 model for Australia - so maybe something is a bit different...

RowdyDoc, 2021-08-15:
Weird - odometer reading is almost always 0 for my Kona - but have seen it displaying the actual correct kms every now and then.  Haven't picked out a pattern as yet...

ecodweeb, 2025-01-23:
@RowdyDoc or @tibcoplord I have a 2019 Kona and I get 0 reported for the ODO reading. Is there something I can modify on the OVMS code to fix this? It is the only vehicle I have running OVMS that doesn't display the odometer reading.

RowdyDoc, 2025-01-24:
Sorry @ecodweeb - I gave up with OVMS on the Kona as I never attempted to do the scripting that was required to prevent the OVMS module keeping the Kona systems "awake" most of the time and draining the 12Vdc battery.

ecodweeb, 2025-01-24:
@RowdyDoc that's a shame, but I am curious. I only had 12Vdc issues when the factory telematics was subscribed and enabled. Since moving to OVMS + Comma AI (sharing 12Vdc power from my OBD) I've had the longest run on a 12Vdc battery. My unit is not cellular, however, perhaps that makes a difference.

lopezjm2001, 2025-01-24:
Yeah, I had to replace the 12v battery with a Lifepo4 type because the OVMS caused the battery to go flat overnight. Even now I get an error message every morning that my 12v battery discharged whilst stationary. My BMW i3 has a BMS just for the 12v battery.

dexterbg, 2025-01-24:
You need to determine the kind of 12V drain you experience.

If the OVMS keeps some Kona ECUs awake, the polling scheme probably can be changed to avoid that.

The Kona does these polls to query the vehicle state:
```
		{ 0x7e4, 0x7ec, VEHICLE_POLL_TYPE_OBDIIEXTENDED,  	0x0101, 		{      9,    9,   9 }, 0, ISOTP_STD }, 	// BMC Diag page 01 - Must be called when off to detect when charging
…
		{ 0x770, 0x778, VEHICLE_POLL_TYPE_OBDIIEXTENDED,  	0xbc03, 		{      7,    7,   7 }, 0, ISOTP_STD },  // IGMP Door status + IGN1 & IGN2 - Detects when car is turned on
```

Try a lower frequency on these in `off` state or pausing these for some time after a state change.

RowdyDoc, 2025-01-25:
I had two issue I couldn't resolve:
1. Preventing the drain by not keeping the Kona subsystems awake
2. A shutdown of the unit when the battery voltage got too low (as a 

A problem with the reference voltage for the Kona and the way the OVMS uses it was problematic and I kind of gave up at that stage due to too much work and too little free time:
https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/issues/931

For me it was not valuable to have a neat system to collect info etc when it critically reduced the reliability of the fundamental requirement (car works when you need it) - so that was the show-stopper.

I would still like to get it up and running again, but not really up to speed with it enough, and would have to invest a fair amount of time learning what can be done with the scripting etc with no guarantee of success.

The fundamental requirement is to go into a low frequency wake-up of polling every say 30-60min when the car is neither on nor charging.  That would stop the excessive battery drain when it matters, but give responsive updating when it matters.