openvehicles/Open-Vehicle-Monitoring-System-3 #282: `can log stop` crashes

dexterbg, 2019-11-02:
This seems to happen with all vfs logging.
The backtraces circle in on fwrite() – maybe a race condition between fclose() and fwrite()?
[bt1.txt](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/files/3800756/bt1.txt)
[bt2.txt](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/files/3800757/bt2.txt)
[bt3.txt](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/files/3800758/bt3.txt)

dexterbg, 2020-04-11:
Am 10.04.20 um 10:54 schrieb Tamás Kovács:
> the crash info:
>
> Last crash: LoadProhibited exception on core 1 Registers: PC : 0x400d742b PS : 0x00060530 A0 : 0x800d6f39 A1 : 0x3ffc2a50 A2 : 0x3ffeda10 A3 : 0x3ffc2a90 A4 : 0x0301a8c0 A5 : 0x00000000 A6 : 0x3ffb70f0 A7 : 0x00000000 A8 : 0x800d7421 A9 : 0x3ffc2a30 A10 : 0x3ffeda10 A11 : 0x00000000 A12 : 0x00000008 A13 : 0x3ffb6e04 A14 : 0x00000003 A15 : 0x3ffb6d60 SAR : 0x00000008 EXCCAUSE: 0x0000001c EXCVADDR: 0x0301a8f0 LBEG : 0x40090a61 LEND : 0x40090a95 LCOUNT : 0xffffffff 
>
> Backtrace: 0x400d742b 0x400d6f36 0x400f9b01 0x400f9d47 0x400f9dbd 0x400f9dcd Version: 3.2.011-169-g4b2f3c64/ota_0/eap (build idf v3.3-beta3-776-g3d198cd50 Apr 5 2020 10:56:45)
>
> Tamás Kovács <kommykt@gmail.com> ezt írta (időpont: 2020. ápr. 10., P, 8:00):
>
>     Hi!
>
>     i want to create some log file with ovms and savvycan, but ovms all time crashed very often.
>     i start the logging with the following command: 
>     can log start tcpserver transmit gvret-b :23 424
>     I now want only the 424 pid, i connect the savvycan with ip address (not direct to the ovms, with the ap both the ovms and laptop connected).
>     I used the  3.2.011-169-g4b2f3c64/ota_0/eap fw updated automatically with ota from dexters-web eap.

Following Tamás' backtrace I found a use-after-free situation: the canlog destructor did not deregister the event callback. That could also explain my previous issues with "can log stop", with earlier backtraces probably misleading.

First tests using crtd vfs logging did no longer cause crashes after "can log stop".

@Tamás: please test with your setup.

KommyKT, 2020-04-12:
i think it fixed now get ~200k frames captured but the module crashed
(18:59 CEST), and when parking the car not restart logging after crash a
new crash (20:59 CEST) with the same  backtrace: 0x4008b03f 0x4008b2d5
0x400e709c 0x40083d96

Michael Balzer <notifications@github.com> ezt írta (időpont: 2020. ápr.
11., Szo, 11:56):

> the crash info:
>
> Last crash: LoadProhibited exception on core 1 Registers: PC : 0x400d742b
> PS : 0x00060530 A0 : 0x800d6f39 A1 : 0x3ffc2a50 A2 : 0x3ffeda10 A3 :
> 0x3ffc2a90 A4 : 0x0301a8c0 A5 : [...]

dexterbg, 2020-04-12:
Thanks for testing. CAN logging does not restart automatically on reboot, that's no fault. Your new crash was caused by a WDT, so that's a new bug.

It's currently impossible to tell which task was stuck from the crash debug data, maybe you can get a hint from the logs. If so, please open a new issue with that info.

closed 2020-04-12 by dexterbg