openvehicles/Open-Vehicle-Monitoring-System-3 #283: Any work done on BMW i3?

WeeBull, 2019-11-06:
I was wondering if anyone has tried the OVMS on the BMW i3 (or any BMW really)?

I'd be interested in giving it a go, but don't want to waste time if somebody has already hit a brick wall with it.

markwj, 2019-11-07:
Not that I know of. The OVMS part shouldn't be hard. The tricky bit is getting the information on CAN bus messages. They are vehicle specific and not publicly available.

WeeBull, 2019-11-07:
Yep, I know. I'm prepared to try to do some reverse engineering, but first of all I need a way of exploring, and so looking at the OVMS.

My concern is that I've never seen anything on the web about reading information on the CAN bus of an i3, only ever doing stuff (i.e. coding the car) through the ethernet connection (also on OBD on BMWs). 

I've tried a few things:

- A couple of bluetooth OBD dongles with program like Torque on Android. One MCP2515 based and the other a Chinese "ELM327". Both reported nothing, even though I've had the MCP one working to get basic information out of a previous car.
- I got a logic analyser on the i3 OBD CANbus, and saw life, but just a single repeating message no matter what mode the car was in. The analyser doesn't have any way of sending messages though, so al I could do was observe.

So I was hoping somebody might have seen even the generic messages, like speed, working, because I'm not having much luck so far.

The car has CAN running through-out it for communication between the various components (I've got schematics), but they all feed into a single unit (The BDC - Body Domain Controller) which drives the OBD socket, so I don't know how filtered the OBD interface is likely to be.

elbow, 2020-11-13:
Hi,

I think I can do i3 support.  I gained understanding by looking at ediabas and decompiling the *.prg files inside Deep OBD.  Also captured bluetooth traces and logging.

I’ve ordered an OVMS from Openenergymonitor and when it arrives here in South Africa I’ll make a start.

Help of course is welcome; happy to share what I have and love an alpha tester or two.

At this stage trying to get on the ovmsdev list - opened an issue about that.

dexterbg, 2020-11-13:
That's great, welcome to the team :-) For alpha testers, see https://www.openvehicles.com/node/2420, and I suggest asking in the german i3 forum: https://www.goingelectric.de/forum/viewforum.php?f=56

Regards,
Michael

elbow, 2020-11-14:
Hi Michael,

Thanks for the pointers.

Steve

dexterbg, 2021-12-30:
Meanwhile, Steve has contributed a very mature BMW i3 implementation:
- https://docs.openvehicles.com/en/latest/components/vehicle_bmwi3/docs/index.html

So closing this. Thanks, Steve!

closed 2021-12-30 by dexterbg