openvehicles/Open-Vehicle-Monitoring-System-3 #284: GPS week number rollover on SIM5360

markwj, 2019-11-07:
We have a report of our SIM5360 suffering from the GPS week rollover. According to a document from SIMCOM, this will affect SIM5360 starting 2019-11-03 (not the usual April 2019 problematic date).

https://www.cika.com/soporte/Information/GSMmodules/GPS-week-rollover_Simcom.pdf

`Time Zone:  CET-1CEST,M3.5.0,M10.5.0/3
UTC Time:   2000-03-22 20:28:27 UTC
Local Time: 2000-03-22 21:28:27 CET
Provider:   gsm-nmea

PROVIDER             STRATUM  UPDATE TIME
*gsm-nmea                  2       0 Wed Mar 22 20:28:27 2000`

A possible workaround will be to add 1024 weeks to the date provided by the GPS NMEA sentence. That will give us another 19 years until we have to think about this again.

markwj, 2019-11-07:
Workaround added to simply add 1024*7*86400 to the julian date/time calculated from the NMEA message. Hopefully this will workaround the issue and give us another 19 years before it comes back around.

1a4364f5d9383d280c03b0501cd3c9fa5d61916d GPS week number rollover on SIM5360 #284

Available in 3.2.005-102-g1a4364f.

dexterbg, 2019-11-07:
The SIMCOM info appears to be wrong in regard to "all firmware versions affected". My older module doesn't have the issue, my newer one has it.

The older module has a newer SIMCOM firmware version:
```
OVMS# simcom cmd AT+CGMR
D (1091803) simcom: rx line ch=4 len=23  : +CGMR: 35316B11SIM5360E
```
This version does not show the bug.

The newer module showing the bug has:
```
D (44168260) simcom: rx line ch=4 len=23  : +CGMR: 35316B09SIM5360E
```

I think the correct solution is to check for date < 2019-11-03 as previously suggested on the mailing list. I'll change the code accordingly.

dexterbg, 2019-11-07:
Now working on both modules.

closed 2019-11-07 by dexterbg