openvehicles/Open-Vehicle-Monitoring-System-3 #293: Empty module version metric on some Smart ED

dexterbg, 2019-12-09:
The "m.version" metric isn't set on boot. This only affects some Smart ED.

Firmware build of the user who reported this was the current 3.2.006 release candidate, it seems to have happened with the user switching from some specific smart firmware build `3.2.005-239-g63204c03-dirty/ota_0/edge (build idf v3.3-beta3-770-ge97f72ea2 Nov 15 2019 12:48:17)` to the current standard build.

Dimitrie78, 2019-12-10:
also the timezone does not load on boot.
ill lock into it on weckend

markwj, 2020-01-31:
The issue here is an event queue overflow on startup (so the 'system.start' event is lost, and MDNS, version, hardware type, etc are all impacted.

Changed default sdkconfig OVMS_HW_EVENT_QUEUE_SIZE from 20 to 40, to try to workaround this and allow enough space during startup. The issue seems to have been triggered by new events added to the firmware (such as egpio, power) contributing to the high number of events issued during startup.

closed 2020-01-31 by markwj