openvehicles/Open-Vehicle-Monitoring-System-3 #301: 2016 Nissan LEAF Remote Climate Control

glynhudson, 2019-12-14:
The [LEAF OVMS docs currently](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/blob/master/vehicle/OVMS.V3/components/vehicle_nissanleaf/docs/index.rst#2016-2017-models) state that remote climate won't work on 2016 Nissan LEAF's.

However, testing myself I've observed that remote climate trigged by OVMS does work if the [Nissan TCU is unplugged](https://photos.app.goo.gl/MuvpCaXQUjbCdoox6), it's in the same location as earlier LEAF models. However, when the 2016 TCU was unplugged Bluetooth calling and voice commands stopped working, presumably since the internal microphone is now routed thought the TCU for some reason on the 2016 model. 

It would be great to be able to find a work around for this. Has anyone else experienced this? Shall I update the docs?

markfinn, 2019-12-15:
The docs do list 2016 as working with a TCU disconnect.  It then later also says it may not work due to the can bus move of the TCU.  Was the pre V3 hardware only on one CAN bus? Maybe that's the "may" in "may not".

dalathegreat, 2019-12-15:
No need to update the docs, I suggest this issue can be closed.

Here is more reading on the bluetooth, maybe something that can help you? https://www.mynissanleaf.com/viewtopic.php?f=36&t=23887&start=10 (Note that the cert for the forum has expired, and they haven't had time to renew it)

dalathegreat, 2020-01-05:
@glynhudson  I suggest we close this?

glynhudson, 2020-01-16:
Sure, happy to close this. Should I add a note to the docs to note that unplugging the TCU on a 2016 leaf will enable remote climate at the expence of BT? I just wanted to confim with other that this was indeed the case.

closed 2020-08-24 by glynhudson