openvehicles/Open-Vehicle-Monitoring-System-3 #311: Start Battery Heating remotely on Nissan Leaf

dalathegreat, 2019-12-27:
Now when we are approaching the really cold months, it would be extremely beneficial for a Nissan Leaf to have more control for the battery heating. As you northern Leaf drivers might know, if the temperature of the battery drops too low, the Leaf will not fast-charge. What if we could control the heating of the battery like on a Tesla? This would be a game changer! 

First of all, to see if your Leaf is equipped with DC heating elements inside the battery, there is a signal in the 0x5C0 frame that has 0/1 if pads are present or not.
[image]

In 0x5C0 frame 0 bit 5, there is a heating **start** send request and in bit4 there is a **stop** heating request.
[image]

The theory is that to start heating, you write bit5 to 1. BUT there is also a send request bit that might be necessary to write also. It's in frame 1 bit 0 (Batt_Heater_Mail_Send_Request)
[image]

Anyone got a Leaf equipped with heater pads? :)

The-Alchemist, 2020-01-04:
@dalathegreat : How can you check if you have those heaters in the battery?  Is that the LEAF Winter Package?

I think I have it, because I've got a 2012 LEAF SL. =)

Could you clarify what you mean by "this should only be possible on AZE0 Leaf"?

Thanks!  Looking forward to working on this. 🎆

dalathegreat, 2020-01-05:
@The-Alchemist The documentation I found doesn't indicate that the heater pad fields are present on the bus. Maybe the ZE0 (2011-2012) model doesn't communicate the heating with the VCM. Maybe it just handles it internally?

You can try to look at the 0x5C0 frame4 bit0, if it's high then you have the heater. This worked on AZE0 (2013-2017) atleast.

mjkapkan, 2020-11-23:
This is on my to-do list as well. I already have a plugin for controlling the heating in the cabin with an additional eating element, so using that logic to control battery temperature should be relatively easy. Will try to test sending the command on my Leaf.

mjkapkan, 2020-12-01:
@dalathegreat Let's add this as new metric. Which one do you prefer?
`heater `or `heaterpresent`?
Also should we have this under standard metrics `v.b.` or specific `xnl.v.b.` ones?

We also need to identify the frames that indicate if the heater is on or off.

Also my first taught was to temporary fake the battery temperature messages to force the heater to start. But I don't have a can bridge yet, so it might take too much effort to do just with OVMS.

dalathegreat, 2020-12-01:
@mjkapkan  `heaterpresent` is very descriptive. No difference about the metrics for me.

The temps are slowly getting closer to being able to test this, -5*C here at the moment, still a bit to get to -17*C

mjkapkan, 2020-12-01:
@dalathegreat Hm, looks like I have it:
```
OVMS# me li heater
v.b.heaterpresent                        yes
```

mjkapkan, 2020-12-04:
Tried sending those messages, but no luck:

`can can1 tx standard 5C0 20
can can1 tx standard 5C0 20 1
can can1 tx standard 5C0 10 1`

We also have an option to fake the battery temp. but not sure it is possible to do that with OVMS.

mjkapkan, 2021-01-14:
@dalathegreat I think we will have below -17 Celsius in the upcoming days. So let's try to record the CAN messages and see what happens. I have made this plugin to send a notification if the temp drops below/above certain threshold:
https://github.com/mjkapkan/OVMS-Plugins/blob/master/battempmon/battempmon.js
we could also trigger start the recording of can messages, but I don't have an SD at the moment.

mjkapkan, 2021-01-19:
Unfortunately the coldest temperatures have passed and we had just one morning with -24 Celsius. My battery did not drop below -15 so the heater did not kick-in. I guess we need a volunteer driving a Leaf in a northern location not yet affected by the climate change. Without that it will be hard to reverse engineer the required messages.

dalathegreat, 2021-01-20:
Same here, even though it was cold my Leaf sat at 3temp bars due to getting so much use daily. We really need the help from someone above the polar circle :D

Araknai, 2021-11-13:
Hey guys, looks like it never got cold enough for you to sniff out the battery heater commands eh?
I was hoping I didn't have to fix my own aftermarket battery heater this winter..
Anyway..it will almost certainly get down to -28C at some point this winter where I live, so if needed I may be able to attempt to be up to speed enough to set up some CAN recording..