openvehicles/Open-Vehicle-Monitoring-System-3 #332: Metrics Set

Otiz1, 2020-02-09:
Hi,

when i use the command "metrics list v.e.headlight" i got an answer with "no" thats right.
But when i type "metrics set v.e.headlights yes" it says "metrics set" and when i type again "metrics list v.e.headlight" it says yes. Car headlight are all time off. Maybe someone could help me please ?

Car: Nissan ENV200 Dec. 2018

Thanks for your help

KommyKT, 2020-02-09:
Which vehicle type set (to get the vehicle type: "me li v.type") in OVMS.
I think the car update the headlight metric after your set.

Otiz1, 2020-02-09:
when i type in what u said the answer is "v.type                                   NL"
don't know what to do now. Thanks for your answer

dexterbg, 2020-02-11:
A metric is a measurement of a physical property. Setting it's value isn't supposed to change reality, it's just a means to inject test values or control offline metrics.

I admit it would really be nice to be able to just set the SOC to 100% any time just with a shell command. But unfortunately that will only work as a simulation.

If you want to control your headlights or anything else, the vehicle adaption needs to provide a command for that.

closed 2020-02-11 by dexterbg

Otiz1, 2020-02-11:
Thanks for the info. So in the development status it isn't possible. But maybe you can tell me how to inject Canbus Codes for testing. I got a few codes that I could try to send. Maybe one works and i can report it. 

wish a nice day

dexterbg, 2020-02-12:
CAN frames can be sent using the `can canX tx` command.